RobotLocomotion/drake #3225: Enable googlestyle in Drake Windows CI builds

sherm1, 2016-08-23:
Our cpplint wrapper test was failing on Windows but it went unnoticed in CI.

@jamiesnape said in #3199:
> There is no Python on the Windows CI images because it not one of the requirements on Windows per the documentation.

@jwnimmer-tri said in #3201:
> If you want us to keep [cpplint_wrapper_test.py] working on windows, you'll have to work / ask to get the MSVC builders to cover this (which I believe means: installing python, and changing ctest_driver_script to enable googlestyle on MSVC configurations).

Please consider enabling this on Windows to avoid future regressions.

jamiesnape, 2016-08-23:
If we are happy with diverging from the installation instructions and assuming that anything else that Python enables on Drake functions as expected, then installing Python is easy enough.

sherm1, 2016-08-23:
@jwnimmer-tri, @RussTedrake: any reason not to require Python as a prerequisite for Windows source builds as for the other platforms? If we ever succeed with director-building (#1786) we'll need it anyway.

jwnimmer-tri, 2016-08-23:
You can add it to the instructions without making it required (we do that with other things already).  Someone just has to write down the steps (precise -- saying exactly whose python and what version to use), update the windows from_source instructions, test that Drake still works locally after doing them, and then have CI add in that same python to the system image as well -- opting into a bunch of the post-merge builds in the PR.

In short: no theoretical reason to avoid python, just that someone has to blow several hours or days on windows support for this, which may or may not be worth the investment.

david-german-tri, 2016-09-22:
Closing this as will-not-fix.

closed 2016-09-22 by david-german-tri