RobotLocomotion/drake #3226: SNOPT-related valgrind issues

ggould-tri, 2016-08-23:
The linux-clang-nightly-memcheck-valgrind build is flagging a lot of SNOPT interactions as memory misuse (bad pointers and uninitialized accesses).  Some of this may be real; much is likely a result of SNOPT's interface and implementation being awful in their use of memory.

All of these issues should be addressed (where real) or suppressed (where, more likely, spurious).

See, eg: https://drake-cdash.csail.mit.edu/viewDynamicAnalysis.php?buildid=88438

sammy-tri, 2016-08-23:
At least one of them is real.  The patch below makes valgrind happy.  I've never submitted it in a PR because I've never been sure what the hell I was doing in there and couldn't figure out how to defend the result of randomly slapping a +1 in the middle of the file.

    Author: Sam Creasey <sam.creasey@tri.global>
    Date:   Thu Jul 21 15:42:24 2016 -0400

        Allocate some extra memory to copy the pointer into.

    diff --git a/drake/solvers/snopt_solver.cc b/drake/solvers/snopt_solver.cc
    index 6409e1b..3a330a6 100644
    --- a/drake/solvers/snopt_solver.cc
    +++ b/drake/solvers/snopt_solver.cc
    @@ -57,15 +57,15 @@ struct SNOPTData : public OptimizationProblem::SolverData {
       std::vector<snopt::integer> iGfun;
       std::vector<snopt::integer> jGvar;
 
       void min_alloc_w(snopt::integer mincw, snopt::integer miniw,
                        snopt::integer minrw) {
         if (lencw < mincw) {
           lencw = mincw;
    -      cw.resize(8 * lencw);
    +      cw.resize(8 * (lencw + 1));
         }
         if (leniw < miniw) {
           leniw = miniw;
           iw.resize(leniw);
         }
         if (lenrw < minrw) {
           lenrw = minrw;

RussTedrake, 2016-09-08:
i have seen the need for that +1 before in our snopt code.  i think we should probably include it, and add a comment line saying that we can't currently justify it except via valgrind.

sammy-tri, 2016-09-12:
I think I'm OK with adding the +1, but I'm not 100% sure our current allocation/usage of the SNOPT workspaces is actually sound.  I'm going to take a couple more hours to dig into things and at least add some comments as to why I think the current approach is OK (if I decide that it is) rather than just landing (basically) the patch above.

PR #3434 (2016-09-13): Allocate extra memory to copy our data pointer into.
Fixes #3226

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/3434)
closed 2016-09-13 by sammy-tri