RobotLocomotion/drake #3232: cpplint_wrapper.py Prints Extraneous Warnings

liangfok, 2016-08-23:
# Example Extraneous Warnings

I believe many of the following are false positives because they involve automatically-generated code.

```
$ ./drake/common/test/cpplint_wrapper.py 
Checking .........................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................
 TOTAL 571 files checked, found 6 warnings
Ignoring "/home/liang/dev/drake-distro-2/drake/thirdParty/bsd/spruce/src/spruce.cc": file excluded by "/home/liang/dev/drake-distro-2/drake/CPPLINT.cfg". File path component "thirdParty" matches pattern "thirdParty"
Ignoring "/home/liang/dev/drake-distro-2/drake/thirdParty/bsd/tinydir/tinydir.h": file excluded by "/home/liang/dev/drake-distro-2/drake/CPPLINT.cfg". File path component "thirdParty" matches pattern "thirdParty"
Ignoring "/home/liang/dev/drake-distro-2/drake/thirdParty/bsd/GetFullPath/GetFullPath.c": file excluded by "/home/liang/dev/drake-distro-2/drake/CPPLINT.cfg". File path component "thirdParty" matches pattern "thirdParty"
/home/liang/dev/drake-distro-2/drake/util/MexWrapper.h:15: warning: Extra space before [  [whitespace/braces] [5]
/home/liang/dev/drake-distro-2/drake/systems/plants/rigidBodyTreeMexFunctions.h:13: warning: Extra space before [  [whitespace/braces] [5]
/home/liang/dev/drake-distro-2/drake/systems/plants/constraint/constructPtrRigidBodyConstraint.h:14: warning: Extra space before [  [whitespace/braces] [5]
```

If I recall correctly, `cpplint_wrapper.py` used to be silent if there are no problems that would cause pre-merge CI to fail.

jwnimmer-tri, 2016-08-23:
#3065 changed the version of `cpplint.py` we use; did you run the superbuild (to have the `git submodule` dance be done) after pulling that PR into your workspace?

liangfok, 2016-08-23:
I ran the superbuild and the output changed a little, but there are still some extraneous warnings:

```
$ ../drake/common/test/cpplint_wrapper.py 
Checking ...............................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................
 TOTAL 577 files checked, found 3 warnings
/home/liang/dev/drake-distro-2/drake/util/MexWrapper.h:15: warning: Extra space before [  [whitespace/braces] [5]
/home/liang/dev/drake-distro-2/drake/systems/plants/rigidBodyTreeMexFunctions.h:13: warning: Extra space before [  [whitespace/braces] [5]
/home/liang/dev/drake-distro-2/drake/systems/plants/constraint/constructPtrRigidBodyConstraint.h:14: warning: Extra space before [  [whitespace/braces] [5]
```
This was on one of my feature branches. I shall next test this on a fresh checkout of master.

liangfok, 2016-08-23:
Actually, the cpplint warnings are showing up on merge-merge CI:

https://drake-cdash.csail.mit.edu/viewBuildError.php?type=1&buildid=89682

Thus, maybe they are not extraneous after-all. 

(This is still my feature branch, which may have accidentally diverged too far from master.)

liangfok, 2016-08-23:
I was unable to replicate the problem on a fresh clone of master. I will thus close this issue.

closed 2016-08-23 by liangfok

jwnimmer-tri, 2016-08-23:
When https://github.com/RobotLocomotion/drake/pull/3228/files is compared to master, it shows that the google_styleguide hash is changed.  I suspect a merge error on the feature branch is the root cause here.

liangfok, 2016-08-24:
@jwnimmer-tri: You're right, I manually switched to the correct SHA by executing the following commands:

```
$ cd drake-distro/externals/google_styleguide
$ git checkout 27929f1fd6e928ba536784e82ec67f7d474725c6
```

Once I made the above change, `cpplint_wrapper.py` no longer printed any warnings. Now I'm trying to figure out how to make this SHA change permanent, which will effectively revert the mistaken change in #3228.

liangfok, 2016-08-24:
I'm not sure what I did but after typing `git status` a few times, the change in the `google_styleguide` submodule's SHA was detected. This enabled me to commit and push the change. #3228 no longer shows the change to submodule `google_styleguide`'s SHA.