RobotLocomotion/drake #3234: drake ros build fails if MATLAB is not available

hongkai-dai, 2016-08-23:
I build the drake_catkin_workspace following the instructions on http://drake.mit.edu/from_source_ros.html#build-from-source-using-ros-catkin, since my MATLAB license does not work temporarily, I build the workspace with the command to disable MATLAB support
```
$ catkin build -DCMAKE_BUILD_TYPE:STRING=RelWithDebInfo -DDISABLE_MATLAB=TRUE
```
But catkin runs into problem, with the following error message
```
Cloning into 'swig-matlab'...
autoreconf: Leaving directory `.'
CMake Warning:
  Manually-specified variables were not used by the project:

    BUILD_SHARED_LIBS
    CMAKE_EXE_LINKER_FLAGS
    CMAKE_JAVA_COMPILE_FLAGS
    CMAKE_MODULE_LINKER_FLAGS
    CMAKE_SHARED_LINKER_FLAGS
    CMAKE_STATIC_LINKER_FLAGS
    Java_JAVAC_EXECUTABLE
    Java_JAVAH_EXECUTABLE
    Java_JAVA_EXECUTABLE
    Java_VERSION_STRING

License checkout failed.
License Manager Error -96
MATLAB is unable to connect to the license server.
Make sure that the MATLAB client machine can communicate with the hostname listed in the SERVER line
of the license file, and that the port listed in the SERVER line is not blocked by a firewall.

Troubleshoot this issue by visiting:
http://www.mathworks.com/support/lme/R2014a/96
```
Should we disable `swig_matlab` when we pass the flag `-DDISABLE_MATLAB=TRUE`?

jamiesnape, 2016-08-23:
You are missing `--cmake-args` in your command.

david-german-tri, 2016-08-23:
> Should we disable swig_matlab when we pass the flag -DDISABLE_MATLAB=TRUE?

I think the `swig_matlab` message is a red herring, as @jamiesnape suggests: your actual problem is that CMake isn't getting the `DISABLE_MATLAB=TRUE` message.

But, to answer the question: no, I don't think so.  We use `swig_matlab` for swigging everything, not just MATLAB, and `swig_matlab` doesn't actually cause the MATLAB executable to run during the build.

PR #3235 (2016-08-24): Added missing --cmake-args to ROS build instructions.
Resolves #3234.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/3235)
closed 2016-08-23 by hongkai-dai