RobotLocomotion/drake #3240: URDF Color Specifications shoudl be 4-Tuples, not 3-Tuples

liangfok, 2016-08-24:
Drake currently contains URDFs that specify colors using just three values (RGB). For example, see [this URDF](https://github.com/RobotLocomotion/drake/blob/9a67372fa00054aedf8a9f30684bdd5dc2ee9b0d/drake/examples/Quadrotor/crazyflie/crazyflie.urdf#L9).

This is in violation of the [URDF specification](http://wiki.ros.org/urdf/XML/link), which states:

> The color of a material specified by set of four numbers representing red/green/blue/alpha, each in the range of [0,1].

Currently, the URDF parser has a bug that accepts the 3-tuple as a 4-tuple. To comply with the URDF specification, We should enhance the URDF parser to throw an exception if a 3-tuple is provided for the color.

scpeters, 2016-09-29:
I will take a look at this issue.

liangfok, 2016-09-29:
@scpeters: Thanks!

scpeters, 2016-10-13:
It would be nice to have a validator program like `check_urdf` that is in [urdfdom](https://github.com/ros/urdfdom/blob/master/urdf_parser/src/check_urdf.cpp) and the `liburdfdom-tools` package on ubuntu.

scpeters, 2016-10-13:
Actually the upstream `check_urdf` doesn't complain about 3-tuple `rgba` values either.

scpeters, 2016-10-13:
It just doesn't parse any of them:

* https://github.com/ros/urdfdom_headers/blob/master/urdf_model/include/urdf_model/color.h#L87-L90

liangfok, 2017-04-12:
OK. Since upstream doesn't complain, I'll close this issue. Maybe once #4337 is resolved we can revisit to ensure consistent behavior between Drake vs. Gazebo.

closed 2017-04-12 by liangfok