RobotLocomotion/drake #3253: Move matlab code into `drake-distro/drake/matlab`

jwnimmer-tri, 2016-08-25:
As begun in #3191, we should move all matlab-related code into `drake-distro/drake/matlab`; this also includes the C++ supporting code for mex.  See https://github.com/RobotLocomotion/drake/pull/3191#issuecomment-240897342 for an overview.

The order of operations will be something like:
- Fix `util`, either:
 - stop using `util` from within non-matlab-related C++ code (ideally); move `util` into `matlab/util`; or
 - failing that, for now just split `util` in twain like we did with `solvers` (as a fallback);
- Fix `GetFullPath`, either:
 - invoke the build of `thirdParty/bsd/GetFullPath` from within `drake/matlab` (ideally); or
 - failing that, guard it with `MATLAB_FOUND` at the top level, just like `add_subdirectory(matlab)` is guarded;
- Fix `systems`  by splitting it in twain like we did with `solvers`.
- Fix `regtests` to partition the tests by either matlab-iness, specific plan TBD.
- Consider what to do with `examples` TBD.

jwnimmer-tri, 2016-08-25:
Given other priorities, and probably wanting to coordinate with other `CMakeLists` refactoring, I am not going to work on this immediately -- but don't be surprised if it crops up again in a few weeks.

jwnimmer-tri, 2016-10-17:
@RussTedrake for these four files, may I move them into `drake-distro/drake/matlab/` as part of this upcoming PR, or would it be easier to keep them where they are now (`drake-distro/drake/`)?
 1. drake/addpath_drake.m
 1. drake/rmpath_drake.m
 1. drake/Contents.m
 1. drake/runAllTests.m

jwnimmer-tri, 2016-10-17:
General FYI... work in progress will be here (N.B. will be rebased often):
https://github.com/jwnimmer-tri/drake/commits/the-great-matlab-relocation-of-aught-sixteen

RussTedrake, 2016-10-17:
Fine to move but you'll need to update the docs in a bunch of places, too.

jwnimmer-tri, 2016-10-21:
I think we can let `addpath_drake` remain where they are; it's not worth disrupting them, because they are not in the way like the library code was.  I also think that I am content to leave `examples` alone as well.  Thus, once #3861 merges, I think the issue can be closed, modulo a final sanity-check pass.

jwnimmer-tri, 2016-11-01:
Wrap-up checklist:
- [x] `find drake -name examples -prune -o -name thirdParty -prune -o -name matlab -prune -o -name doc -prune -o -name \*.m -print` is clean.
- [ ] Test with rm-rf matlab and make sure all C++ tests pass.
- [ ] Do we want to rework examples also?

jwnimmer-tri, 2016-11-01:
One nitpick issue filed as #3982, otherwise this is done.  Closing the meta-ticket.  :100:.

closed 2016-11-01 by jwnimmer-tri