RobotLocomotion/drake #3278: Carve out the non-matlab portions of `drake/util` into new homes

jwnimmer-tri, 2016-08-27:
Related to #3253, we should find new homes for the C++ forward-looking portions of `drake/util`.  This has been happening incrementally (see prior `common` and `math` related PRs, such as #2666, #2779, #2785 #3041 #3121, and cleanups like #3166 #3179 #3180 #3276 #3277).  This issue can serve as the burn-down list for finishing up.

jwnimmer-tri, 2016-08-27:
Here is my rough cut for uses of `util` that remain:
```
./systems/controllers/controlUtil.cpp:7:#include "drake/util/drakeUtil.h"
./systems/controllers/InstantaneousQPController.cpp:15:#include "drake/util/lcmUtil.h"
./systems/controllers/InstantaneousQPController.cpp:17:#include "drake/util/yaml/yamlUtil.h"
./systems/controllers/LQR.h:10:#include "drake/util/drakeGradientUtil.h"
./systems/controllers/LQR.h:11:#include "drake/util/drakeUtil.h"
./systems/controllers/QPCommon.h:11:#include "drake/util/drakeUtil.h"
./systems/controllers/simple_feedback_trajectory_controller.h:8:#include "drake/util/drakeUtil.h"
./systems/plants/constraint/RigidBodyConstraint.cpp:11:#include "drake/util/drakeGeometryUtil.h"
./systems/plants/constraint/RigidBodyConstraint.h:13:#include "drake/util/drakeUtil.h"
./systems/plants/joints/FixedAxisOneDoFJoint.h:12:#include "drake/util/drakeGradientUtil.h"
./systems/plants/joints/QuaternionFloatingJoint.h:7:#include "drake/util/drakeGeometryUtil.h"
./systems/plants/joints/RollPitchYawFloatingJoint.h:7:#include "drake/util/drakeGeometryUtil.h"
./systems/plants/RigidBody.cpp:5:#include "drake/util/drakeGeometryUtil.h"
./systems/plants/RigidBodyTree.cpp:13:#include "drake/util/drakeGeometryUtil.h"
./systems/plants/RigidBodyTree.cpp:14:#include "drake/util/drakeUtil.h"
./systems/plants/RigidBodyTree.h:26:#include "drake/util/drakeGeometryUtil.h"
./systems/plants/RigidBodyTree.h:27:#include "drake/util/drakeUtil.h"
./systems/plants/test/testKinematicsCacheChecks.cpp:11:#include "drake/util/drakeGeometryUtil.h"
./systems/robotInterfaces/QPLocomotionPlan.cpp:19:#include "drake/util/drakeGeometryUtil.h"
./systems/robotInterfaces/QPLocomotionPlan.cpp:20:#include "drake/util/drakeUtil.h"
./systems/robotInterfaces/QPLocomotionPlan.cpp:21:#include "drake/util/lcmUtil.h"
./systems/robotInterfaces/QPLocomotionPlan.cpp:22:#include "drake/util/convexHull.h"

./examples/Atlas/atlas_dynamics_demo.cc:7:#include "drake/util/drakeAppUtil.h"
./examples/Cars/car_sim_lcm.cc:8:#include "drake/util/drakeAppUtil.h"
./examples/Cars/multi_car_sim_lcm.cc:8:#include "drake/util/drakeAppUtil.h"
./examples/kuka_iiwa_arm/run_kuka_iiwa_arm_dynamics.cc:11:#include "drake/util/drakeAppUtil.h"
./examples/Pendulum/pendulum_run_dynamics.cc:11:#include "drake/util/drakeAppUtil.h"
./examples/Pendulum/pendulum_run_energy_shaping.cc:9:#include "drake/util/drakeAppUtil.h"
./examples/Pendulum/pendulum_run_lqr.cc:10:#include "drake/util/drakeAppUtil.h"
./examples/Pendulum/pendulum_run_swing_up.cc:18:#include "drake/util/drakeAppUtil.h"
./examples/Quadrotor/Quadrotor.h:9:#include "drake/util/drakeGeometryUtil.h"
./examples/Quadrotor/runDynamics.cpp:10:#include "drake/util/drakeAppUtil.h"
./examples/Quadrotor/runLQR.cpp:10:#include "drake/util/drakeAppUtil.h"
```

The systems stuff is a mixed bag; some are probably spurious, some will be one-liners, and some will required a little thought (e.g., yaml).

The examples mostly just need porting to use `gflags`.

jwnimmer-tri, 2016-09-10:
@david-german-tri could I nominate you for fixing the drakeGeometryUtil uses that are relevant to this issue?
```
./systems/plants/constraint/RigidBodyConstraint.cpp:11:#include "drake/util/drakeGeometryUtil.h"
./systems/plants/joints/FixedAxisOneDoFJoint.h:12:#include "drake/util/drakeGradientUtil.h"
./systems/plants/joints/QuaternionFloatingJoint.h:7:#include "drake/util/drakeGeometryUtil.h"
./systems/plants/joints/RollPitchYawFloatingJoint.h:7:#include "drake/util/drakeGeometryUtil.h"
./systems/plants/RigidBody.cpp:5:#include "drake/util/drakeGeometryUtil.h"
./systems/plants/RigidBodyTree.cpp:13:#include "drake/util/drakeGeometryUtil.h"
./systems/plants/RigidBodyTree.h:26:#include "drake/util/drakeGeometryUtil.h"
./systems/plants/test/testKinematicsCacheChecks.cpp:11:#include "drake/util/drakeGeometryUtil.h"
./systems/robotInterfaces/QPLocomotionPlan.cpp:19:#include "drake/util/drakeGeometryUtil.h"
```

RussTedrake, 2016-09-10:
@hongkai-dai was the author for many of these and might also be able to help.

PR #3848 (2016-10-21): Rip out drake/matlab/util from the drake/util directory
Fixes #3278.  A ~~second~~ _third_ portion of issue #3253.  A second attempt at PR #3832.
closed 2016-10-21 by jwnimmer-tri