RobotLocomotion/drake #3286: Mysterious CI Failure: Win32 - The process cannot access the file because it is being used by another process.

liangfok, 2016-08-28:
# Example Error

```
20:20:46 ERROR: Failed to clean the workspace
20:21:03 java.nio.file.FileSystemException: C:\Users\Administrator\workspace\experimental\0b4ef046\drake\build\.ninja_deps: The process cannot access the file because it is being used by another process.
```

# Example Build Instances
- https://drake-jenkins.csail.mit.edu/job/experimental/4171/compiler=msvc-ninja-32,label=windows,type=debug/console

jamiesnape, 2016-08-30:
Looks like instance `i-04bb1705` went bad. Also looks like the "do not failure the build if clean of workspace fails" checkbox in the Jenkins plugin does not actually work.

david-german-tri, 2016-10-06:
Windows is unsupported now.

closed 2016-10-06 by david-german-tri