RobotLocomotion/drake #3287: Missing drakeConfig.cmake - Documentation Needed for Applications That Want to Link Against Drake's Libraries

liangfok, 2016-08-28:
# Problem Definition

I'd like to create a software project that is built using CMake and links against Drake's exported libraries. In my top-level CMakeLists.txt, I tried to use `find_package(drake)` but encountered the following error when running `cmake`:

```
CMake Warning at CMakeLists.txt:9 (find_package):
  By not providing "Finddrake.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "drake", but
  CMake did not find one.

  Could not find a package configuration file provided by "drake" with any of
  the following names:

    drakeConfig.cmake
    drake-config.cmake

  Add the installation prefix of "drake" to CMAKE_PREFIX_PATH or set
  "drake_DIR" to a directory containing one of the above files.  If "drake"
  provides a separate development package or SDK, be sure it has been
  installed.
``` 
I searched for files in `drake-distro` called `*Config.cmake` and `*-config.cmake` and confirmed that none existed for Drake (though numerous others existed for other libraries like lcm, yaml, etc.).

# Suggested Course of Action

## Step 1: Add the Missing .cmake Files

## Step 2: Document How Users of Drake Can Configure Their Build Systems to Find and Use Drake

## Step 3: Provide a "Hello World" Implementation of a Project that Uses Drake

jamiesnape, 2016-08-30:
FYI @mwoehlke-kitware.

mwoehlke-kitware, 2016-08-30:
Yes, we definitely want to fix this, and it's on my to-do list for cleaning out all the pods junk. Unfortunately, adding proper exported packages to something the size of drake and with bunches of existing tweaks to how it creates libraries is not trivial. (But the ease of use for consumers is definitely worthwhile!)

FWIW, see e.g. LCM (the modern, CMake-ified version that we're using) for a good example of doing this right...

liangfok, 2016-09-08:
I just realized that this may be a duplicate of #2693.

david-german-tri, 2017-01-05:
From f2f: This is nearly ready.  Once it is, we should try to consider convincing OpenHumanoids to switch to it, so that we could drop pkg-config.

david-german-tri, 2017-01-05:
(However, Director also needs to build against older SHAs of Drake.)

david-german-tri, 2017-03-24:
`drake-config.cmake` is on master, though there are trouble reports like #5456.  I'm going to close this issue as partially fixed, and partially obsolete due to #3129.

closed 2017-03-24 by david-german-tri