RobotLocomotion/drake #3289: Extraneous Build Warning Regarding simulator.cc: "Type attributes ignored after type is already defined"

liangfok, 2016-08-28:
# Example of Build-time Warning

```
/home/liang/dev/drake-distro-3/drake/systems/analysis/simulator.cc:9:43: warning: type attributes ignored after type is already defined [-Wattributes]
 template class DRAKESYSTEMANALYSIS_EXPORT Simulator<double>;
                                           ^
/home/liang/dev/drake-distro-3/drake/systems/analysis/simulator.cc:10:43: warning: type attributes ignored after type is already defined [-Wattributes]
 template class DRAKESYSTEMANALYSIS_EXPORT Simulator<AutoDiffXd>;
```

# Initial Prognosis

I suspect this warning appeared after #3202 was merged. Specifically: https://github.com/RobotLocomotion/drake/pull/3202/files#diff-47396c61ef559c4753189635c419c199.

sherm1, 2016-08-30:
See issue #3305. There is a problem on Windows also.

jwnimmer-tri, 2016-12-22:
Pretty sure this is fixed now...

sherm1, 2016-12-22:
Yup.

closed 2016-12-22 by sherm1