RobotLocomotion/drake #3294: Standardize on `length` instead of `size` in systems/framework

david-german-tri, 2016-08-29:


sherm1, 2016-08-29:
I think this should be `size()` for reasons just mentioned on Slack. They are:
- length is a dimension in physical simulation (for us measured in meters)
- size is used for C++ containers and is the term Eigen uses for number of elements in a vector

I would argue that our adoption of Eigen means we have already chosen a convention for this, and it is `size()`.

sammy-tri, 2016-08-29:
I missed the voting on slack, and agree with both of @sherm1's points.

liangfok, 2016-08-29:
Sounds like a re-vote is in order. Maybe instead of just a simple binary :+1: and :-1: , we should make it "by approval" and/or "everybody has one hundred tokens; place them in your preferred bin" type voting.

For example, since I'm voting "present", I shall put 50 tokens in :+1: and 50 tokens in :-1: .

sherm1, 2016-08-29:
/cc @mitiguy

jwnimmer-tri, 2016-09-22:
Any new consensus here?  I'm about to write more code that will be littered with either one or the other.

david-german-tri, 2016-09-22:
> Any new consensus here?

Not to my knowledge; team sentiment was quite divided.  Personally, I have no interest in this bikeshed, nor does the inconsistency particularly bother me.  If it doesn't bother you either, feel free to perpetuate it!

If it does bother you, maybe PR the fix you prefer?  I'll LGTM it, and anyone who objects can either get over it or escalate to Russ.  (That goes for anyone else as well!)

jwnimmer-tri, 2016-09-23:
Alright.  I will work on a PR for `size` soon(ish).

PR #3574 (2016-09-24): Use 'size' when referring to vector dimension
Resolves #3294.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/3574)
closed 2016-09-24 by david-german-tri