RobotLocomotion/drake #3302: Unable to open dreal/external/src/CAPD4/capdMake/INSTALL

jwnimmer-tri, 2016-08-30:
Since #3221, several of Drake's demos and unit tests display a warning like this any time they run:
`Unable to open directory: /home/jwnimmer/jwnimmer-tri/drake-distro/build/externals/dreal/external/src/CAPD4/capdMake/INSTALL`.

It would be nice to suppress this.  (I assume it is unimportant?)

soonho-tri, 2016-08-30:
I see. I'll take care of this one.

soonho-tri, 2016-09-08:
@jwnimmer-tri, could you give me a simple repro step for this one? I ran `ctest -VV` at `drake-distro/build/drake` directory, but I couldn't get the message from the run.

jwnimmer-tri, 2016-09-08:
Sure, no problem.  On master:
```
$ drake-distro/build/debug/drake/bin$ ./lidarTest 
Running main() from gtest_main.cc
[==========] Running 1 test from 1 test case.
[----------] Global test environment set-up.
[----------] 1 test from LidarTest
[ RUN      ] LidarTest.BasicTest
Unable to open directory: /home/jwnimmer/jwnimmer-tri/drake-distro/build/debug/externals/dreal/external/src/CAPD4/capdMake/INSTALL
Welding joint base
Welding joint base
Welding joint base
Welding joint base
[       OK ] LidarTest.BasicTest (434 ms)
[----------] 1 test from LidarTest (434 ms total)

[----------] Global test environment tear-down
[==========] 1 test from 1 test case ran. (434 ms total)
[  PASSED  ] 1 test.
```

soonho-tri, 2016-09-08:
@jwnimmer-tri, what outputs do you have for the following commands?

 - `ls -l /home/jwnimmer/jwnimmer-tri/drake-distro/build/debug/externals/dreal/external/src/CAPD4/capdMake`
 - `ls -l /home/jwnimmer/jwnimmer-tri/drake-distro/build/debug/externals/dreal/external/src/CAPD4`

BTW, I think this warning message comes from https://github.com/RobotLocomotion/drake/blob/master/drake/systems/plants/xmlUtil.cpp#L241-L243 .

jwnimmer-tri, 2016-09-08:
```
jwnimmer@call-cc:~/jwnimmer-tri/drake-distro$ ls -l /home/jwnimmer/jwnimmer-tri/drake-distro/build/debug/externals/dreal/external/src/CAPD4/capdMake
total 624
-rw-rw-r-- 1 jwnimmer jwnimmer      0 Sep  7 15:45 AUTHORS
-rw-rw-r-- 1 jwnimmer jwnimmer     96 Sep  7 15:45 capd_library_version_number.m4
-rw-rw-r-- 1 jwnimmer jwnimmer     88 Sep  7 15:45 capd_version_number.m4
-rwxr-xr-x 1 jwnimmer jwnimmer   7333 Sep  7 15:46 compile
-rwxr-xr-x 1 jwnimmer jwnimmer  43499 Sep  7 15:46 config.guess
-rw-rw-r-- 1 jwnimmer jwnimmer      0 Sep  7 15:45 config.rpath
-rwxr-xr-x 1 jwnimmer jwnimmer  36144 Sep  7 15:46 config.sub
lrwxrwxrwx 1 jwnimmer jwnimmer     10 Sep  7 15:45 COPYING -> ../COPYING
drwxrwxr-x 3 jwnimmer jwnimmer   4096 Sep  7 15:45 debian
-rwxr-xr-x 1 jwnimmer jwnimmer  23566 Sep  7 15:46 depcomp
drwxrwxr-x 3 jwnimmer jwnimmer   4096 Sep  7 15:45 docs
drwxrwxr-x 3 jwnimmer jwnimmer   4096 Sep  7 15:45 examples
lrwxrwxrwx 1 jwnimmer jwnimmer     10 Sep  7 15:45 INSTALL -> ../INSTALL
-rwxr-xr-x 1 jwnimmer jwnimmer  15155 Sep  7 15:46 install-sh
drwxrwxr-x 5 jwnimmer jwnimmer   4096 Sep  7 15:47 libcapd
-rw-r--r-- 1 jwnimmer jwnimmer 324404 Sep  7 15:46 ltmain.sh
drwxrwxr-x 2 jwnimmer jwnimmer   4096 Sep  7 15:46 m4
drwxrwxr-x 2 jwnimmer jwnimmer   4096 Sep  7 15:45 make
-rwxr-xr-x 1 jwnimmer jwnimmer   6872 Sep  7 15:46 missing
-rw-rw-r-- 1 jwnimmer jwnimmer      0 Sep  7 15:45 NEWS
drwxrwxr-x 2 jwnimmer jwnimmer   4096 Sep  7 15:45 osx
lrwxrwxrwx 1 jwnimmer jwnimmer      9 Sep  7 15:45 README -> ../README
drwxrwxr-x 2 jwnimmer jwnimmer   4096 Sep  7 15:45 release_changelog
-rwxr-xr-x 1 jwnimmer jwnimmer   4640 Sep  7 15:46 test-driver
drwxrwxr-x 3 jwnimmer jwnimmer   4096 Sep  7 15:45 utils
jwnimmer@call-cc:~/jwnimmer-tri/drake-distro$ ls -l /home/jwnimmer/jwnimmer-tri/drake-distro/build/debug/externals/dreal/external/src/CAPD4
total 524
-rw-rw-r--  1 jwnimmer jwnimmer  41151 Sep  7 15:45 aclocal.m4
drwxr-xr-x  2 jwnimmer jwnimmer   4096 Sep  7 15:46 autom4te.cache
drwxrwxr-x  2 jwnimmer jwnimmer   4096 Sep  7 15:47 bin
drwxrwxr-x  9 jwnimmer jwnimmer   4096 Sep  7 15:46 capdAlg
drwxrwxr-x  7 jwnimmer jwnimmer   4096 Sep  7 15:46 capdAux
-rw-rw-r--  1 jwnimmer jwnimmer  21862 Sep  7 15:47 capd.cache
drwxrwxr-x  8 jwnimmer jwnimmer   4096 Sep  7 15:47 capdDynSys4
drwxrwxr-x  6 jwnimmer jwnimmer   4096 Sep  7 15:46 capdExt
drwxrwxr-x 11 jwnimmer jwnimmer   4096 Sep  7 15:46 capdMake
-rw-rw-r--  1 jwnimmer jwnimmer    146 Sep  7 15:45 ChangeLog
-rw-rw-r--  1 jwnimmer jwnimmer  17264 Sep  7 15:47 config.log
-rwxrwxr-x  1 jwnimmer jwnimmer  31799 Sep  7 15:46 config.status
-rwxrwxr-x  1 jwnimmer jwnimmer 169008 Sep  7 15:46 configure
-rw-rw-r--  1 jwnimmer jwnimmer   2167 Sep  7 15:45 configure.ac
-rw-rw-r--  1 jwnimmer jwnimmer  35147 Sep  7 15:45 COPYING
drwxrwxr-x  4 jwnimmer jwnimmer   4096 Sep  7 15:46 lib
-rw-rw-r--  1 jwnimmer jwnimmer  28359 Sep  7 15:46 Makefile
-rw-rw-r--  1 jwnimmer jwnimmer   1911 Sep  7 15:45 Makefile.am
-rw-rw-r--  1 jwnimmer jwnimmer  26887 Sep  7 15:46 Makefile.in
lrwxrwxrwx  1 jwnimmer jwnimmer      9 Sep  7 15:45 README -> README.md
-rw-rw-r--  1 jwnimmer jwnimmer      0 Sep  7 15:45 README.md
```

jwnimmer-tri, 2016-09-08:
Wow.  So the XML hunter is searching through _all the things_ due to https://github.com/RobotLocomotion/drake/blob/master/drake/systems/plants/xmlUtil.cpp#L308-L310.

soonho-tri, 2016-09-08:
OK. I think there is a symbolic link from `externals/dreal/external/src/CAPD4/capdMake/INSTALL` to `externals/dreal/external/src/CAPD4/INSTALL`. However somehow the target `externals/dreal/external/src/CAPD4/INSTALL` does not exist.

soonho-tri, 2016-09-08:
Can you try to delete dreal under drake and rebuild it? I guess this solves this problem.

soonho-tri, 2016-09-08:
> Wow. So the XML hunter is searching through all the things

Yes. I think this can be improved by narrowing down the search scope.

soonho-tri, 2016-09-08:
This issue could be caused by the change that I made in dReal. We used to build an external project (i.e. CAPD) using `BUILD_IN_SOURCE 1`. Recently, I turned that off. I guess there could be some remaining files/directories/broken-symlinks in the source directory and this XML hunter picks it up.

jwnimmer-tri, 2016-09-08:
Ah; perhaps I had `Ctrl-C`'d the superbuild at some inopportune moment, and it doesn't recover incrementally?  In any case, I blew away these directories, re-ran the superbuild, and I am still seeing the message (and the symlink is still broken):
```
externals/dreal
build/debug/install/lib/dreal
build/debug/install/include/dreal
build/debug/externals/dreal
```

soonho-tri, 2016-09-08:
Could you make a tar.gz of `/home/jwnimmer/jwnimmer-tri/drake-distro/build/debug/externals/dreal/external/src/CAPD4-stamp/` directory and send it to me? This directory should include enough information for the debugging.

soonho-tri, 2016-09-08:
I think I fixed the problem on dReal side. Could you try to rebuild dReal again by removing those directories that you mentioned above? If the problem persists, please let me know.

jwnimmer-tri, 2016-09-08:
Works now (no error message) -- and the symlink is good now (to an empty file in the parent directory).
But how did you fix it without a PR?  Is some part of the build process not tied to a specific sha?

closed 2016-09-08 by jwnimmer-tri

soonho-tri, 2016-09-08:
> But how did you fix it without a PR? Is some part of the build process not tied to a specific sha?

We're maintaining our own separate repositories for the external packages of dReal. I fixed the problem there. From drake's point of view, there are two levels of indirection.

jwnimmer-tri, 2016-09-08:
And the dReal build uses master of those externals, instead of a sha?  So builds are not reproducible?

soonho-tri, 2016-09-08:
> And the dReal build uses master of those externals, instead of a sha? So builds are not reproducible?

Yes, that's the case for now. I'll change it soon.