RobotLocomotion/drake #3305: Need more nuanced EXPORT macros for template instantiations

sherm1, 2016-08-30:
Currently CMake is providing us with export macros for indicating which symbols in a shared library should be visible. These are set to different compiler directives for gcc/clang and MSVC, but placed identically. That works for ordinary symbols but not for library-instantiated template classes. For those there is code required in two places:
```c++
// A declaration in the header file:
extern template class EXPORT_A Simulator<double>;
// The instantiation in the .cc file:
template class EXPORT_B Simulator<double>;
```
To avoid warnings, `EXPORT_A` should apparently be `__attribute__((visibility("default")))` for gcc/clang and empty for MSVC. `EXPORT_B` should be `__declspec(dllexport)` for MSVC and empty for gcc/clang. 

The problem currently is that there is only a single "EXPORT" macro available so it puts either `__attribute__((visibility("default")))` or `__declspec(dllexport)` in both positions, causing warnings and possibly misbehavior on all platforms. See #3289 for a report of the ".cc" file warning on Linux. See [here](https://github.com/RobotLocomotion/drake/blob/master/drake/systems/analysis/simulator.h#L298) for the crude workaround currently in Drake for MSVC.

Please consider adding a pair of new export macros that we can use in this circumstance. I don't think the existing macro can be used in either place. We're expecting to use this a lot in the near future and would rather not see the hacks proliferate.

jamiesnape, 2016-08-30:
FYI @mwoehlke-kitware.

mwoehlke-kitware, 2016-08-30:
FYI @bradking we should probably address this in `GenerateExportHeader.cmake`. I'm thinking `<name>_EXPORT_TEMPLATE` (A) and `<name>_EXPORT_INSTANTIATE` (B); does that work for everyone?

bradking, 2016-08-30:
> `EXPORT_B` should be `__declspec(dllexport)` for MSVC and empty for gcc/clang. 

The dllexport may be needed on MinGW too.  Also one will need to investigate whether default visibility is needed on the instantiation when compiling with `-fvisibility=hidden`.

Ideally the `EXPORT_B` case will just be able to use the normal export macro instead of needing a dedicated one.  It should only appear in a `.cc` file and so will be compiled with the macro set to the exporting side's value.

sherm1, 2016-08-30:
> I'm thinking <name>_EXPORT_TEMPLATE (A) and <name>_EXPORT_INSTANTIATE (B);

I'd like to see "extern" mentioned in (A) since it should only appear in "extern template" declarations, maybe `<name>_EXPORT_EXTERN_TEMPLATE` ?

sherm1, 2016-08-30:
> Ideally the EXPORT_B case will just be able to use the normal export macro 

I think that's what is leading to the Linux warnings reported in #3289. That one has to be empty on Linux but would still have the gcc/clang directive.

jwnimmer-tri, 2016-08-30:
:100: to @bradking's comment.  First figure out the correct compiler inputs for all of the declarations (citations to the compilers' documentation, and/or platform ABI details appreciated), and _then_ design the macros to suit.  I don't think the PR spike test that led to this ticket is sufficient guidance for how this should all end up.

mwoehlke-kitware, 2016-08-31:
> To avoid warnings, `EXPORT_A` should apparently be `__attribute__((visibility("default")))` for gcc/clang and empty for MSVC.

I confirm this via empirical testing. See also https://msdn.microsoft.com/en-us/library/bb531392.aspx, which says the same thing.

That said, I should point out that disabling C4910 could be another option...

> `EXPORT_B` should be `__declspec(dllexport)` for MSVC and empty for gcc/clang.

I have *not* been able to confirm this. Can someone please provide a [SSCCE](http://sscce.org/) and note what compiler(s) show the issue? Bonus points for using [this CMake test](https://gitlab.kitware.com/matthew-woehlke/cmake/tree/geh-template-symbols/Tests/GenerateExportHeader) as a starting point.

mwoehlke-kitware, 2016-08-31:
>> `EXPORT_B` should be `__declspec(dllexport)` for MSVC and empty for gcc/clang.
>
> I have *not* been able to confirm this.

Um... never mind. Confirmed with GCC 4.9.2. Maybe I was only trying with clang, and misunderstood which compiler was the issue...

david-german-tri, 2016-10-06:
Closing, since we are only going to be removing export goop now that Windows is no longer supported.

closed 2016-10-06 by david-german-tri