RobotLocomotion/drake #3308: Determine a scope policy for accessors and mutators on Systems

david-german-tri, 2016-08-30:
Do we always pass very general pointers (e.g. `Context`), or do we only expose the inputs and outputs a particular function requires (e.g. `ContinuousState`)?

david-german-tri, 2016-08-30:
From discussion on #3275:

Pros for always passing `Context`: consistent API, beginner-friendly, ensures Cache is always present.

Pros for passing subordinate pointers: Implementation doesn't have to dig out the sub-pointers it cares about, implementation can't touch bits that it shouldn't.

sherm1, 2016-08-30:
I liked the idea of carefully restricting what is mutable, both for safety and to lighten the load on the method implementer who otherwise has to hunt down and extract the right piece of context. But @amcastro-tri pointed out that this shifts the hunt & extract burden onto the method *caller* who is even less likely to be qualified to do that, and the extraction code would now appear at every call rather than just once in the implementation.

A few other arguments in favor of the "whole mutable context" API:
- we really only pass the local subcontext which is defined by the subsytem whose method we are invoking
- The APIs are simpler and more uniform: const Context to extract info, mutable Context for state changes; no further Context cruft in the API (and generally the less said about Context the better)
- At least in Debug builds we can add assertions to verify that only legitimate changes were made (because we always note mutable access as a change)
- Simbody has done it this way for a decade with no apparent adverse effects

david-german-tri, 2016-12-13:
I think the emerging consensus across a bunch of examples in `systems/framework` and `systems/analysis` is that users of Systems should generally only have to pass the Context, _and_ implementers of Systems should implement methods where the only mutable arguments are the things that are allowed to change.   NVI is useful for achieving this.

closed 2016-12-13 by david-german-tri