RobotLocomotion/drake #3310: TODO: Create a Stand-In for Director's IK Trajectory Planner

liangfok, 2016-08-30:
# Problem Definition

Few of us are able to compile not to mention _run_ the custom version of Director that we're using to generate the joint trajectories for the IIWA robot. Thus, to facilitate development, it would be helpful to have an application that can be a lightweight stand-in for it.

# Action Item

Create a simple application that waits for a key stroke. When the user hits a key, the application should:

1. Create an IK trajectory by calling [GenerateIKDemoTrajectory()](https://github.com/naveenoid/drake/blob/dev/kukaIKControlDemo/drake/examples/kuka_iiwa_arm/iiwa_simulation.h#L83) to obtain a sample IK trajectory.

2. Save the trajectory in a [robot_plan_t](https://github.com/sammy-tri/drake/blob/director_ik/drake/lcmtypes/lcmt_robot_plan_t.lcm) LCM message. Specifically, in the "plan" member variable within that LCM message, which is an array of [bot_core.robot_state_t](https://github.com/mwoehlke-kitware/bot_core_lcmtypes/blob/0434090f6f54228c35514f7c9216e3bf1e003ea3/lcmtypes/bot_core_robot_state_t.lcm) objects.

3. Publish the message on LCM channel "COMMITTED_ROBOT_PLAN".

liangfok, 2016-08-31:
FTR: WIP is here: https://github.com/liangfok/drake/blob/feature/director_kuka_ik_standin/drake/examples/kuka_iiwa_arm/kuka_director_ik_trajectory_publisher_standin.cc

liangfok, 2016-08-31:
# Demo compilation

```
$ cd drake-distro
$ mkdir build
$ cd build
$ cmake .. -DCMAKE_BUILD_TYPE:STRING=RelWithDebInfo -DDISABLE_MATLAB=TRUE -DWITH_DREAL=FALSE -DWITH_SNOPT=FALSE
$ make -j8
```

# Demo Execution

To run the demo, open three terminals:

## Terminal 1 - Drake Visualizer

```
$ cd drake-distro/build
$ ./install/bin/drake-visualizer
```

## Terminal 2 - Robot Simulation

```
$ cd drake-distro/build
$ ./drake/bin/kuka_plan_runner
```

## Terminal 3 - Director IK Planner Stand In

```
$ cd drake-distro/build
$ 
```

sammy-tri, 2018-09-21:
Let's call this fixed by #7728.

closed 2018-09-21 by sammy-tri