RobotLocomotion/drake #3312: Mysterious CI Failure: "Can't unlink already-existing object"

liangfok, 2016-08-30:
# Example Error

```
15:55:03 Extracting Ninja for Windows...
15:55:03 x ninja.exe
15:55:03 CMake Error: Problem with archive_write_header(): Can't unlink already-existing object
15:55:03 CMake Error: Current file: ninja.exe
15:55:03 CMake Error: Problem extracting tar: C:/Users/Administrator/workspace/experimental/0b4ef046/ninja-win.zip
15:55:03 
15:55:03 CMake Error at ci/ctest_driver_script.cmake:208 (message):
15:55:03 *** CTest Result: FAILURE BECAUSE EXTRACTING NINJA FOR WINDOWS WAS NOT
15:55:03 SUCCESSFUL
```

# Example Log Files

* https://drake-jenkins.csail.mit.edu/job/experimental/compiler=msvc-ninja-32,label=windows,type=debug/4248/

Lucy-tri, 2016-08-30:
The next build passed. Probably a spurious failure.

jamiesnape, 2016-08-30:
Same issue as #3286.

liangfok, 2016-08-30:
OK thanks.

closed 2016-08-30 by liangfok