RobotLocomotion/drake #3314: Must System::EvalOutput() Be Pure Virtual?

liangfok, 2016-08-31:
`BotVisualizerSystem`, and `LcmPublisherSystem` are leave systems with zero outputs. Despite to having any outputs, they must still provide an `EvalOutput()` method since this method is a pure virtual method in ancestor class `drake::systems::System`. Must this be the case?

jwnimmer-tri, 2016-08-31:
Having publish-only systems seems like a special case.  They alone don't seem to me like a reason to change the base framework class.  We can make a PublishOnlySystem base for them if we want to save keystrokes.

liangfok, 2016-08-31:
:+1: for making a `PublishOnlySystem` that is a child of `LeafSystem`.

liangfok, 2016-08-31:
I'm going to wait for @david-german-tri to chime in before submitting a PR that adds a `PublishOnlySystem` to the System 2.0 framework.

sherm1, 2016-08-31:
:-1: for PublishOnlySystem. It is easy to tell if a System is publish-only since it will have 0 output ports. Let's just make `EvalOutputPort()` optional with a default implementation that throws an error if someone calls it. I don't think that calls for yet another class.

david-german-tri, 2016-08-31:
> Let's just make EvalOutputPort() optional with a default implementation that throws an error if someone calls it.

I don't want to give up the compile-time reminder that children of `System<T>` must implement `EvalOutputPort`.

> I don't think that calls for yet another class.

I don't understand why another class is a bad thing.  It's small, it's optional, and the name is self-documenting.

sherm1, 2016-08-31:
> I don't understand why another class is a bad thing. 

I don't think it solves a real problem and I think this class picks an essentially random axis about which to divide Systems. Next we will need a PublishOnlyDiagram. Last time we rejected this route after it looked like we would get ContinuousAndDiscreteNonPhysicalSystems, etc. Why is it a good idea to represent the presence or absence of particular System properties as classes? How does this scale when we add a new method to System? There are already plenty of non-pure virtuals which are mandatory for Systems that turn out to have certain properties.

liangfok, 2016-08-31:
:+1: for providing a default implementation of `System::EvalOutput()` that simply contains the following line of code:

```
DRAKE_ABORT_MSG("This system has no outputs meaning it's a mistake to call EvalOutput()")
```

david-german-tri, 2016-08-31:
> Next we will need a PublishOnlyDiagram.

I agree that would be wrong.  Diagrams can, do, and should always implement the entire `System<T>` interface in framework code, not in user code.

> Last time we rejected this route after it looked like we would get ContinuousAndDiscreteNonPhysicalSystems, etc.

That is not analogous.  We rejected multiple interface classes for different System attributes.  Those interface classes formed a type tree with many roots, which required multiple inheritance, which led to a variety of problems, none of which apply to the proposed PublishOnlyDiagram.  It's just a branch on the type tree that saves a little bit of typing.  

I admit I don't entirely see the point in PublishOnlyDiagram, but if someone else values the convenience enough to write it, I think it should be allowed on master.  It's not going to break anything.

david-german-tri, 2016-08-31:
```
DRAKE_ABORT_MSG("This system has no outputs meaning it's a mistake to call EvalOutput()")
```
This, on the other hand, I do not believe we should allow on master.  It will be confusing if your System does, in fact, have outputs and you just didn't remember to implement `EvalOutput`.  Worse, it defers the discovery of your mistake to run time.

@liangfok, it looks like the answer to your original question may be: Yes, it is necessary.  It's a pretty small and self-explanatory burden.

liangfok, 2016-08-31:
OK. Closing the issue. Thanks!

closed 2016-08-31 by liangfok