RobotLocomotion/drake #3316: Create a rigid body plant translator to poses.

amcastro-tri, 2016-08-31:
`BotVisualizer` needs poses and bodies' names to send over LCM. RBP outputs a state vector which doesn't directly translate into useful physical body poses. 
There are other cases in which systems equations are directly coded without using an RBT (see simple pendulum example) but however a `BotVisualizer` is used to visualize it as a three dimensional rigid body system. For those cases a translator from state to body poses could be also used.

Therefore it seems to make sense to have a separate System that takes a state vector and outputs body poses. Uses could include not only communicating with `BotVisualizer` as stated in these examples but could also be writing to a file or even using three state space poses for control.

sherm1, 2016-08-31:
@amcastro-tri, wouldn't that System have to have a kinematic model in it to map states->poses? If so it would have the same problem as the current BotVisualizer in that it duplicates the RB Tree.

liangfok, 2016-09-01:
Yes, the proposed system will a reference to the `RigidBodyTree`. However it's still beneficial since the 6-DOF computations can be computed once per simulation cycle and shared by multiple downstream systems. Two systems that I know will use such information include `BotVisualizerSystem` and `RosTfTransformPublisherSystem`. I can also imagine numerous `ImuSensorSystem` objects that measure the pose of each rigid body consuming the pose data. Without a dedicated system that computes the body poses, the calculations will have to be done in each individual system that requires the poses.

Since the `RigidBodyPlant` needs to compute the 6-DOF poses of each body anyway (I think), it would be most efficient to include the computation in it. However, not all simulations use the `RigidBodyPlant` meaning the proposed system will still be useful in those situations.

liangfok, 2016-09-01:
Once `Context` supports parameters, it'll probably make sense to make the names of the rigid bodies parameters rather than outputs since they will rarely change. I assume doing this will increase computational efficiency at the cost of less flexibility w.r.t. changing the `RigidBodyTree` model during run-time.

sherm1, 2016-09-01:
> Yes, the proposed system will need a reference to the RigidBodyTree. However it's still beneficial since the 6-DOF computations can be computed once per simulation cycle and shared by multiple downstream systems. 

I can't see any advantage to separating a "PoseSystem" out from RigidBodyPlant which could just as easily output the poses on an output port, which can of course be shared downstream. And with the caching mechanism in place they will also be updated only when changed. No code would be duplicated, and the RBTree would not have to leak out of RBPlant. 

PoseSystem would *not* be useful for simulations that don't use RBPlant -- instead the custom equations in such a system would have to be duplicated in the PoseSystem. That seems far worse than adding a pose output port to that system, which very likely has to compute the poses anyway.

Am I misunderstanding the proposal?

sherm1, 2016-09-01:
> Once Context supports parameters, it'll probably make sense to make the names of the rigid bodies parameters rather than outputs since they will rarely change. I assume doing this will increase computational efficiency ...

No, not at all! It is perfectly fine to have a constant output port (in fact "constant" is a standard "sample rate" in Simulink and should be for us as well). In any case our caching mechanism will ensure that nothing gets computed unless it is out of date. The names will never be out of date once set.

Parameters should be reserved for values you might want to change from one simulation to the next. It seems unlikely that body names would be in that category, but you could make them parameters if appropriate. But don't do it for performance reasons.

david-german-tri, 2016-09-01:
FYI, I agree with everything @sherm1 said above.

> Should the pose and body name information be stored in an output port or can they simply saved in the Context?

Stored in an output port - which will eventually be backed by a Cache entry, once that functionality exists.

liangfok, 2016-09-01:
Can you provide more details? Why output port instead of context? Thanks!

amcastro-tri, 2016-09-01:
I just realize that my proposal does not make sense.
`BotVisualizer` needs to be able to send `drake-visualizer` all the information of what is contained in the world (body names, geometry types and their parameters, color, offsets). All of this currently happens in 
`BotVisualizerSystem::initialize_drake_visualizer()` which of course need a reference to the `RigidBodyTree`.

Therefore I think the best option is to have `BotVisualizer` to receive RBP states and emit LCM messages. To do this efficiently the RBP will need to output its KinematicsCache as an abstract value.

For systems like a simple pendulum dynamics but that we want to render as a three dimensional RBP, we will just leave the `KinematicsCache` port unconnected case in which the `BotVisualizer` will just instantiate its own `KinematicsCache` as it is done right now.

I am closing this however because this solution is not a good one. I opened #3325 to continue discussion there.

closed 2016-09-01 by amcastro-tri