RobotLocomotion/drake #3319: Can Diagram::GetMutableSubSystemContext() abort instead of returning nullptr?

liangfok, 2016-08-31:
While debugging #3318, @naveenoid and I encountered a situation where [Diagram::GetMutableSubSystemContext()](https://github.com/RobotLocomotion/drake/blob/cf4d7d2d6b69dbf05c0619c0de13506c4cd778af/drake/systems/framework/diagram.h#L221) was returning `nullptr`. This naturally resulted in a segmentation fault in the application's code. It would be helpful if the method could abort rather than return `nullptr`, so the problem can be detected earlier.

liangfok, 2016-09-01:
Another way this issue can be resolved is to update the method's documentation to specify under what conditions `nullptr` may be returned.

david-german-tri, 2016-09-01:
Yes, documenting it is a good idea.  I'll do that.

PR #3348 (2016-09-06): Use std::vector instead of std::map to store DiagramContext constituents.
Clarify the cases in which subcontext/suboutput retrieval aborts, and the cases in which it returns nullptr.  Resolves #3319.

+@liangfok for feature review

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/3348)
closed 2016-09-06 by david-german-tri