RobotLocomotion/drake #3320: Untangle multibody tree terminology

sherm1, 2016-08-31:
(Resurrecting this discussion from #1714 since it has cropped up recently.) This is an FYI for people who have to deal with RigidBodyTree internals: the terms "joint" and "parent/child" are meaningful to users but are technically imprecise. We need more precise terms internally.

Joint
-----
Roboticists typically understand "joint" as a connection between two bodies that parameterizes relative motion of the two bodies. However, there are at *least* three possible mathematical models of a joint. These distinctions don't matter to many end users but are paramount for the implementation. Here are some terms we can use internally:
 1. "**Mobilizer**": ODEs governing joint coordinates -- representing permitted motion (DOFs) across the joint.  
 2. "**Constraint**": a set of algebraic constraints restricting the motion of a body (a.k.a. "loop joint") 
 3. "**Bushing**": a force element allowing limited compliance (a.k.a. "penalty method").
 
Parent/child
-------------
A general multibody system consists of a set of bodies and a set of joints. Each joint connects a pair of bodies. In `.urdf` or `.sdf`, the user designates one of the bodies as the "parent" and the other as the "child" which defines the sign and order of the joint coordinates. Reversing parent and child simply reverses the joint coordinates. 

If you draw a graph with bodies as nodes and joints as edges, it does not form a tree; it is a general directed graph, with edges directed from child to parent. This description corresponds exactly, for example, to Gazebo's `.sdf` file format. Many multibody codes model the system in directed graph form. Gazebo typically uses ODE or Bullet physics engines which give every body 6 degrees of freedom with respect to the world, and model every joint as a set of algebraic constraints. On the other hand, Drake, Simbody, and SD/FAST use internal coordinates which require a (non-unique) spanning tree for the directed graph.

When forming the spanning tree, "world" is the root, each body will appear as a node, and every edge of the spanning tree will correspond to a joint (possibly with some joints left over). In a tree there is an inherent direction to an edge: you can think of each edge as pointing *towards* the root. Let's call the direction towards the root the "inboard" direction and the direction away from the root the "outboard" direction. Here is the ambiguity: in general it is not possible for the inboard direction always to correspond to the user's parent body! This is why internal coordinate multibody codes provide "reverse" joints.

So the unique joint that connects a body into the Drake `RigidBodyTree` (which uses the "mobilizer" model above) shouldn't be referred to as the "parent joint" (because the user's "child" body turned out to be closer to world). Instead, call it the body's "inboard joint" or body's "mobilizer" (my preferred terminology). Then it is possible to speak independently of the tree topology and the (parent,child) sign convention that a user establishes for a joint.

We would like all us internals people to recognize that these ambiguities exist, and to migrate to more precise terminology internally.

Paul and Sherm

/cc @mitiguy, @amcastro-tri, @liangfok, @tkoolen, @RussTedrake, @patmarion

liangfok, 2016-08-31:
:+1: Thanks for proposing this clarifying terminology. I recommend that it be posted as official documentation on [Drake's website](http://drake.mit.edu/). Perhaps it should go in the [design notes](http://drake.mit.edu/design.html)? Or maybe the [style guide](http://drake.mit.edu/developers.html#programming-style-notes)?

**Question:** The second paragraph of the "Parent/child" subsection mentions that some simulators require a spanning tree while others do not. What are the pros / cons of each? Is it generality vs. computational efficiency?

liangfok, 2016-08-31:
Can we also adopt the following terms?

* **generalized state vector** - The concatenation of a generalized coordinate vector with a generalized velocity vector.
* **generalized coordinate vector** - A vector of generalized coordinates.
* **generalized velocity vector** - A vector of generalized velocities.
* **generalized coordinate** - A distance (m) or angle (radian) measurement.
* **generalized velocity** - A speed (m / s) or angle (radians / second) measurement.

I would also be OK with replacing the word "coordinate" with "position". Thoughts?

sherm1, 2016-08-31:
> What are the pros / cons of [maximal vs. minimal coordinate multibody systems]?

That is a very long conversation, @liangfok! Both can work fine but have different strengths and weaknesses. Definitely not a GitHub-level chat.

sherm1, 2016-08-31:
@liangfok, in this issue let's see if we can get the joints vs. mobilizers and parent/child vs. inboard/outboard terms settled first. Your other proposed definitions are not precise -- e.g. the term "generalized" means exactly what it says -- there is no implication that the coordinates are lengths or angles; those are just common special cases.

liangfok, 2016-08-31:
Oh, are you saying that the term "mobilizer joint" is redundant that we should standardize on just "mobilizer". That is OK with me. Fewer key strokes.

sherm1, 2016-08-31:
> are you saying the term "mobilizer joint" is redundant?

Yes, it's not wrong since a mobilizer is one of the implementations for a joint. But I prefer just "mobilizer" since it is short and unambiguous. [Here](https://simtk.org/docman/view.php/47/1536/Seth-2010-ShermanEastmanDelp-MinimalJointFormulationForBiomechanisms-NonlinearDyn-v62-p291.pdf) is a paper that precisely defines it.

liangfok, 2016-09-01:
Thanks.  I will standardize on using the term "mobilizer".

RussTedrake, 2016-09-01:
I've never seen mobilizer before.  Is it standard?

sherm1, 2016-09-01:
> I've never seen mobilizer before. Is it standard?

No. It was coined by Ajay Seth and me at Stanford in 2010 in [this paper](https://simtk.org/docman/view.php/47/1536/Seth-2010-ShermanEastmanDelp-MinimalJointFormulationForBiomechanisms-NonlinearDyn-v62-p291.pdf) in desperation trying to deal with biomechanics users who have their own ideas about joints. The closest thing in the multibody literature is "hinge" which is inadequate in any context where it might have its conventional meaning of "revolute joint" (like a door hinge). We used the "mobilizer" term successfully for years to avoid usurping the terms "joints", "parent", and "child" which our users felt they already understood. We used it also while working with OSRF to add an internal coordinate multibody engine to Gazebo -- they also had an understanding of joint/parent/child (as seen in urdf and sdf) that did not generalize cleanly to trees.

Because tree joints *permit* motion (as opposed to constraint joints that *restrict* it), "mobilizer" is a very nice term for use as "the thing that grants mobility to a body". A tree joint's dofs can be referred to as it mobilities, and the terms hang together smoothly and unambiguously. 

I should add that it is always better to use joint/parent/child if you can! Mobilizer should be reserved as a term of art for people who need to know how a model is being implemented internally.

liangfok, 2016-09-01:
Should [DrakeJoint](https://github.com/RobotLocomotion/drake/blob/master/drake/systems/plants/joints/DrakeJoint.h) be deprecated in favor of a new class called `RigidBodyMobilizer`?

Notice that it is in a directory called "joints". Should this directory be renamed "mobilizers"?

RussTedrake, 2016-09-01:
I have to say that I’m really not a fan of the newly invented name.

liangfok, 2016-09-01:
OK. How about let's leave the public-facing API use "joint" but the internal implementation details use "mobilizer"?

amcastro-tri, 2016-09-01:
I should say that we don't need to rush this decision (notice the "low" priority).

> Should DrakeJoint be deprecated in favor of a new class called RigidBodyMobilizer?

I don't think so. Joints, as in the concept that @sherm1 describes, can be implemented as a collection of `Mobilizers`, `Constraints` (some wacky mechanism not just constrained to rotate around an axis but more generally on a surface) and `ForceElements` (say friction). Right now `DrakeJoint` is all of that in one single class. I think that ideally we would like to have this concept of `Mobilizers` split from the concept of a `Joint` but it'd require more than just a simple refactoring.

> OK. How about let's leave the public-facing API use "joint" but the internal implementation details use "mobilizer"?

I think that is a good idea. But again, if we go forward with the concepts here proposed (which I personally support), it'd require more than a simple refactoring. So please do not rush on this and take this issue as an open place for discussion. I think that was the original intent @sherm1?

sherm1, 2016-09-01:
> take this issue as an open place for discussion. I think that was the original intent @sherm1?

Yes. My main goal was to point out that there is an ambiguity in the current terminology. Thinking clearly about it is important; the particular terminology less so. That said, it would be nice to agree on some bulletproof terminology and there are few good choices.

sherm1, 2016-09-01:
> I have to say that I’m really not a fan of the newly invented name.

I'd be happy with anything unambiguous. There is no widely-accepted term that distinguishes a user's conception of "joint" from its implementation in a minimal-coordinate tree. The terms "inboard" and "outboard" for tree directions have a long history (from Tom Kane). "Inboard joint" is synonymous with "mobilizer" (we used that term in sd/fast) but doesn't work as well in practice because it pulls in the tree direction concept at times when you just want to to talk about the joint properties. But it is still far preferable to "parent joint" which muddles the user-defined joint coordinate conventions with the hidden tree structure.

sherm1, 2016-09-01:
> Should DrakeJoint be deprecated in favor of a new class called RigidBodyMobilizer?

Discussion seems more appropriate at this point. For comparison, Simbody has a class `Body` to hold a description of mass properties, geometry, and the like and a class `MobilizedBody` that contains a body and its mobilizer. MobilizedBodies are the nodes of Simbody's multibody tree. While "joints" have an independent existence, a mobilizer is an ephemeral implementation concept that only makes sense as part of a tree containing the body it mobilizes. There is a utility class [`MultibodyGraphMaker`](https://simtk.org/api_docs/simbody/latest/classSimTK_1_1MultibodyGraphMaker.html) for mapping a set of bodies and joints to a spanning tree of mobilized bodies plus some constraint-based loop-closing joints.

tkoolen, 2016-09-01:
> I should say that we don't need to rush this decision (notice the "low" priority).

Exactly, let's talk about this before acting.

My opinion is that a single `Joint` class should at one point have methods that allow all of the functionality of a 'mobilizer', 'constraint', and 'bushing'. A joint is/can be all three of those things; why split it up in a (to me) artificial way? So I guess my opinion is completely opposite of @amcastro-tri's.

The terms parent and child are currently being used in the context of a tree, where there is no ambiguity, except for loop joints. Featherstone uses the terms predecessor and successor to define the direction of a loop joint in a kinematic tree; I like those terms (at the very least better than `frameA_` and `frameB_` currently in use in `RigidBodyLoop`). In my opinion, loop joints should not be separate classes. Instead, they should just be `Joint`s that are only *used* differently from 'tree' `Joint`s. They should be stored along with references to their successor and predecessor rigid bodies in `RigidBodyTree`, and `Joint` should have a method that returns the constraint-force subspace (terminology from Featherstone), as well as one for computing the constraint violation given a transform (to be used for constraint stabilization and constraint resolution).

In a general directed acyclic graph, parent/child relationships are undefined and shouldn't be used, but we're not at that point yet: we simply don't have such a representation at this point in time. Surely, `RigidBody`s should not store their 'parent joint', since that is particular to the kinematic tree structure and may not be relevant for a maximal coordinates implementation, so I think that should be changed to future-proof things. But I disagree that the current terminology is ambiguous *within the context of `RigidBodyTree`*. I think the perceived ambiguity is due to unjustified extrapolation to a future world bigger than `RigidBodyTree`, and a desire to split up what should not be split up in the first place.

amcastro-tri, 2016-09-01:
> In my opinion, loop joints should not be separate classes. Instead, they should just be Joints that are only used differently from 'tree' Joints.

I do agree loop joints should be `Joints`. However, the implementation of a loop joint is **very** different from say a `RevoluteJoint`. While a `RevoluteJoint` falls in the category of a `Mobilizer` (for the lack of a better terminology in this discussion), `LoopJoint` falls in the category of a `Constraint`. Mathematically, numerically and programatically they are handled in different ways.

In my mind it makes sense to have some sort of `Mobilizer` (or any other term) granting DOF's to a body, `Constraint`'s removing DOF's from a system of rigid bodies and then `Joint`'s. Particular cases of `Joint` might be `RevoluteJoint` (implemented with a 1 DOF mobilizer) and `LoopJoint` (implemented as a holonomic constraint).  The user always (unless there is a very specific problem) plays around with `Joint`'s.

Splitting the implementation in these two orthogonal axes allows for more flexibility of the system. For instance, if we actually had a flexible way to specify constraints, then #3131 would'be been super simple to implement without even mentioning joints in the design.

tkoolen, 2016-09-01:
> the implementation of a loop joint is very different from say a RevoluteJoint

The way I see it, to `RevoluteJoint` only needs to implement two additional methods (constraint-force subspace and constraint violation) to be *used* as a loop joint in either `RigidBodyTree` or some future class that does maximal coordinates. The way such a `RevoluteJoint` would be used depends on its role as either a loop joint or a 'tree' joint (in the context of `RigidBodyTree`), but to me the concept remains the same.

If `LoopJoint` is a `Joint`, how do you envision the inheritance diagram? I'm assuming there's going to be more than one type of loop joint; so could you clarify the relationship with other `Joint`s?

I'm not really opposed to using `mobilizer` as a variable name inside `RigidBodyTree` to signify a 'tree' joint if it needs to be distinguished from a loop joint, I'm just very opposed to splitting up `Joint` into `Mobilizer`, `Constraint`, and `Bushing` classes.

sherm1, 2016-09-01:
> I'm just very opposed to splitting up Joint into Mobilizer, Constraint, and Bushing classes.

To be clear: I didn't intend to propose anything about class structure here. I wanted to point out the technical distinction between the logical concept of a joint (with arbitrary parent/child or frameA/frameB or whatever) and that joint's implementation in internal coordinates where there is a very real ordering established by the tree topology. From your comment, I think you got it!

How best to map from logical joint to particular implementation is a separate question, likely with lots of good answers as long as the concepts are clear.

sherm1, 2016-09-01:
> Featherstone uses the terms predecessor and successor to define the direction of a loop joint in a kinematic tree; I like those terms (at the very least better than frameA_ and frameB_ currently in use in RigidBodyLoop).

BTW, frames `A` and `B` actually sound better to me than predecessor and successor which imply a sequence that doesn't exist. IIRC, that is exactly how Paul @Mitiguy's dynamics textbook defines joints (as a relationship between frames A and B), and I think it better captures the arbitrariness of the choice of direction. I'm happy with "parent frame" and "child frame" for this same purpose since that is a long-standing convention in robotics for joint definition. I just don't want us then to use the same terms to describe the multibody tree topology, which is distinct.

mitiguy, 2016-09-07:
Hi Twan,

I am trying to send you a 2-page .pdf.
However, I cannot find your e-mail.

Please send me your e-mail address.

Thanks,

Paul
paul.mitiguy@tri.global

P.s.  To help speed things, we may want
to employ Michael Posa (one of my former
students - who was a TA for one of my classes).

EricCousineau-TRI, 2019-03-01:
Given that we've now adopted usage of `MultibodyPlant` and have hashed most terminology debates there, is this issue still alive?
(I believe more specific debates have their own issue, like Jacobians?)

sherm1, 2019-03-01:
Let's close this fine discussion and open new ones as specific needs arise in MBPlant.

closed 2019-03-01 by sherm1