RobotLocomotion/drake #3325: Output KinematicsChache from RBP

amcastro-tri, 2016-09-01:
Per discussion in #3316, we need to output the `KinematicsCache` as an output for `BotVisualizer` not having to recomputed already available quantities in the `KinematicsCache`.
The output will be in the form of an `AbstractValue` and `BotVisualizer` will have an input port for it.

For systems like a simple pendulum dynamics but that we want to render as a three dimensional multibody system, we will just leave the `KinematicsCache` port unconnected case in which the `BotVisualizer` will just instantiate its own `KinematicsCache` as it is done right now. Therefore we would need the capability to leave inputs unconnected and for `System<T>`'s to query whether these are connected or not.

Two questions arise @david-german-tri and @sherm1:

1. Can we leave input ports disconnected? would it be a good idea to do so?
2. Could we (should we?) provide a `System<T>` method to check whether an input port is connected or not? like `System<T>::is_input_port_conneted(port_number)`.

jwnimmer-tri, 2016-09-01:
I think this design proposal is a super-confusing way to address the challenges that #3316 was trying to solve, and that #3316 was more on the right track.

amcastro-tri, 2016-09-01:
I am open to close this issue and re-open #3316. But let me see if I can share with you some thoughts. 
Just think about how the design in #3316 would get connected.

**Example 1.** RBP --> RbpTransalator --> BotVisualizer
In order for RbpTransalator to not recompute the `KinematicsCache` it'd need it as an input form RBP. So the desing here proposed does not add this new `AbstractValue` port but it was implicit in #3316 (in case with "super-confusing" this is what you were referring to).

**Example 2.** SimplePendulum --> RbpTransalator --> BotVisualizer
In this case `SimplePendulum` does not have a `KinematicsCache`. Somebody needs to compute it. Well, in that case it seems we'd then need to leave the `KinematicsCache` port unconnected and then `RbpTransalator` would compute the kinematics cache.

So, all the super-confusing components were already present in #3316. Actually here I am just simplifying to two classes instead of three. Again, I might be missing something. Let me know what you think.

jwnimmer-tri, 2016-09-01:
This is tagged as high priority; is that true?  Is the only top-level goal to be able to write something like `BotVisualizer`?  I feel like a whiteboard would really help, which means deferring to next week.

liangfok, 2016-09-01:
And a System 2.0 versions of `RosTfPublisherSystem`, and `ImuSensorSystem`. OK for deferring till next week.

amcastro-tri, 2016-09-01:
Yes, that is why the high priority. It should be easy to implement once we agree on what exactly to implement. The high priority is because we at least need RBP, and BotVisuzlizer to simulate something and visualize it (as done with the Kuka demo).
I like the idea of a f2f with whiteboard! and for next week, yes.

sherm1, 2016-09-01:
KinematicsCache is going to have a short lifetime since its functionality should be absorbed by System2's built-in caching facility. So I don't think we should be designing in any new dependence on it (unless this is meant as a short-term hack). Exposing meaningful information on output ports makes more sense and is more general since any System could output such data, while KinematicsCache is only available from Systems that have a RigidBodyTree embedded somewhere.

liangfok, 2016-09-01:
Is it possible to incrementally transform the `KinematicsCache` into System2's cache? Are we still planning on incrementally morphing `RigidBodyTree` into the version used by Dynamics 2.0?

sherm1, 2016-09-01:
> Is it possible to incrementally transform the KinematicsCache into System2's cache?

That will be much easier if we don't expose KinematicsCache outside RBTree and RBPlant. Then we can gradually move its contents in the Sys2 cache until it is empty.

tkoolen, 2016-09-04:
Sharing of `KinematicsCache` (or whatever will replace it in the future) is also pretty much required for an efficient implementation of `RigidBodySensor`s. Perhaps a quick and dirty solution is to just construct these RigidBodySensors with a reference to the `KinematicsCache` internal to RigidBodyPlant for now (which should by the way not be constructed on every call to `EvalTimeDerivatives`!).

liangfok, 2016-09-04:
Would it be too much against the spirit of System 2.0 to have `RigidBodyPlant` maintain a pointer to a mutable `KinematicsCache` object and all consumers of the `KinematicsCache` maintain a `const` reference to it?

sherm1, 2016-09-05:
System 2 has its own nascent caching mechanism which should not be strangled in the womb by outputting some other caching mechanism!

tkoolen, 2016-09-06:
For future reference, I'll restate an example I gave on slack earlier: consider a robot with 20 accelerometers attached to it (we've actually discussed doing this for Valkyrie; this is not some contrived example). If every RigidBodyAccelerometer has to redo the work to compute the data currently stored in `KinematicsCache`, that would be terrible for performance. In my view, this demonstrates the need for some sort of sharing of cached results between Systems. 

So what is the long-term vision for this? The discussion on slack seemed to be leaning towards only outputting a minimal representation of state (q and v for a RigidBodyPlant), and that cache is only an implementation detail, hidden to someone assembling a Diagram. That would imply that sharing cached results should be done 'underhand'.

Instead, I'd like to argue that passing data between systems should always be done through ports; otherwise this nice, decoupled system design is in name only. So I'm in favor of @amcastro-tri's original plan of having a `KinematicsCache` output port for `RigidBodyPlant` (or whatever the cache for a `RigidBodyPlant` will be renamed to in the future).

liangfok, 2016-09-06:
How about instead of having _one_ output port that contains a `KinematicsCache` object, we have _n_ output ports, one for each of the _n_ intermediate values stored in what is now the `KinematicsCache`? We can always add more output ports if and when we want more intermediate values.

sherm1, 2016-09-06:
The System 2 cache is contained inside the Context and is available to all subsystems for every computation performed at run time (in a carefully managed way that guarantees up to date results). In a sense it is *always* exported implicitly. It should never be exposed explicitly on an output port.

> If every RigidBodyAccelerometer has to redo the work to compute the data currently stored in KinematicsCache, that would be terrible for performance.

Agreed! That would never happen making proper use of the System 2 facilities.

amcastro-tri, 2016-09-06:
> The System 2 cache is contained inside the Context and is available to all subsystems for every computation performed at run time

I wonder what that API would look like. Right now we can access subsystem contexts like done [here](https://github.com/RobotLocomotion/drake/blob/master/drake/systems/framework/primitives/pid_controller.cc#L85) for the PID controller. The access can be done if you have the pointer to the system which subcontext you want to access to. 
Consider a simple example like `BotVisuzlizer`, should `BotVisualizer` keep a (non-owning) pointer to the `RigidBodyPlant`?

tkoolen, 2016-09-06:
Hmm, I mean it works, but the output ports of a `RigidBodyPlant` are kind of just for show then (except for properly setting up the dependency graph). `RigidBodySensors` would be able to do their work just fine without any information from `RigidBodyPlant` output ports.

I suppose maybe you're planning to require having access to the state vector in order to access the cache for a system (this is not what's currently implemented), but in that case, why not just have a scalar output port, outputting a hash value computed from the state?

david-german-tri, 2016-09-06:
From @sherm1:

> > If every RigidBodyAccelerometer has to redo the work to compute the data currently stored in KinematicsCache, that would be terrible for performance.

> Agreed! That would never happen making proper use of the System 2 facilities.

I agree as well, and for the same reasons.  This would be much clearer if the System 2 facilities actually existed, so I will dust off that branch this week.  It was previously PR #3135.  We tabled it because there were competing priorities, and because @sherm1 and I have an unresolved design disagreement.  (That disagreement is tangential to this thread, where we entirely agree.)

From @tkoolen:

> Instead, I'd like to argue that passing data between systems should always be done through ports; otherwise this nice, decoupled system design is in name only.

I absolutely, 100% agree with this.  If we allowed `mutable` members of `System<T>` subclasses, the whole framework would be pointless.

> So I'm in favor of @amcastro-tri's original plan of having a KinematicsCache output port for RigidBodyPlant (or whatever the cache for a RigidBodyPlant will be renamed to in the future).

As an interim solution, I think this (or anything similar) is fine.  It also gets at an important point: the System2 cache infrastructure will supersede the aspects of KinematicsCache that manage data lifetime.  It will _not_ supersede the aspects of KinematicsCache that are a notation for particular, expensive computations about RigidBodyTree.  We will still need that notation, to be stored in System2 cache, and to be transmitted on a `RigidBodyPlant` output port to downstream consumers.

david-german-tri, 2016-09-06:
Sigh.  My post crossed the most recent from @amcastro-tri and @tkoolen in flight.

I do not think @sherm1 meant to claim that multiple subsystems would _share_ a cache.  If he did, then I strongly disagree!  I believe Sherm meant, and I also would assert, that (a) each subsystem always has its _own_ cache available and (b) cache entries which depend on a given input are invalidated whenever that input changes.

So, a downstream consumer of KinematicsCache can check whether its KinematicsCache input has changed, and redo resulting computations only if it hasn't.

sherm1, 2016-09-07:
> I do not think @sherm1 meant to claim that multiple subsystems would share a cache. 

Right! What I meant is that the full cache is always lurking behind every computation within a Diagram so that (as managed carefully via input and output ports) all needed information can be obtained without (a) passing caches around,  (b) recomputing anything that has already been computed, or (c) computing things that don't get used. Each subsystem sees only its own cache, but the framework as a whole has the Diagram perspective and can efficiently deliver computations between subsystems via the ports.

tkoolen, 2016-09-07:
OK, that makes sense. Back to the original issue :-). What ports do you think should be added to RigidBodyPlant to enable efficient RigidBodySensors and visualizers?

RussTedrake, 2016-09-08:
fyi - this is also blocking the valkyrie sim.

liangfok, 2016-09-08:
To get the conversation started how about let's make it output the following at least initially? We can always add more output ports with additional data as necessary.

1. A vector of joint positions (`q`).
2. A vector of joint velocities (`q_dot`).
3. A newly defined `WorldVisualizationDescription` object that contains, for each `RigidBody` in the world, a description of how it should be visualized.
4. A newly defined `RigidBodyPoses` object that contains, for each `RigidBody` in the world, its 6-DOF pose relative to the world frame.
5. A newly defined `ContactWrenches` object that contains, for each contact in the world, the two bodies in contact and the contact wrench.

RussTedrake, 2016-09-09:
@liangfok asked me to comment on this

@liangfok, i think that the conceptual output of the rigid body plant is very clear -- it should contain the generalized state of the plant (which is positions and velocities).  in my view most of the discussion here is about passing the result of computations on that state (transforms, accelerations, etc) simply to make downstream computations more efficient for sensors, etc.

liangfok, 2016-09-09:
Yes, the additional state that we were thinking about outputting from `RigidBodyPlant` are indeed mainly for computational efficiency. Another reason I've heard is to prevent downstream systems from needing to have a `const` reference to the `RigidBodyTree` (i.e., an encapsulation argument).

I believe one of the original instigators that resulted in this discussion is here, where it was pointed out that `BotVisualizer 2.0` could be vastly simplified if it were provided rigid body pose information: https://github.com/RobotLocomotion/drake/pull/3228#issuecomment-242509888

tkoolen, 2016-09-09:
> i think that the conceptual output of the rigid body plant is very clear -- it should contain the generalized state of the plant (which is positions and velocities)

@RussTedrake, I disagree; I think that the *state* of a rigid body plant is very clear, but I don't think that the state should be the only possible output presented on a rigid body plant's output ports, even conceptually.

amcastro-tri, 2016-09-10:
I will summarize here a f2f conversation we just had with @sherm1 about this.

### About visualizing a `RigidBodyPlant`

**First**, We thought that a more appropriate name for System 2 `BotVisualizer` would be something like `RbtLcmPublisher` (other similar name suggestions are welcome) so that it is abolutely clear what this system does: it publishes LCM messages for our `drake-visualizer` that are only valid to visualize a `RigidBodyTree`.
Also, it then makes sense that this system is then initialized from a constant reference to a `RigidBodyTree`.

**Second**, `RigidBodyPlant` would have output ports for each of its `RigidBody` poses so that they can be inputs to `RbtLcmPublisher`. Since outputs in System 2.0 will be cached, accessing these from `RbtLcmPublisher` won't incur in any additional computation solving the problem of unnecessarily re-doing computations.

### About visualizing a system like a `Pendulum`.

We think that in a case of a system like say a simple pendulum that clearly has all the information to plot itself and was written by a user with a clear idea in mind of how that system would be visualized, it makes sense for this system to also output the poses for its links.
Therefore System 2.0 pendulum would output the rigid body pose of its link. Similarly for other systems like say `SpringMassSystem`.

jwnimmer-tri, 2016-09-10:
I think it will be a common occurrence for Systems that connect to the `RigidBodyPlant` output to want the position and velocity for all bodies, not just some.  The above proposal reads to me like it proposes one port per body.  Offering that is fine if you want, but I would also like to see offered a single output port that provides the entire state for all bodies (in addition to whatever other outputs we also end up offering).  The so-called `RbtLcmPublisher` would use that all-in-one port, of course.

liangfok, 2016-09-10:
I would also prefer a single port that contains all of the poses. It seems cleaner and more tidy that way.

tkoolen, 2016-09-10:
> `RigidBodyPlant` would have output ports for each of its RigidBody poses

and

> I would also like to see offered a single output port that provides the entire state for all bodies

This is a difficult one.

I think the 'small' output ports for each of the RigidBody poses should exist in any case, because otherwise you're probably requiring some systems downstream to accept more data than what they need to do their jobs: 'fat' output ports require fat input ports. This would be kind of like only wanting a little pet rat, but getting the whole pet store and having to deal with it. Fat output ports increase coupling in the code base, and make it harder to write unit tests for individual systems. Small output ports provide better data abstraction.

But should there be a fat output port with all poses in addition to the small output ports? I could be persuaded both ways, but I'm leaning towards a "no", just because it feels a bit redundant. Maybe what's needed instead is some syntactic sugar for connecting a whole bunch of ports in a one-liner?

> `RbtLcmPublisher`

I prefer not to abbreviate RigidBodyTree (or rather, to abbreviate in general). LCM is a very well-established abbreviation, but RBT is more likely to confuse people that are new to the code base.

liangfok, 2016-09-10:
:+1: for providing sugar for wiring up a whole bunch of related ports into one.

Question: Is it OK for the number of output ports to be dependent on the model that's being loaded? In other words, will the decision to make the number of output ports proportional to the number of rigid bodies introduce too much coupling between the diagram and the model (URDF, SDF, etc.) that it can load?

RussTedrake, 2016-09-10:
I still think the output of the RigidBodySystem should always be the generalized state.  We could have a sensor/modifier that turns that into poses of some end effector.  Again, the design which was thought out and discussed (at length) before is described briefly in
https://github.com/RobotLocomotion/drake/blob/master/drake/systems/plants/RigidBodySystem.h#L15

I think we should either consider RigidBodyPlant to be what I was calling RigidBodyTree in that doc, or include sensors/actuators and call it RigidBodySystem (where the output is now the sensors).

liangfok, 2016-09-10:
OK. I suppose we can stick with the System 1.0 design of lumping the `RigidBodyTree`, `Sensors`, and `ForceElements` all into a single System 2.0-based `RigidBodySystem`.

@RussTedrake, just to be clear, are you opposed to having this system also output the individual rigid body poses in addition to its generalized state? Only outputting the generalized state will force downstream system like `BotVisualizer` to recompute the pose information.

liangfok, 2016-09-10:
Thinking about this some more, I suppose there's nothing preventing us from aggregating the `RigidBodyTree`, `Sensors`, and `Force Elements` all into a single `RigidBodySystem`, but still have multiple output ports, one for the generalized state of the tree, and other ports for each individual sensor and force element.

tkoolen, 2016-09-10:
> lumping the RigidBodyTree, Sensors, and ForceElements all into a single System 2.0-based RigidBodySystem.

That works too.

> I still think the output of the RigidBodySystem should always be the generalized state.

I really think that just can't work because of https://github.com/RobotLocomotion/drake/issues/3325#issuecomment-244989985. If sensors are separate systems (as was the case in the design you linked) and the only outputs available to the sensors are q and v, then you either have to redo lots of work, or share caching between systems, making the port-based design be just for show.

amcastro-tri, 2016-09-10:
@jwnimmer-tri, @liangfok. Yes, I am sorry if I didn't explain correctly but my proposal is to provide a single port with all body poses, not a port per body.

@RussTedrake 
> I still think the output of the RigidBodySystem should always be the generalized state. We could have a sensor/modifier that turns that into poses of some end effector.

Yes, the proposal here is to have that output port with the state plus an additional port with poses that could for instance be used by the bot visualizer. The reason for this port is to avoid re-doing computations in the bot visualizer which were already performed by the RigidBodyPlant. These poses (and potentially other kinematic quantities) could additionally be used by sensor systems which then won't require to recompute the body poses or any kinematics in order to figure out say an acceleration.

> Thinking about this some more, I suppose there's nothing preventing us from aggregating the RigidBodyTree, Sensors, and Force Elements all into a single System 2.0-based RigidBodySystem

I do not like that idea. Dividing our system into a `RigidBodyPlant` and into a bigger `RigidBodySystem` containing now additional sensors and actuators IMO allows for greater flexibility and a more incremental development. That discussion does not belong to this issue though.

amcastro-tri, 2016-09-10:
Please let me summarize the plan of action into what I think covers all of your concerns. I am intentionally leaving out the discussion of sensors/actuators and other systems that can be attached to a `RigidBodyPlant` because System 2.0 now allow us to develop into smaller increments by composing simpler systems into a `Diagram`. `RigidBodyPlant` ONLY contains the `RigidBodyTree` model of the world with:
**Inputs**: the actuators' generalized forces.
**Outputs**: state + convenience output port (a single one) with bodies' kinematics. 

The later can be used for either visualization or by smaller systems like sensors so that this information is already cached and does not need re-computations.

The output port with the system kinematics will be wrapped in a nice API so that a request on per-body basis can easily be performed as well. So this output port won't just be a bare array of doubles but more like an array of `RigidBodyKinematics` (for the lack of a better name).

liangfok, 2016-09-10:
@amcastro-tri I in general agreement with you, but was only considering mirroring System 1.0's approach of combining everything into a `RigidBodySystem` based on @RussTedrake's comments above.

In what way does making sensors and force elements increase flexibility and incremental development? Even if they were contained in a `RigidBodySystem` along with a `RigidBodyTree`, they would still be their own independent objects that are instantiated at run-time either programmatically or via a specification file like SDF.

amcastro-tri, 2016-09-10:
> I am in general agreement with you

Good.

> but was only considering mirroring System 1.0's approach 

System 2.0 does not need to mirror System 1.0 (of course we still have similar concepts, but not implementations). We can do better with @david-german-tri's design now.

> In what way does making sensors and force elements independent systems increase flexibility and incremental development?

A simple example: #3245 could be merged right now as it is without having to design right now the right abstractions to add sensors. Even adding the output poses I propose here would be a separate PR. IMO that is incremental development. 

How it is more flexibly to design sensor systems? well, that'd require a f2f or VC but I don't think that belongs to this issue anyway.

liangfok, 2016-09-10:
I see, so you're saying incremental w.r.t. pull requests. That is true.

However, would you agree that lumping the tree, sensors, and actuators does not enable additional incremental development w.r.t the creation of a model?

sherm1, 2016-09-10:
> @tkoolen: I think the 'small' output ports for each of the RigidBody poses should exist in any case, because otherwise you're probably requiring some systems downstream to accept more data than what they need to do their jobs: 'fat' output ports require fat input ports. 

In System 2 there is no cost to having a fat poses port even if only a few poses are used, because RigidBodyPlant has to compute them anyway. When the System 2 implementation is complete there will be no copying from output ports to input ports. The connecting lines in a System Diagram should not be thought of as pipes along which data flows, but just an expression of the connectivity. So picking a particular pose from the output port is actually just obtaining a reference to a particular element of RBPlant's internal pose cache.

There would be nothing to prevent us from adding options to RBPlant to make it produce more specific output ports. That would be a performance win for quantities that wouldn't otherwise be computed, but for poses we wouldn't gain anything.

tkoolen, 2016-09-10:
@sherm1, I wasn't talking about runtime cost.

sherm1, 2016-09-10:
> I wasn't talking about runtime cost.

Oh. To me it seems messier to add construction options to an RBPlant to generate body-specific pose outputs, then use those to make a custom-wired Diagram, than simply to select the poses you care about from the always-available general supply. Same goes for state -- it's easy just to output the whole `x` even if downstream only needs some of it.

liangfok, 2016-09-10:
I was just thinking that if we end up lumping `RigidBodyTree`, `Sensors`, and `Force Elements` all within a single System 2.0 version of `RigidBodySystem`, we might as well also throw `BotVisualizer` into `RigidBodySystem`. I'm not sure if this is an argument for or against lumping everything together. Just something to think about. Maybe the port-based architecture is only for connecting things that are less coupled like planner vs. controller vs. plant?

liangfok, 2016-09-11:
@RussTedrake, I believe @david-german-tri's post above (https://github.com/RobotLocomotion/drake/issues/3325#issuecomment-245105482) concisely describes the System 2.0 vision of caching. Basically, each system will have it's own private cache that's _not_ shared with other systems.

sherm1, 2016-09-11:
@RussTedrake, KinematicsCache will be replaced (gradually) by System 2's more-general caching mechanism so we don't want to expose KC in a System 2 system. Sys2 has more information so can trigger computation when necessary and avoid computing expensive quantities that aren't being consumed.

RussTedrake, 2016-09-11:
If that is the model, then i would argue for the rigodbodysystem approach, where the single system contains the list of sensors and actuators -- acting effectively as a very special case of diagram where cache sharing is allowed. 

Perhaps the botvisualizer tools also live as a part of this system so they share the same privilege?

RussTedrake, 2016-09-11:
@sherm1 -- totally agreed that the more general caching mechanism will be in play here.  But is that mechanism restricted to a single system, or can it be shared across systems?

sherm1, 2016-09-11:
> is that mechanism restricted to a single system, or can it be shared across systems?

It can be shared among subsystems contained in the same Diagram (however deeply nested). Sharing is carefully controlled though to prevent out-of-date references. Subsystem A can do anything it wants with its own portion of the Context (which contains the cache), but if it wants to use something from subsystem B that is mediated by the Diagram.

liangfok, 2016-09-11:
Are we accidentally delaying ourselves by prematurely optimizing things? @sherm1 has previously emphasized that the entire caching mechanism is _optional_ and that it will be easily togglable on / off. From a system user's point of view, the cache does not even exist. It will only have non-functional impacts on the overall execution of the software.

Given the above, how about let's charge forward implementing the `RigidBodyPlant`, `Sensors`, `ForceElements`, `BotVisualizer`, etc., as individual System 2.0 systems that input and output only the minimal generalized state (and have multiple instantiations of `RigidBodyTree` as necessary)? We can then perform some benchmarks to prove that we have a performance problem, which will justify (1) a sophisticated caching mechanism, (2) exposure of non-minimal state through ports, and/or (3) lumping multiple software abstractions into a single `RigidBodySystem` so they can share the same cache?

tkoolen, 2016-09-11:
In https://github.com/RobotLocomotion/drake/issues/3325#issuecomment-246115243, I meant sharing caching 'under the table' - not through ports. That's what I'm against, and so is @david-german-tri. If q and v are the only outputs, but all of the sensors actually only interact with cache that's not explicitly shared through a port, then the output port is just for show in my opinion.

> i would argue for the rigodbodysystem approach, where the single system contains the list of sensors and actuators -- acting effectively as a very special case of diagram where cache sharing is allowed.

This is certainly a viable option. Clearly we're having trouble defining the output ports of `RigidBodyPlant`, so maybe we just haven't drawn the system borders in the right place. On the other hand, if we're going to have sensors with state at some point (hysteresis perhaps), then it's going to be awkward to combine the state of the sensor with the state of the `RigidBodyPlant`, while it would be trivial if they were their own systems hooked up in a `Diagram`.

These are just some considerations, I don't know what the right answer is, but I'm still modestly in favor of a design with many small output ports that are created upon user request after constructing the `RigidBodyPlant` but before wiring up the `Diagram`.

tkoolen, 2016-09-11:
@liangfok, I don't think it's premature. It's pretty clear to me that e.g. each contact sensor system has to redo all of the kinematics, collision checking, and contact force computation, then that's going to be bad for performance.

liangfok, 2016-09-11:
I agree it will be bad performance. However, I'm unconvinced that it will be so bad as to be unusable. Having baseline numbers will be super useful to guide the discussion and design choices we make.

tkoolen, 2016-09-11:
@RussTedrake, thinking about it some more, the thing I'm proposing ('many small output ports created upon user request') is actually the same as the original `RigidBodySystem` design with `RigidBodySensor`s; I was just using different terminology. Each small output port would have the same role as a `RigidBodySensor` in the original `RigidBodySystem` design. Hysteresis and noise can be handled by hooking up a (future) sensor noise model to the perfect information coming out of the small output ports.

liangfok, 2016-09-11:
@tkoolen, in System 1.0, the outputs of `RigidBodySensor` objects is lumped together with the generalized state of the `RigidBodyTree` forming one potentially gigantic output coming out of the System 1.0 `RigidBodySystem`. I assume you're trying to avoid this in System 2.0? I would agree with that.

RussTedrake, 2016-09-11:
to summarize my stance, there are two proposals that I have seen before which I agree with.  
1) RigidBodySystem, which contains a list of sensors, etc, with the output being those sensors.  a fullstatefeedback sensor would output the generalized state if it is present.  that was the design i initially implemented in system1.0
2) RigidBodyPlant, who's output is only the generalized state.  sensors, etc are wired up using Diagram, and the Diagram's caching mechanism (eventually) optimizes away repeated calls.  @tkoolen -- in this model I am not proposing that anything is hidden.  the sensor blocks would still call the relevant methods, like forwardKinematics, on the generalized state, but those calls would be made essentially free because of the caching mechanism.

I like 2 better, and do agree with @liangfok that we could put it in place without worrying about performance first.  but I'm also ok with 1.

tkoolen, 2016-09-11:
@liangfok, ah yes, I'd forgotten about that. Yeah, let's not concatenate all measurements together.

For proposal 2 in @RussTedrake's comment, the blocking issue that needs to be resolved is whether or not caches are really private to each subsystem (but possibly shared explicitly via output ports), or shared with the whole diagram (implicitly, not through ports).

Just to summarize the discussion on this topic:
* @RussTedrake is in favor of sharing caches implicitly with the whole diagram (https://github.com/RobotLocomotion/drake/issues/3325#issuecomment-246153583).
* @david-german-tri (https://github.com/RobotLocomotion/drake/issues/3325#issuecomment-245101995, https://github.com/RobotLocomotion/drake/issues/3325#issuecomment-245105482)  and I (https://github.com/RobotLocomotion/drake/issues/3325#issuecomment-245099507) are in favor of keeping caches private to each subsystem.
* I'm a little unclear on @sherm1's stance (https://github.com/RobotLocomotion/drake/issues/3325#issuecomment-245092887, https://github.com/RobotLocomotion/drake/issues/3325#issuecomment-245354356, https://github.com/RobotLocomotion/drake/issues/3325#issuecomment-246155598).

If we end up going with subsystem-private caches, the next decision is whether to have many small `RigidBodyPlant`/`RigidBodySystem` output ports, or one, or a few fat output ports.
* @amcastro-tri and @sherm1 (https://github.com/RobotLocomotion/drake/issues/3325#issue-174531541, https://github.com/RobotLocomotion/drake/issues/3362#issuecomment-245170332), as well as @liangfok (https://github.com/RobotLocomotion/drake/issues/3325#issuecomment-245548677) are in favor of fat output ports.
* I am in favor of many small output ports (https://github.com/RobotLocomotion/drake/issues/3362#issuecomment-246158590, https://github.com/RobotLocomotion/drake/issues/3325#issuecomment-246160130). @RussTedrake is as well, but would prefer sharing caches so that this isn't relevant in the first place (https://github.com/RobotLocomotion/drake/issues/3325#issuecomment-246168296).

We'll just need to take decisions on these two topics to make progress. Please let me know if I misrepresented your stance.

jwnimmer-tri, 2016-09-11:
I'm glad there's more focus now!  I'll throw in one tangential point in case it helps you all decide.

A fat output port doesn't mean that you need to suffer System 1's manual indexing math (the RigidBodySystem 1.0 idiom with "get_num_positions" / "get_sensor_start_index" and whatever).  If you choose the fat port, it should provide a RigidBodySystem-specific subclass of BasicVector that consumers would downcast.  That vector subclass can offer any number of API calls to access the underlying data in portions-at-a-time, such as back-referencing to model_ids in the tree, separating position data from sensor readings, etc.  The caching of the output and underlying Eigen matrix representation would be consolidated into one output port, but the using-code can be as sugar'd up for convenience as we like.

tkoolen, 2016-09-11:
@jwnimmer-tri, I'm aware of that. My reasons for not liking fat outputs so much are outlined here: https://github.com/RobotLocomotion/drake/issues/3325#issuecomment-246087775. But fat output ports certainly are a valid solution to this issue; just a matter of preference.

liangfok, 2016-09-11:
One option is to initially provide both fat ports and small ports, then over time we can see if both are in fact useful for different purposes.

sherm1, 2016-09-11:
> I'm a little unclear on @sherm1's stance [on sharing of cache]

I believe mine is the same as @david-german-tri's: we have a beautiful caching mechanism designed for System 2 that is completely sufficient for efficiently and correctly connecting subsystems via input/output ports. So my stance is that this is a non-issue. The cache should be invisible and the solutions should involve designing useful output ports that expose meaningful data.

[24 more replies]

closed 2017-02-01 by amcastro-tri