RobotLocomotion/drake #3326: Use upstream Eigen instead of POD wrapper

jamiesnape, 2016-09-01:
Yes: https://bitbucket.org/eigen/eigen/
No: https://github.com/RobotLocomotion/eigen-pod/

tkoolen, 2016-09-01:
I'm in favor of this (and the related issues). Just one thing to think about: up until a few days ago (before https://github.com/RobotLocomotion/eigen-pod/pull/7), we were applying our own patches to Eigen. These patches were stored in eigen-pod. Where should they be stored in the future if we needed manual patches again? Or would we just fork Eigen and point to our own branch?

jamiesnape, 2016-09-01:
First step is to try to upstream them.

mwoehlke-kitware, 2016-09-05:
Another problem... I don't know that we can use an hg repo as a submodule... maybe we could use e.g. https://github.com/RLovelett/eigen as the upstream instead? (It's... not clear if or how well `git-hg` can be used for a submodule...)

tkoolen, 2016-09-05:
Another option is to go back to `externalProject_add`.

mwoehlke-kitware, 2016-09-05:
We never stopped using `externalproject_add`. Yes, it's wrapped in `drake_add_external` right now, but teaching that to handle externals that aren't submodules isn't a problem. The problem is whether we are willing to lose the benefits (change tracking, especially) of having externals as submodules.

jamiesnape, 2016-09-06:
Being on upstream is more important than having a submodule in my opinion. Otherwise, I guess we mirror to a Git repository.

PR #3439 (2016-09-16): Use non-POD eigen
Fixes #3326.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/3439)
closed 2016-09-16 by jamiesnape