RobotLocomotion/drake #3330: Use upstream GTK+ instead of POD wrapper

jamiesnape, 2016-09-01:
Yes: http://win32builder.gnome.org/gtk+-bundle_3.6.4-20130921_win32.zip
Yes: http://win32builder.gnome.org/gtk+-bundle_3.6.4-20131201_win64.zip
No: https://github.com/RobotLocomotion/gtk-pod/

mwoehlke-kitware, 2016-09-05:
I'm not sure this one makes sense. "gtk-pod" isn't so much a POD as a wrapper for installing GTK from the aforementioned upstream sources. In order to get rid of it, we would have to duplicate all of the logic in its CMakeLists.txt in Drake proper, and TBH I'd just as soon not do that, as it makes things much more complicated. (Also, I don't think we wouldn't want to move from a private, CMake-ified version of some project to an upstream using some other build system, which is similar to what we would be doing here.)

We might rename the repo to "gtk-wrapper" or some such and possibly get rid of its `Makefile`, though.

jamiesnape, 2016-09-06:
Sounds reasonable.

mwoehlke-kitware, 2016-09-13:
Okay, so just downloading the packages (#3392) is *not* going to work; we either have to unpack the entire GTK "distro" into drake's install root (and ensure nothing tries to use the broken `.pc` files), or else deal with the non-trivial logic to install only the required subset and/or fix up the `.pc` files.

Note that someone (drake? libbot?) is still insisting on searching for `lcm.pc` which inturn needs `glib-2.0.pc`, which is why the experimental builds of #3392 are failing.

Thoughts?

jamiesnape, 2016-09-13:
Unpack GTK and delete all the pkg-config files maybe?

mwoehlke-kitware, 2016-09-13:
That just means anyone trying to use them will fail in a more obvious manner; we'd still have to find and fix any place looking for them.

jamiesnape, 2016-10-05:
Closing since Windows is no longer supported.

closed 2016-10-05 by jamiesnape