RobotLocomotion/drake #3344: Move fork of bot_core_lcmtypes to RobotLocomotion organization

jamiesnape, 2016-09-01:
Fork: https://github.com/mwoehlke-kitware/bot_core_lcmtypes

jamiesnape, 2016-09-01:
@david-german-tri Do you have sufficient tokens to create a repo?

david-german-tri, 2016-09-01:
I do.  Can you remind me quickly (a) why we are using a fork from https://github.com/openhumanoids/bot_core_lcmtypes and (b) why bot_core_lcmtypes belongs to OpenHumanoids and not RobotLocomotion in the first place?

@wxmerkt FYI

jamiesnape, 2016-09-01:
https://github.com/openhumanoids/bot_core_lcmtypes/pull/18

>Hi again, after thinking about this over the weekend, I'm going say that we cannot take on this change.
> I think that what you are suggesting would bring improvements and certainly more modern.
>
> The issue is the we do not have any software engineers working on the openhumanoids project to carry it though our ecosystem. (Compared to the drake, the project is tiny.)
>
> I think the preferred route might be for drake to avoid using this lcmtypes repository.

mwoehlke-kitware, 2016-09-01:
Please *don't* do this... it should move upstream to the libbot organization. (But feel free to change this issue description accordingly.)

jamiesnape, 2016-09-01:
Either works for me.

david-german-tri, 2016-09-01:
There's a `libbot` organization?

jamiesnape, 2016-09-01:
> There's a `libbot` organization?

News to me too.

mwoehlke-kitware, 2016-09-01:
> There's a `libbot` organization?

There is in the github sense: https://github.com/libbot2.

mwoehlke-kitware, 2016-09-07:
On further investigation, the current repo seems to be a dumping ground of LCM types from several projects, including drake itself. It would be good if that situation could be normalized somehow...

sammy-tri, 2016-09-28:
https://github.com/RobotLocomotion/lcmtypes now exists to collect lcm types, and is a clone of @mwoehlke-kitware's current drake branch.

jamiesnape, 2016-10-07:
Though we are not actually using it yet. Is there a follow up ticket or should this stay open until we do?

david-german-tri, 2016-10-07:
Oh, we aren't?  Oops.

mwoehlke-kitware, 2016-10-07:
Not yet. I am still working on changes on the LCM side (which will eventually become changes to RL/lcmtypes) with the desire/intent to fix all of #3397, #3683 and this at once, rather than needing 2-3 PR's for incremental updates. (Also, there are going to be non-trivial code changes to switch over, since the package name changed.)

sammy-tri, 2016-10-17:
In https://github.com/RobotLocomotion/lcmtypes/issues/3 we effectively decided not to do this.  Any objections to closing this ticket?

closed 2016-10-21 by sammy-tri