RobotLocomotion/drake #3346: Augment drake::systems::Simulator::StepTo() with a custom abort function.

liangfok, 2016-09-01:
A custom abort function takes as input the current simulation time and returns a boolean value that specifies whether the simulation should be terminated. This function should be called each cycle of the `while()` loop within `drake::systems::Simulator::StepTo()`.  When running Drake within a ROS node, the normal `ctrl+c` doesn't work because ROS [installs its own SIGINT handler](http://wiki.ros.org/roscpp/Overview/Initialization%20and%20Shutdown#Shutting_Down).

The current `while()` loop looks like the following:

```
while (context_->get_time() <= final_time) {
  ...
}
```
It needs to be modified to effectively be:

```
while (context_->get_time() <= final_time && ros::ok()) {
  ...
}
```

In Solver 1.0, this was achieved by providing a function called `should_stop()` that is part of the `while()` loop's control expression. To see how this feature was added to Solver 1.0, refer to issue #2192 and PR #2660.

liangfok, 2016-09-01:
@sherm1, any advice? I won't want to muck up your beautiful API any more than necessary? :smile:

jwnimmer-tri, 2016-09-01:
I would prefer not to pollute the analysis API with inversion of control hooks like this callback.

For right now, I believe the ROS simulator could just step something like one second at a time, and poll `ros::ok()` inbetween the calls to StepTo.  That should serve us well for quite a while, correct?

I could post several other "real" solutions that are more conceptually elegant, but I'd rather KISS here if we can.

liangfok, 2016-09-01:
OK, I think you're proposed simple solution will be fine for now.

sherm1, 2016-09-01:
> I think you're proposed simple solution will be fine for now.

Good. A nice solution to this will integrate cleanly with the simulator's not-yet-extant event handling facility, which will permit controlled interruption of simulation for execution of user code (potentially trajectory-altering) when designated events occur. One of the standard event types is "start of step" (Simulink calls that "major step"). That would be a reasonable time to check for user input such as `ros::ok()`. An alternative would be to poll with a regular sample time (say 100ms) the occurrence of which is also a type of event.

jwnimmer-tri, 2016-09-01:
So you need to publish time from Simulator 2.0 -> ROS?  In that case, you just need a time publishing System (block).  The details of StepTo matter not-at-all.

liangfok, 2016-09-01:
Oh yeah. That is a good point. I shall try that.

closed 2016-09-01 by liangfok

liangfok, 2016-09-02:
FTR: Creating a system dedicated to publishing the clock worked.

https://github.com/liangfok/drake/commit/2dddae76a5f802455322347ad9f2c6072a7e26d3