RobotLocomotion/drake #3351: Can't include bot_core LCM types that refer to other bot_core LCM types

tkoolen, 2016-09-03:
I'll use `systems/plants/test/lidarTest.cpp` as an example. In this file, we include `planar_lidar_t.hpp` as follows:
```C++
#include "lcmtypes/bot_core/planar_lidar_t.hpp"
```

This works. However, if I pattern match this to include `position_3d_t.hpp` right below, using
```C++
#include "lcmtypes/bot_core/position_3d_t.hpp"
```
then I get the following error:
```
In file included from /Users/twan/code/drake-distro/drake/systems/plants/test/lidarTest.cpp:9:
/Users/twan/code/drake-distro/build/install/include/lcmtypes/bot_core/position_3d_t.hpp:12:10: fatal error: 'bot_core/vector_3d_t.hpp' file not found
#include "bot_core/vector_3d_t.hpp"
         ^
1 error generated.
make[5]: *** [systems/plants/test/CMakeFiles/testLidar.dir/lidarTest.cpp.o] Error 1
make[4]: *** [systems/plants/test/CMakeFiles/testLidar.dir/all] Error 2
```

This is because `position_3d_t` has a `vector_3d_t` field, and (generated) bot_core LCM type hpp files include other bot_core LCM types as
```C++
#include "bot_core/$SOME_TYPE.hpp"
```

Please advise on the right way to fix this. This is currently a blocking issue for me.

tkoolen, 2016-09-03:
I can work around this by adding 
```CMake
include_directories(${CMAKE_INSTALL_PREFIX}/include/lcmtypes) #TODO(#3351)
```
below
```CMake
if(lcm_FOUND)
```
in drake/CMakeLists.txt.
Reduced priority.

mwoehlke-kitware, 2016-09-05:
Apparently, `lcmgen` currently generates code that expects to have the direct location of the headers in the include paths (i.e. for us, `include/lcmtypes`). If we don't like this, I think we need to add an option to `lcmgen` to specify an include prefix to use in the generated files. (Otherwise, your "work-around" is actually the correct solution, and we probably shouldn't name LCM type headers as `lcmtypes/...`.)

mwoehlke-kitware, 2016-09-05:
> your "work-around" is actually the correct solution

Scratch that... sort of. We're also missing `target_include_directories` for the LCM type libraries... including the directory as in your work-around is correct, but it should be hung off the LCM type libraries as a target usage requirement. (And in fact, we should have a `target_include_directories` regardless of what the actual directory is...)

tkoolen, 2016-09-05:
Thanks for looking into this. Is this the right way to do it then:
```cmake
target_include_directories(drake_lcmtypes_hpp INTERFACE ${CMAKE_INSTALL_PREFIX}/include/lcmtypes)
```
I can open a pull request if there's not more to it.

> [...] we probably shouldn't name LCM type headers as lcmtypes/ [...]

So this is currently possible because `${CMAKE_INSTALL_PREFIX}/include` is an include directory. I've always wondered whether it should be. What are your opinions on this? `${CMAKE_INSTALL_PREFIX}/include` is listed in the `*_include_dirs` properties of lcm, yaml-cpp, and GTEST (and possibly others that I don't have installed currently), whereas e.g. eigen3 and bullet only have their respective subdirectories of `${CMAKE_INSTALL_PREFIX}/include` in their `*_include_dirs` properties.

mwoehlke-kitware, 2016-09-05:
> Is this the right way to do it then?

...no. The `target_include_directories` is right, but the headers won't be in `${CMAKE_INSTALL_PREFIX}/include` until drake is installed. The correct directory would be wherever the headers are dumped at build time.

(Eventually, we should be creating exported targets, also, at which point it should also be guarded with `$<BUILD_INTERFACE:...>`, but don't worry about that for now.)

See `externals/lcm/lcm/CMakeLists.txt` for a "correct" example (includes stuff for exported targets).

> `${CMAKE_INSTALL_PREFIX}/include` is an include directory. I've always wondered whether it should be.

It depends on the package. Some packages have all their headers in a subdirectory, and expect to be included without said subdirectory in the path given `#include`. For these, it is correct that their include directories should be `<prefix>/include/<subdir>`. Others expect their headers to be included with said subdirectory as part of the include path (or, worse, don't *have* a subdirectory; fortunately most developers realize these days why this is bad). For these, `<prefix>/include` is correct.

Internally, LCM types appear to fall into the first category, but we are using them according to the second. This is inconsistent and suggests we should pick one and make both uses match, either by removing the leading `lcmtypes/` from our uses, or teaching lcmgen to use such a prefix.

I don't have a strong opinion which...

tkoolen, 2016-09-06:
@mwoehlke-kitware, since you have a better view of the build system at this point, would you mind putting together a pull request that fixes this?

mwoehlke-kitware, 2016-09-07:
Before we "fix this properly", I'd like to agree on what constitutes "properly". Should LCM type includes always start with `lcmtypes/`, including in generated files? Or should they never start with `lcmtypes/`?

mwoehlke-kitware, 2016-09-07:
As a related question, we ought to have a "library"¹ for the C++ types. Right now we have drake_lcmtypes (the C types library) and no "library" for the C++ types. What should we name it? My vote would be drake_lcmtypes-cpp (and rename drake_lcmtypes → drake_lcmtypes-c)...

Note that drake_lcmtypes_hpp is just a target, and can only convey build-order dependencies. With a proper "library" for the C++ types, this would become an implementation detail not used outside `drake/lcmtypes/CMakeLists.txt`.

(¹ Specifically, an `INTERFACE` "library", which is an entity in CMake off of which you can hang dependencies and target usage requirements and otherwise use like a library, but doesn't exist outside of propagating these.)

tkoolen, 2016-09-13:
Sorry about the response time.

> Should LCM type includes always start with lcmtypes/, including in generated files? Or should they never start with lcmtypes/?

I don't have a strong opinion either way.

> As a related question, we ought to have a "library"¹ for the C++ types [...]

I'm not 100% clear on what having an `INTERFACE` "library" would give us over a target, but if you think it's a good idea, I'm all for it. Those renames sound good too.

Somewhat related: https://github.com/RobotLocomotion/drake/issues/3426.

mwoehlke-kitware, 2016-09-13:
> I'm not 100% clear on what having an INTERFACE "library" would give us over a target[...]

See my previous comment. Specifically, "an `INTERFACE` "library" [...] is an entity in CMake off of which you can hang dependencies and target usage requirements and otherwise use like a library". So instead of:

    target_link_libraries(consumer libA libB...)
    add_dependency(consumer lcmtypes_hpp)
    target_include_directories(consumer PRIVATE path/to/lcmtypes)

...you only need to write:

    target_link_libraries(consumer libA libB... lcmtypes-cpp)

(In particular, the detail of where the headers are located becomes an implementation detail of the `INTERFCE` "library", rather than all consumers of the LCM types C++ bindings needing to know it. Encapsulation is good :smile:.)

tkoolen, 2016-09-13:
I see, yes, that's better.

sammy-tri, 2016-09-15:
This is also a blocking issue for me, as I'm also working on preparing a PR which wants to use bot_core LCM types which refer to other bot_core LCM types.  I stumbled on the same workaround which Twan did with adding an include_directories line, but I don't really want to include that in my PR.

I'm not sufficiently clear on the CMake changes suggested above to attempt to implement myself without additional guidance.

PR #3550 (2016-09-23): Fix LCM types includes
Update bot_core_lcmtypes, and actually look for it, so we can leverage it's interface include directories for consumers of the C++ bindings. Add an `INTERFACE` library with interface include directories for Drake's own C++ LCM type bindings. Use this instead of `add_dependencies` for consumers of the same (the `INTERFACE` library now propagates the dependency on the bindings being generated before trying to build code that uses them). Remove addition of drake's LCM types include directory as a global include. Fix a couple of spots where someone was using LCMSystem header [... cut]
commit 7e1f9402 (2016-09-23): Remove LCM types include work-around

Don't add Drake's LCM types include directory to global includes. Fix
some consumers of drakeLCMSystem that were using the headers but not
linking the library (and were unable to find the LCM type headers as a
result).

Fixes #3351.
closed 2016-09-23 by mwoehlke-kitware