RobotLocomotion/drake #3352: pose_t LCM type has shadowing issue

tkoolen, 2016-09-03:
On Clang, we use -Werror=shadow. `bot_core/pose_t.hpp` has a member `pos`. It also uses `pos` as a local variable in several methods, causing -Wshadow errors.

Right now, I'm working around this using
```C++
#pragma clang diagnostic push
#pragma clang diagnostic ignored "-Wshadow"
#include "lcmtypes/bot_core/pose_t.hpp"
#pragma clang diagnostic pop
```

What's the right, long-term way to fix this?

jwnimmer-tri, 2016-09-03:
In order of my personal preferences:

I suppose that renaming `pos` in the message is a non-starter, since it is widely used and shared?

Or, we could ask and/or PR the LCM code generator to produce warning-free code -- or at least generate separate `h` and `cc` files, so the warnings are locally controllable.  Either or both would benefit the wider community.

Or, if we just want to suppress all warnings for the lcm headers, a flag like `--system-header-prefix` or `#pragma clang system_header` would be a way to do so.

Or, we could just override on an ad-hoc basis, like the sample in the top post.  Hopefully only a few message fields are overloaded by the generated code's locals.

tkoolen, 2016-09-04:
I'm in favor of suppressing all warnings for LCM headers for now, and possibly working with the LCM people to split their generated files up into `h` and `cc` later.

Renaming `pos` in the message should not be necessary, and yeah, the message is used in other projects as well.

It's debatable whether it can be considered a bug in their code generator, since they can't predict the names of the fields we use, and they don't share the same coding standards (Wshadow isn't enabled by default, and not everybody considers shadowing bad practice). Splitting things up into `h` and `cc` files would improve things, but that's a big project.

liangfok, 2016-09-04:
I'm OK with suppressing all warnings related to auto-generated LCM code for now.

I created an issue on LCM's issue tracker to track a more permanent solution (see link above).

jwnimmer-tri, 2016-09-05:
The LCM codegen is plenty aware of the variable names the message uses -- its emitting the entire struct and function!  It would be fair to ask that the codegen be compatible with a wide variety of downstream warning flags (ours and others') -- so that as many project as possible can use LCM as easily as possible.  The codegen could add salt to its own local variable names (possibly only when needed) in order to avoid conflicting with user-provided names.

mwoehlke-kitware, 2016-09-05:
> Or, we could ask and/or PR the LCM code generator to [...] generate separate `h` and `cc` files, so the warnings are locally controllable.

(Pedantic: I assume you mean `.hpp` and `.cpp`? `.h` would be the C headers...)

This would be a pretty substantial change, not least because LCM C++ bindings are currently header-only. OTOH, at least for CMake, having a proper library makes build-order dependencies and target usage requirements easier, at least for CMake < 3.3. I'm not sure, though, that upsetting everyone's build systems is worth it. (To be clear.. if this change was made, every project that builds LCM C++ bindings ­would have to be updated.) Another reason I'm not sure it's worth it is you would lose inlining, and my understanding is that the C++ bindings are a relatively light wrapper over the C bindings.

Renaming the variable in the generated code might be doable...

> Or, if we just want to suppress all warnings for the lcm headers

I think `target_include_directories(... SYSTEM ${lcm_headers_dir})` could work here; they are always in a separate directory, no?

EricCousineau-TRI, 2018-02-08:
Closing, given that `bot_core/pose_t` is not presently used directly in the codebase (from doing a quick `grep`).

closed 2018-02-08 by EricCousineau-TRI