RobotLocomotion/drake #3353: Can no longer build Drake on Windows with VS 2015 32 bit toolchain

sherm1, 2016-09-06:
I updated my drake-distro clone, rebuilt, and got this message:
```
------ Build started: Project: drakeRBM, Configuration: Debug x64 ------
RigidBodyTree.cpp
RigidBodyTree.cpp(2651): fatal error C1060: compiler is out of heap space
```
Watching the compilation, the compiler had just hit 4GB address space when it died (this was a Debug build). 

Apparently Visual Studio uses 32 bit cross compilers by default even when installed on a 64 bit machine and building 64 bit binaries. There is allegedly a way to get it to use a 64 bit toolset instead; I will investigate. If so, we should modify the build system to force that (if possible) or at least update documentation.

sherm1, 2016-09-06:
This [stackoverflow article](http://stackoverflow.com/questions/19820718/how-to-make-visual-studio-use-the-native-amd64-toolchain) was helpful. The trick is to set an environment variable `PreferredToolArchitecture=x64` before running Visual Studio. Can CMake force this through the solution file it generates so we don't have to set that variable?

BTW, RigidBodyTree.cpp now needed almost 7GB of address space to compile (Debug). I think that is almost twice what it was using the last time I checked.

liangfok, 2016-09-06:
> BTW, RigidBodyTree.cpp now needed almost 7GB of address space to compile (Debug). I think that is almost twice what it was using the last time I checked.

Yeah, the bump from ~3.9GB to ~7GB of RAM happened between April 18, 2016 and April 21, 2016, see: https://github.com/RobotLocomotion/drake/issues/2074#issuecomment-244525072.

I have a feature branch where I've refactored the code to reduce RAM usage to ~5.1GB of RAM and have plans to decouple the parsers from the `drakeRBM` library, which I believe will further reduce memory consumption. I will be submitting PRs that bring the RAM utilization down soon.

david-german-tri, 2016-09-22:
Closing this as will-not-fix.

closed 2016-09-22 by david-german-tri