RobotLocomotion/drake #3361: Deprecate AddFloatingJoint() in Favor of AttachModelInstanceToTree()

liangfok, 2016-09-06:
Method `AddFloatingJoint()` was originally added because rigid bodies potentially belonging to multiple model instances were being added to the `RigidBodyTree` in a manner that was impossible to determine which bodies belong to which model instance (this is why it takes a list of body indices as an input parameter).

Now that we have the `model_instance_id` variable in `RigidBody`, this method should no longer be needed. Instead, a new method called `AttachModelInstanceToTree()` should be provided that identifies a model instance's root body and connects it to the `RigidBodyTree`.

sherm1, 2016-09-06:
I don't understand this. A model instance doesn't necessarily have a unique root. Didn't we already agree that there can be multiple disconnected trees in a model instance? And what do you mean by "attaching to a tree"? Doesn't this always attach a body to the world?

liangfok, 2016-09-18:
> A model instance doesn't necessarily have a unique root. Didn't we already agree that there can be multiple disconnected trees in a model instance?

Yes. The existing [AddFloatingJoint()](https://github.com/RobotLocomotion/drake/blob/ad24abbd7213876169e978fbb383d0561f99e199/drake/systems/plants/parser_common.h) works by adding the same joint to every one of the disconnected trees in a model instance. The new method should be more expressive in terms of being able to specify different types of joints for each disconnected tree.

>  Doesn't this always attach a body to the world?

No. I believe some of the disconnected trees can be attached to a branch belonging to another tree. For example, a disconnected tree representing Valkyrie's right hand may be attached to Valkeyrie's right arm.

liangfok, 2017-04-12:
@amcastro-tri, should this be added to the wish list in #5013?

amcastro-tri, 2017-04-12:
Model instances (or some other representation) is in the wishlist in #5013.

tri-ltyyu, 2019-03-22:
This is likely specific to RigidBodyPlant. If this is applicable to MultBodyPlant/SceneGraph, please file new issue. Closing for now.

closed 2019-03-22 by tri-ltyyu