RobotLocomotion/drake #3362: Output contact/collision information from RBP

amcastro-tri, 2016-09-06:
It would be useful to output the collision information and contact wrenches computed during `EvalTimeDerivatives` from `RigidBodyPlant`. 
What would the most convenient format of this output be? Maybe a structure containing the collision manifold plus contact wrench would be best? so that users of that information can request positions as well as forces for each contact. 

cc'ing @tkoolen and @sherm1.

tkoolen, 2016-09-06:
From the slack discussion: @ggould-tri rightly mentioned that in full generality, there could be a contact wrench between each pair of bodies in a mechanism. For now, all I personally need is wrenches acting between the bodies of the mechanism and the world (ground reaction wrenches).

So far, I like @jwnimmer-tri's  idea of having a separate output port per ground reaction wrench the best. Then `RigidBodyPlant` could have a std::map<const RigidBody*, SystemPortDescriptor<T>> member called `ground_reaction_wrench_ports_`, where each port is just a regular vector port. `map` lookups only need to happen when you connect the ports, not at every simulation time step.

tkoolen, 2016-09-06:
This is one of the big issues currently blocking a proper C++ Valkyrie/Atlas simulator that interfaces with the `InstantaneousQPController` and the OpenHumanoids stack.

tkoolen, 2016-09-06:
Some more thoughts:
* to address @ggould-tri's comment, perhaps we could have a `std::map<std::pair<const RigidBody*, const RigidBody*>, SystemPortDescriptor<T>>` instead, but we probably wouldn't want to fill all N^2 ports by default, N being the number of bodies in the mechanism.
* @amcastro-tri brought up the point of how rich the contact information should be. One wrench per body may not cut it in the long run; perhaps we'll need pressure distributions in the future. But for now, one contact wrench per body is good enough for me.

sherm1, 2016-09-06:
Sorry, late to the party. How are the contact forces to be consumed? Is this supposed to be something that can be wired abstractly between Systems, so must go through an output port, or does the consumer know it is talking to a RigidBodyPlant so can call an arbitrary method there provided for extracting this information?

tkoolen, 2016-09-06:
In this case, the consumer *could* know that it is talking to a RigidBodyPlant and call a method, but I think it shouldn't. That would defeat the purpose of having a port-based system.

sherm1, 2016-09-07:
I like the solution @amcastro-tri proposed initially (assuming I understood it correctly): an abstract-valued output port that delivers a `ContactResults` object containing metadata describing which contact surfaces are in contact, and the numerical values associated with the contacts. Any System that declares an input port expecting that type could connect.

In general a single body may have any number of contact surfaces, and individual surfaces may be extended meshes with detailed pressure distributions available. Even if RBPlant currently supports only much simpler contact, by returning a class now we can extend the returned object later to provide more elaborate results as they become available.

sherm1, 2016-09-07:
/cc @edrumwri

RussTedrake, 2016-09-08:
fyi - this is blocking the valkyrie sim.

sherm1, 2016-09-11:
After f2f with @tkoolen, @amcastro-tri is proceeding with a plan to add an output port that puts out a `ContactResults` object that will contain appropriate metadata (which contact surface pairs) and numerical data (per-pair contact wrenches). This will be extensible for later addition of more elaborate contact manifold or patch information.

RussTedrake, 2016-09-11:
I continue to think that the cleanest design would be to define a force sensor which consumes the generalized state (which would be the only output from rigid body plant), and outputs the contact forces.  With caching working behind the scenes to avoid duplicate computation.

RussTedrake, 2016-09-11:
but i do think that contact forces might deserve special treatment, and could be convinced that generalized state + contact results might be the right thing to output.  clearly related to https://github.com/RobotLocomotion/drake/issues/3325

tkoolen, 2016-09-11:
For the record, I'm still more in favor of having many small output ports created upon user request instead of a fat one containing `ContactResults` for reasons mentioned here: https://github.com/RobotLocomotion/drake/issues/3325#issuecomment-246087775. But a fat output port will do the job.

SeanCurtis-TRI, 2016-11-15:
#3821 merged.

closed 2016-11-15 by SeanCurtis-TRI