RobotLocomotion/drake #3365: Are the SpringMassSystemTest integrator tests obsolete?

david-german-tri, 2016-09-06:
Has Simulator 2.0 superseded some or all of the bulky tests in https://github.com/RobotLocomotion/drake/blob/master/drake/systems/framework/examples/test/spring_mass_system_test.cc?  If so, perhaps they should be deleted.

sherm1, 2016-09-06:
They aren't obsolete yet but will be as soon as the Simulator can provide semi-explicit and implicit integrators since those are needed to exercise the full API.

jwnimmer-tri, 2019-03-29:
I don't know where those tests ended up the past three years.

We now have `./multibody/plant/test/spring_mass_system_test.cc` which seems like it should live on, and `./systems/plants/spring_mass_system/test/spring_mass_system_test.cc` also is okay?  Assuming it's OK to close this for now.

closed 2019-03-29 by jwnimmer-tri