RobotLocomotion/drake #3369: runAtlasWalking example error

dengxianga, 2016-09-06:
Hello,
I am using the drake toolbox compiled from source code.
Without using snopt, I am getting errors when running the runAltasWalking.m example code. I am attaching the error message: 

Best,
Xiang
Warning: Snopt not found, using (slower) fmincon 
> In footstepPlanner.nonlinearCollocation (line 20)
  In footstepPlanner.alternatingMIQP (line 37)
  In Biped/planFootsteps (line 104)
  In Biped/runWalkingDemo (line 44)
  In runAtlasWalking (line 30) 
Warning: Cannot use sparse matrices with active-set algorithm: converting to full. 
> In fmincon (line 420)
  In footstepPlanner.nonlinearCollocation (line 191)
  In footstepPlanner.alternatingMIQP (line 37)
  In Biped/planFootsteps (line 104)
  In Biped/runWalkingDemo (line 44)
  In runAtlasWalking (line 30) 
____________________________________________________________
   DerivativeCheck Information

Objective function derivatives:
Maximum relative difference between user-supplied 
and finite-difference derivatives = 8.74281e-06.
  User-supplied derivative element (66,1):     -0.0153209
  Finite-difference derivative element (66,1): -0.0153122
____________________________________________________________

Error using validateFirstDerivatives (line 97)
DerivativeCheck failed:
User-supplied and forward finite-difference derivatives do not match within 1e-06 relative tolerance.

Error in fmincon (line 739)
    validateFirstDerivatives(funfcn,confcn,X, ...

Error in footstepPlanner.nonlinearCollocation (line 191)
  [xstar, fval, exitflag] = fmincon(@objfun, x0, sparse(A), b, ...

Error in footstepPlanner.alternatingMIQP (line 37)
  [plan, t] = footstepPlanner.nonlinearCollocation(obj, miqp_plan, weights, goal_pos);

Error in Biped/planFootsteps (line 104)
  [plan, solvertime] = options.method_handle(obj, plan, weights, goal_pos);

Error in Biped/runWalkingDemo (line 44)
footstep_plan = obj.planFootsteps(x0(1:nq), goal_pos, [], struct('step_params', walking_options));

Error in runAtlasWalking (line 30)
r.runWalkingDemo(walking_options);

samkhal, 2016-09-08:
Would love to hear from the team if there's a better way to do this, but here's what I've been doing:

You can use the student version of snopt.
Request the snoptmex file from here: https://ccom.ucsd.edu/~optimizers/downloads.php
You can use my wrapper: https://github.com/samkhal/snopt-matlab
Once you copy your snoptmex file into there and build it, you can add the directory to your matlab path and drake should be able to find it.

Also, here's how to fix nonlinearCollocation.m to work with snopt 7.5: https://github.com/samkhal/drake/commit/56c5c21b80c19f432c290b610dc7cd30ef176201

RussTedrake, 2016-09-08:
@rdeits -- perhaps we should decide whether we officially support fmincon under the hood for this or not?

soonho-tri, 2016-09-08:
@RussTedrake, @rdeits : We have three test failures from `examples/Atlas/runAtlasJumping` last seven days. The error messages are different from what @dengxianga reported. Here they are ([cdash link](https://drake-cdash.csail.mit.edu/testDetails.php?test=2486885&build=102046)):

```

Unexpected Standard exception from MEX file.
What() is:In BulletModel::findClosestPointsBetweenElements: No closest point found between 140587930465968 and 140587930495008
..

Error in RigidBodyManipulator/collisionDetect (line 69)
  [xA,xB,normal,distance,idxA,idxB] = collisionDetectmex(obj.mex_model_ptr, kinsol.mex_ptr, allow_multiple_contacts,active_collision_options);

Error in RigidBodyManipulator/contactConstraints (line 48)
[phi,normal,xA,xB,idxA,idxB] = collisionDetect(obj,kinsol_no_gradients,allow_multiple_contacts,active_collision_options);

Error in TimeSteppingRigidBodyManipulator/solveMexLCP (line 269)
        [phiC,normal,d,xA,xB,idxA,idxB,mu,n,D] = obj.manip.contactConstraints(kinsol, obj.multiple_contacts);

Error in TimeSteppingRigidBodyManipulator/solveLCP (line 289)
        [obj,z,Mvn,wvn] = solveMexLCP(obj,t,x,u);

Error in TimeSteppingRigidBodyManipulator/update (line 204)
        [obj,z,Mvn,wvn] = solveLCP(obj,t,x,u);

Error in SimulinkModelHandle/sim (line 112)
      varargout{:} = sim(obj.name,varargin{:});

Error in DynamicalSystem/simulate (line 120)
simout = sim(mdl,pstruct);

Error in runAtlasJumping (line 347)
traj = simulate(sys,[0 ts(end)+1],xtraj.eval(0));
Error using SimulinkModelHandle/sim (line 112)
Error reported by S-function 'DCSFunction' in 'MIMOCascade_3957874/system1/system2/system2/system2/system2/system2/system2/system2/system2/system2/system2/DrakeSys':

DrakeSystem S-Function: error MATLAB:unexpectedCPPexception in MATLAB callback.
See additional debugging information above

Error in DynamicalSystem/simulate (line 120)
simout = sim(mdl,pstruct);

Error in runAtlasJumping (line 347)
traj = simulate(sys,[0 ts(end)+1],xtraj.eval(0));
```

Please advise us if you think it's a good to disable this test temporarily.

RussTedrake, 2016-09-08:
The runAtlasJumping is a different problem (this probably isn’t the right thread for it).  That error message happens when the robot falls down (and the LCP fails to return a feasible solution).  Please open a separate issue and assign @hongkai-dai (who wrote that example initially).  Disabling is reasonable, but I’ll let Hongkai decide.

dengxianga, 2016-09-09:
Just followed @samkhal's suggestion. Unfortunately the robot falls...

Error using Biped/runWalkingDemo (line 57)
Center-of-mass during execution differs substantially from the plan.

Error in runAtlasWalking (line 30)
r.runWalkingDemo(walking_options);

dengxianga, 2016-09-16:
It looks like a lot of bugs have been fixed this week, after building from the most recent code, Altas no longer fails, the walking looks pretty cool! Thank you!

RussTedrake, 2016-09-16:
great news!  thanks to @tkoolen !

closed 2016-09-16 by RussTedrake

chang-liang, 2016-10-07:
@dengxianga Hello, did you have Snopt built during Drake building or just download separately by yourself?