RobotLocomotion/drake #3371: Can't build on Windows: No package 'bot2-lcmgl-client' found

sherm1, 2016-09-07:
After a merge with master for PR #3354 my Windows build fails with this message:
```
15>------ Build started: Project: drake, Configuration: Debug x64 ------
15>  Performing configure step for 'drake'
15>  -- CMAKE_INSTALL_PREFIX=C:/Users/Sherm/Documents/GitHub/sherm1/drake-distro-build_d/install
15>  -- MATLAB is disabled because the CMake option DISABLE_MATLAB is set to ON.
15>  Package bot2-lcmgl-client was not found in the pkg-config search path.
15>
15>  Perhaps you should add the directory containing `bot2-lcmgl-client.pc'
15>
15>  to the PKG_CONFIG_PATH environment variable
15>
15>  No package 'bot2-lcmgl-client' found
15>    --------------------------------------------------------------------------------
15>      *** IMPORTANT: bot2-lcmgl build directory not found. ***
15>    --------------------------------------------------------------------------------
15>
```

Help! Wild guess: @jamiesnape ?

sherm1, 2016-09-07:
Apologies -- this was a side effect of unrelated merge damage; my bad -- please ignore.

closed 2016-09-07 by sherm1

jwnimmer-tri, 2016-09-07:
The take-away for me is a build system that outputs loud warnings in the default configuration is a build system that will have lots of false-positive bug reports.

sherm1, 2016-09-07:
Yup. So much crap came out during the build I missed the line or two that included the actual problem. Luckily the gcc CI build was cleaner and I saw the real error there.

jamiesnape, 2016-09-07:
Using CDash is good for things like this:

https://drake-cdash.csail.mit.edu/index.php?project=drake-superbuild&showfilters=1&filtercount=2&showfilters=1&filtercombine=and&field1=label&compare1=61&value1=jenkins-windows-msvc-ninja-32-experimental-3354-debug-4370&field2=buildstarttime&compare2=84&value2=now