RobotLocomotion/drake #3379: MATLAB JVM detection logic can produce spurious warnings in ccmake

david-german-tri, 2016-09-07:
Steps to reproduce:

1. Clean checkout of drake on a system where the JDK version mismatches the MATLAB JVM version.
2. `mkdir build && cd build && ccmake ..`
3.  `c` to configure
4. When presented with a warning about mismatched JVM version, `e` to exit and `q` to quit.
5. `ccmake ..` again
6. `c` to configure

Expected behavior: Another warning about JDK version mismatch.
Actual behavior: `Could not determine MATLAB JVM version because MATLAB exited with nonzero result.`

Presumably something is being scribbled on the CMakeCache.txt that accounts for the change.  Investigating now.  In the meantime, it's perfectly OK to press `g` to generate and exit after configuring.  The build will work fine; the warning is spurious.

david-german-tri, 2016-09-07:
@siyuanfeng-tri @Lucy-tri FYI

jamiesnape, 2016-09-07:
I see the issue. Patch will be up shortly.

david-german-tri, 2016-09-07:
Yeah, looks like a nested-conditional thing.

jamiesnape, 2016-09-07:
(Would cause the warning with repeated `cmake` calls, not just `ccmake`, though it would be less obvious.)

PR #3380 (2016-09-07): Fix spurious MATLAB JVM warning
Fixes #3379.

FYI @Lucy-tri @siyuanfeng-tri .

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/3380)
closed 2016-09-07 by jamiesnape