RobotLocomotion/drake #3382: Mysterious CI Failure - Certain Text in PR's Description Reults in Failure to Identify Matlab Root

liangfok, 2016-09-08:
In #3358, I encountered the following [mysterious pre-merge CI failure](https://drake-jenkins.csail.mit.edu/job/experimental/4407/compiler=gcc,label=linux_matlab,type=release/) in configuration `gcc_linux_matlab`:

```
CMake Error at /home/ubuntu/workspace/experimental/c5b85b52/cmake/mex.cmake:102 (string):
  string sub-command STRIP requires two arguments.
Call Stack (most recent call first):
  CMakeLists.txt:143 (mex_setup)

SWIG_EXECUTABLE: /home/ubuntu/workspace/experimental/c5b85b52/build/install/bin/swig
Checking if Swig has Matlab support... yes
Found Swig with Matlab support
CMake Error at /home/ubuntu/workspace/experimental/c5b85b52/build/install/share/cmake/SwigMatlab.cmake:43 (message):
  Please run 'mex_setup(REQUIRED)' which is provided by mex.cmake in
  https://github.com/RobotLocomotion/cmake before using this macro.
Call Stack (most recent call first):
  bindings/swig/CMakeLists.txt:55 (add_swig_matlab_module)
```

To debug this, I [instrumented](https://github.com/RobotLocomotion/drake/pull/3358/commits/b767d4ba06e97a3b98a9b3ef27e8e9c4ca428bfd) `mex.cmake` with a debug print statement. This resulted in the following being logged:

```
18:35:00 -- ************************** MATLAB ROOT: _matlab_root=This change was shown to decrease build times by
```

Since `_matlab_root` appeared to get part of the PR's description, I removed the following text from the description:

> Using an HP z640 workstation with an Intel(R) Xeon(R) CPU E5-2620 v3 @ 2.40GHz, 40GB of Buffered DDR4 1866 MHz RAM, and commit 6d1f32d1c351e4c646009c3ee6e3e906bb2a4395, the build time and memory footprint without dReal is ~14 minutes and 5.1GB:
> 
> ```
> $ cd drake-distro
> $ rm -rf build; externals
> $ git reset --hard HEAD
> $ mkdir build; cd build
> $ cmake .. -DCMAKE_BUILD_TYPE:STRING=RelWithDebInfo -DDISABLE_MATLAB=TRUE -DWITH_DREAL=FALSE -DWITH_SNOPT=FALSE 
> $ /usr/bin/time --format=%E,%U,%M make -j12
> 13:55.23,4906.59,5097528
> ```

This resulted in the PR's pre-merge CI tests to pass: https://drake-jenkins.csail.mit.edu/job/experimental/4412/.

Thus, I believe there is a bug in `mex.cmake` that arises when the PR's description contains certain text.

david-german-tri, 2016-09-08:
Congratulations, this is the most surprising bug report of the month.

@mwoehlke-kitware, @BetsyMcPhail, @jamiesnape any ideas?

jamiesnape, 2016-09-08:
I will look into it.

mwoehlke-kitware, 2016-09-08:
Why does `mex.cmake` look at the PR description? Or indeed have any concept of PR's? I suspect the problem is in the CI scripts...

jamiesnape, 2016-09-08:
Among other things, Jenkins sets various environment variables, one of which contains the PR description.

jamiesnape, 2016-09-08:
Fail: https://drake-jenkins.csail.mit.edu/job/experimental/4407/compiler=gcc,label=linux_matlab,type=release/injectedEnvVars/
Pass: https://drake-jenkins.csail.mit.edu/job/experimental/4412/injectedEnvVars/

mwoehlke-kitware, 2016-09-08:
Yes, but why would `mex.cmake` look at a PR description?

Oh, I see the problem... looking at that error, it looks like `matlab -e | grep -e MATLAB= | cut -d'=' -f2` went sideways... and the PR has `...STRING=RelWithDebInfo -DDISABLE_MATLAB=TRUE -DWITH_DREAL..`. The `grep` needs to be `-E -e '^MATLAB=`.

PR #3391 (2016-09-12): Use more logic from upstream FindMatlab in mex_setup()
Fixes #3382. Removes any use of `grep` which really was not necessary anyway.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/3391)
commit fca35118 (2016-09-12): Use more logic from upstream FindMatlab in mex_setup()

Fixes #3382.
closed 2016-09-12 by david-german-tri