RobotLocomotion/drake #3383: runAtlasWalking MEX exception

samkhal, 2016-09-08:
I used to be able to run everything fine, but recent versions have been giving me problems with Atlas code. The only relevant changes I've made to drake are ones to get snopt working (https://github.com/RobotLocomotion/drake/issues/3369#issuecomment-245497973)

From runAtlasWalking, I'm getting the following error:

```
Unexpected Standard exception from MEX file.
What() is:don't know the variable name
..
Error in qpSpline (line 27)
  [coefs, ts, objval] = nWaypointCubicSplineFreeKnotTimesmex(ts(1), ts(end), xs, xd0, xdf);
Error in Biped/planSwingPitched (line 206)
[coefs, ts] = qpSpline(ts,...
Error in Biped/planZMPTraj (line 104)
  [new_foot_knots, new_zmp_knots] = planSwingPitched(biped, st, sw0, sw1, initial_hold, target_frame_id);
Error in QPLocomotionPlanSettings.fromBipedFootstepPlan (line 200)
      [zmp_knots, foot_motion_data] = biped.planZMPTraj(x0(1:biped.getNumPositions()), footstep_plan.footsteps, zmp_options);
Error in Biped/planWalkingZMP (line 7)
settings = QPLocomotionPlanSettings.fromBipedFootstepPlan(footstep_plan, obj, x0);
Error in Biped/runWalkingDemo (line 50)
walking_plan_data = obj.planWalkingZMP(x0(1:obj.getNumPositions()), footstep_plan);
Error in runAtlasWalking (line 30)
r.runWalkingDemo(walking_options); 
```

or 

```
Unexpected Standard exception from MEX file.
What() is:times must be increasing
..
Error in qpSpline (line 30)
    [coefs, objval] = nWaypointCubicSplinemex(ts, xs, xd0, xdf);
Error in Biped/planSwingPitched (line 206)
[coefs, ts] = qpSpline(ts,...
Error in Biped/planZMPTraj (line 104)
  [new_foot_knots, new_zmp_knots] = planSwingPitched(biped, st, sw0, sw1, initial_hold,
  target_frame_id);
Error in QPLocomotionPlanSettings.fromBipedFootstepPlan (line 200)
      [zmp_knots, foot_motion_data] = biped.planZMPTraj(x0(1:biped.getNumPositions()),
      footstep_plan.footsteps, zmp_options);
Error in Biped/planWalkingZMP (line 7)
settings = QPLocomotionPlanSettings.fromBipedFootstepPlan(footstep_plan, obj, x0);
Error in Biped/runWalkingDemo (line 50)
walking_plan_data = obj.planWalkingZMP(x0(1:obj.getNumPositions()), footstep_plan);
Error in runAtlasWalking (line 30)
r.runWalkingDemo(walking_options); 
```

I haven't tried debugging the mex but I'd guess the problem is related to the value of ts ([1.8500         0         0    3.2393]) which should be increasing monotonically but isn't.

liangfok, 2016-09-08:
Can you provide a list of commands that I can execute to replicate this problem?

I'm making a wild guess but maybe the non-monotonically-increasing time problem is related to our [switch from an RK1 to a RK2 integrator](https://github.com/RobotLocomotion/drake/commit/b939dbd1e35e76582dd46f2898edb573b42773ec)?

What was the SHA of the most recently working version?

RussTedrake, 2016-09-08:
unfortunately, i cannot test this due to the segfault issues.  that test is currently disabled.  :(

samkhal, 2016-09-08:
> Can you provide a list of commands that I can execute to replicate this problem?

I was just calling `runAtlasWalking` within MATLAB; I'm not sure how to help you reproduce it.

> What was the SHA of the most recently working version?

Unfortunately this isn't likely to help either, since I haven't been keeping up to date; I just moved up from https://github.com/RobotLocomotion/drake/commit/26a9a6e5531e2fb46fbfb249f2e6b730f0143bad from half a year ago. I'll try to build a version before/after the integrator switch and see how it goes.

It seems like there's currently some instability with some of the atlas demos (https://github.com/RobotLocomotion/drake/issues/3369#issuecomment-245616179); could you recommend a recent hash where runAtlasJumping and the other Atlas demos are rock solid?

liangfok, 2016-09-09:
Unfortunately, I'm unable to pinpoint a more recent SHA where `runAtlasJumping` and other Atlas demos were rock solid. Are they rock solid when you switch back to https://github.com/RobotLocomotion/drake/commit/26a9a6e5531e2fb46fbfb249f2e6b730f0143bad?

liangfok, 2016-09-09:
Also, which version of MATLAB are you using?

samkhal, 2016-09-09:
I'm running 2015a. runAtlasWalking was solid back in https://github.com/RobotLocomotion/drake/commit/26a9a6e5531e2fb46fbfb249f2e6b730f0143bad, but runAtlasJumping was broken.

Unfortunately I'm having a very rough time trying to get any version to run Atlas rock solid; I've got four snapshots of drake built, each of which is giving me it's own unique errors when I try to run Atlas demos. Even my old version from https://github.com/RobotLocomotion/drake/commit/26a9a6e5531e2fb46fbfb249f2e6b730f0143bad gave me errors that I had to resolve and is now running very slowly, likely because I'm compiling it with newer versions of the externals. So I'm having trouble finding a "stable" version that I can stick with for the next half-year.

RussTedrake, 2016-09-09:
fwiw -- the reason that we are not prioritizing a fix is because the backbone of these tests is being replaced by the c++ versions which will be much more stable.  i believe that will be happening very soon.

closed 2016-09-09 by RussTedrake

hongkai-dai, 2016-10-16:
FWIW, I received an email from Bryan Hsu, saying that he encountered the same problem.
```
Hi Hongkai,

Matlab paused at qpSpline.m Line 27, when I used 

dbclear all
dbstop if error

And matlab compained like this,

>> dbclear all
>> dbstop if error
>> runAtlasWalking

ans =

     1

Unexpected Standard exception from MEX file.
What() is:don't know the variable name
..

Error in qpSpline (line 27)
  [coefs, ts, objval] =
  nWaypointCubicSplineFreeKnotTimesmex(ts(1), ts(end),
  xs, xd0, xdf);

Error in Biped/planSwingPitched (line 206)
[coefs, ts] = qpSpline(ts,...

Error in Biped/planZMPTraj (line 104)
  [new_foot_knots, new_zmp_knots] =
  planSwingPitched(biped, st, sw0, sw1, initial_hold,
  target_frame_id);

Error in
QPLocomotionPlanSettings.fromBipedFootstepPlan (line
200)
      [zmp_knots, foot_motion_data] =
      biped.planZMPTraj(x0(1:biped.getNumPositions()),
      footstep_plan.footsteps, zmp_options);

Error in Biped/planWalkingZMP (line 7)
settings =
QPLocomotionPlanSettings.fromBipedFootstepPlan(footstep_plan,
obj, x0);

Error in Biped/runWalkingDemo (line 50)
walking_plan_data =
obj.planWalkingZMP(x0(1:obj.getNumPositions()),
footstep_plan);

Error in runAtlasWalking (line 30)
r.runWalkingDemo(walking_options);

And ts, xs, xd0 and xdf values are,

K>> settings

settings = 

    optimize_knot_times: 1
                use_mex: 1

K>> ts

ts =

    0.3500         0         0    2.8511

K>> xd0

xd0 =

     0
     0
     0
     0
     0
     0

K>> xdf

xdf =

     0
     0
     0
     0
     0
     0

K>> xs

xs =

    0.7309    0.8208    1.2402    1.3301
    0.1453    0.1438    0.1367    0.1352
    0.0000    0.0500    0.0500    0.0000
         0   -0.0001         0         0
         0    0.3927         0         0
    0.0003    0.0003    0.0001    0.0001

K>> 

Best,

Bryan
```
I think `ts` is not the problem. I can run this test on my machine, using the most up-to-date drake. On my machine `ts` is `ts = [1.8500,         0,         0,    3.2338]` , it is not in increasing order either.

samkhal, 2016-10-17:
Thanks for bringing this up again! I think I have it figured it out; it's the same issue I had earlier with footstep planning (https://github.com/RobotLocomotion/drake/issues/3737#issuecomment-252703438). It appears that the root of it is caused by my use of the student version of SNOPT 7.5 vs the full SNOPT 7.2. If SNOPT is used for alternatingMIQP for footstep planning, then the footstep plan differs between the two versions.

There are two simple solutions to the above error that could help Bryan:
1. Disable SNOPT and let alternatingMIQP fall back to fmincon. This will cause it to fail a derivative check, so for the fmincon call in nonlinearCollocation.m, `DerivativeCheck` must be set to `off` on [this line](https://github.com/RobotLocomotion/drake/blob/last_sha_with_windows_support/drake/systems/robotInterfaces/+footstepPlanner/nonlinearCollocation.m#L191).
2. Alternatively, instead of using alternatingMIQP for footstep planning, use humanoids2014. So instead of [this line](https://github.com/RobotLocomotion/drake/blob/last_sha_with_windows_support/drake/systems/robotInterfaces/@Biped/runWalkingDemo.m#L44), use this one:

    footstep_plan = obj.planFootsteps(x0(1:nq), goal_pos, [], struct('method_handle',@footstepPlanner.humanoids2014,'step_params', walking_options));

So, why does this happen? I found this cause when I noticed that alternatingMIQP was returning non-turning plans when given a rotated goal; in a straight line, my version gives a footstep plan that looks feasible visually. If somebody wants I can upload the "broken" footstep plan I get with alternatingMIQP, so you could compare it to what it's supposed to output. Interestingly, when I was running a version of Drake from last Feb, it was able to execute the "broken" footstep plan just fine, instead of getting the error above, but changes since then (to planWalkingZMP or futher down) cause it to fail.

Fixing this: it could be that SNOPT 7.5 trial and 7.2 full are giving different results, and it's also possible I'm just them wrong. SNOPT 7.5 changed the function interfaces, so I made a change that I posted here a long time ago: https://github.com/RobotLocomotion/drake/issues/1696. That might be causing the issue, but I'd have to dig deeper to figure out why, and I don't have access to 7.2 to compare.