RobotLocomotion/drake #3388: Frequent failure to upload file on msvc win32 build

tkoolen, 2016-09-08:
I think this has happened on nearly half of the builds I've triggered in the last week:
```
12:16:25    Error when uploading file: C:/Users/Administrator/workspace/experimental/0b4ef046/drake/build/Testing/20160908-1530/Update.xml
12:16:25    Error message was: The requested URL returned error: 504 Gateway Time-out
12:16:25    Problems when submitting via HTTP
```
Most recently, on https://github.com/RobotLocomotion/drake/pull/3378, [build results](https://drake-jenkins.csail.mit.edu/job/experimental/4425/compiler=msvc-ninja-32,label=windows,type=debug/console).

david-german-tri, 2016-09-08:
See previous discussion on #2791 (which in turn has pointers to other instances of the same issue).

From my 2016-07-13 TRI/Kitware sync notes, I have:

- CDash returning 500 occasionally is a known, unexplained issue.
- No convenient way to make CTest exit 0 even when this failure happens.
- This has been rare; for now, we'll just work around with override merge.

So, I'll go ahead and override-merge #3378.  I'll leave it to @jamiesnape and @BetsyMcPhail to follow up on whether this can be fixed with upstream changes to CTest and/or CDash.

jamiesnape, 2016-09-08:
@billhoffman is working on adding the necessary options to ctest upstream.

jamiesnape, 2016-09-15:
Unfortunately, nothing I can do to fix the network issues. The `ctest` return value issue is captured as #2817. Therefore, closing this as a duplicate of that issue.

closed 2016-09-15 by jamiesnape