RobotLocomotion/drake #3389: runAtlasJumping fails in the simulation

hongkai-dai, 2016-09-08:
I am copying this from #3369 
```
Unexpected Standard exception from MEX file.
What() is:In BulletModel::findClosestPointsBetweenElements: No closest point found between 140587930465968 and 140587930495008
..

Error in RigidBodyManipulator/collisionDetect (line 69)
  [xA,xB,normal,distance,idxA,idxB] = collisionDetectmex(obj.mex_model_ptr, kinsol.mex_ptr, allow_multiple_contacts,active_collision_options);

Error in RigidBodyManipulator/contactConstraints (line 48)
[phi,normal,xA,xB,idxA,idxB] = collisionDetect(obj,kinsol_no_gradients,allow_multiple_contacts,active_collision_options);

Error in TimeSteppingRigidBodyManipulator/solveMexLCP (line 269)
        [phiC,normal,d,xA,xB,idxA,idxB,mu,n,D] = obj.manip.contactConstraints(kinsol, obj.multiple_contacts);

Error in TimeSteppingRigidBodyManipulator/solveLCP (line 289)
        [obj,z,Mvn,wvn] = solveMexLCP(obj,t,x,u);

Error in TimeSteppingRigidBodyManipulator/update (line 204)
        [obj,z,Mvn,wvn] = solveLCP(obj,t,x,u);

Error in SimulinkModelHandle/sim (line 112)
      varargout{:} = sim(obj.name,varargin{:});

Error in DynamicalSystem/simulate (line 120)
simout = sim(mdl,pstruct);

Error in runAtlasJumping (line 347)
traj = simulate(sys,[0 ts(end)+1],xtraj.eval(0));
Error using SimulinkModelHandle/sim (line 112)
Error reported by S-function 'DCSFunction' in 'MIMOCascade_3957874/system1/system2/system2/system2/system2/system2/system2/system2/system2/system2/system2/DrakeSys':

DrakeSystem S-Function: error MATLAB:unexpectedCPPexception in MATLAB callback.
See additional debugging information above

Error in DynamicalSystem/simulate (line 120)
simout = sim(mdl,pstruct);

Error in runAtlasJumping (line 347)
traj = simulate(sys,[0 ts(end)+1],xtraj.eval(0));
```

tkoolen, 2016-09-08:
In my experience, this error often means that there are NaNs in q and/or v, or that a vector of all zeros is used as the configuration vector for a quaternion-parameterized robot (norm-zero quaternion).

jwnimmer-tri, 2017-06-19:
Closing as won't-fix, because #6369 will be removing the relevant software.

If I'm confused and this should in fact stay open, please re-open an clarify the remaining unsolved problem(s) in a new comment here.

closed 2017-06-19 by jwnimmer-tri