RobotLocomotion/drake #3397: ImportError: liblcm.so.1: cannot open shared object file: No such file or directory

jwnimmer-tri, 2016-09-09:
The `steering_command_driver.py` is broken (as perhaps are other python users of LCM?):
```
jwnimmer@call-cc:~/jwnimmer-tri/drake-distro/drake/examples/Cars$ env PYTHONPATH=../../../build/debug/install/lib/python2.7/site-packages:$PYTHONPATH ./steering_command_driver.py 
Traceback (most recent call last):
  File "./steering_command_driver.py", line 32, in <module>
    import lcm
  File "/home/jwnimmer/jwnimmer-tri/drake-distro/build/debug/install/lib/python2.7/site-packages/lcm/__init__.py", line 3, in <module>
    from ._lcm import LCM, LCMSubscription
ImportError: liblcm.so.1: cannot open shared object file: No such file or directory
```

jwnimmer-tri, 2016-09-09:
This used to work.  I wonder if an `-rpath` got dropped in some build system update?

I am working on root causing and understanding this (as part of a wider PR to fix up the Cars demos to work again at all ever), but though I'd post in case a wider audience was curious.

jwnimmer-tri, 2016-09-09:
```
readelf -d install/lib/python2.7/site-packages/lcm/_lcm.so 

Dynamic section at offset 0x1ad98 contains 26 entries:
  Tag        Type                         Name/Value
 0x0000000000000001 (NEEDED)             Shared library: [libglib-2.0.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libpthread.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libc.so.6]
```

jwnimmer-tri, 2016-09-09:
Working `_lcm.so` from back in time:
```
readelf -d install/lib/python2.7/site-packages/lcm/_lcm.so 

Dynamic section at offset 0x1ad98 contains 26 entries:
  Tag        Type                         Name/Value
 0x0000000000000001 (NEEDED)             Shared library: [libglib-2.0.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libpthread.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libc.so.6]
```

Broken `_lcm.so` from head:
```
readelf -d install/lib/python2.7/site-packages/lcm/_lcm.so

Dynamic section at offset 0x4da0 contains 26 entries:
  Tag        Type                         Name/Value
 0x0000000000000001 (NEEDED)             Shared library: [liblcm.so.1]
 0x0000000000000001 (NEEDED)             Shared library: [libpython2.7.so.1.0]
 0x0000000000000001 (NEEDED)             Shared library: [libc.so.6]

jwnimmer-tri, 2016-09-09:
Unsurprisingly, git bisect concludes that #2582 / 6fcf875cc91308fa3b26549571c6c0c2eef43fe2 is what broke this:
```
jwnimmer@call-cc:~/tmp/bisect/drake-distro$ git bisect bad
6fcf875cc91308fa3b26549571c6c0c2eef43fe2 is the first bad commit
```

jwnimmer-tri, 2016-09-09:
So I guess what I don't know is, is python code that wants to `import lcm` supposed to set a library path first now?  Or is something like liblcm_pic.a supposed to be included in `_lcm.so` and the CMake variant of the LCM build system broken?  Or is there an `rpath` missing somewhere?  Or ...?

jwnimmer-tri, 2016-09-09:
@mwoehlke-kitware do you have any thoughts?

jwnimmer-tri, 2016-09-09:
FWIW, the "`import lcm` doesn't work" work-around, which I will have in a PR fairly soon:
```
try:
    import lcm
except ImportError as e:
    if ('cannot open shared object file' in e.message and
        '--no-retry' not in sys.argv):
        print 'warning:', e.message, '(will retry now)'
        LD_LIBRARY_PATH = ':'.join([
            os.path.join(DRAKE_DIST_BUILD_DIR, 'install/lib'),
            os.getenv('LD_LIBRARY_PATH', default='')])
        new_env = dict(os.environ)
        new_env.update(LD_LIBRARY_PATH = LD_LIBRARY_PATH)
        sys.exit(subprocess.call(
            [sys.executable] + sys.argv + ['--no-retry'],
            env=new_env))
    raise
```

jwnimmer-tri, 2016-09-12:
Note that whatever solution we come up with here should remove the TODO hacks in #3403.

mwoehlke-kitware, 2016-09-13:
> is python code that wants to import lcm from other than a system directory supposed to set a library path first now

IMHO, yes. Probably the drake install prefix should be in `[DY]LD_LIBRARY_PATH` anyway; who knows what other libraries might be needed and can't be loaded, otherwise. (I suspect this used to work because LCM was being built static; likely it would work again if built static.)

jwnimmer-tri, 2016-09-13:
What is the "drake install prefix"?  It is `drake-distro/build/install/lib" do you mean"?  Or `drake-distro/build/drake/lib`?

mwoehlke-kitware, 2016-09-13:
The first one: `<drake-distro>/build/install/lib<lib-suffix>`. "Install prefix" is normally synonymous with `${CMAKE_INSTALL_PREFIX}`.

 (Note that drake currently neither respects nor propagates (to externals) `${LIB_SUFFIX}`. It *should* do both, but even then I'm probably the only one actually setting it. Just a word of warning though that at some point you may need to add e.g. `lib64` either instead of, or possibly in addition to, `lib`.)

jwnimmer-tri, 2016-09-13:
> > is python code that wants to import lcm from other than a system directory supposed to set a library path first now
> IMHO, yes. Probably the drake install prefix should be in [DY]LD_LIBRARY_PATH anyway; who knows what other libraries might be needed and can't be loaded, otherwise.

All of my Drake programs apparently have a an `rpath` set already, so they don't need this setting (line breaks added for readability):
```
jwnimmer@call-cc:~/jwnimmer-tri/drake-distro/build/debug/drake/bin$
readelf -d rigidBodyLCMNode
...
Library rpath: [
  /home/jwnimmer/jwnimmer-tri/drake-distro/build/debug/drake/lib:
  /home/jwnimmer/jwnimmer-tri/drake-distro/build/debug/install/lib:]
```

Having to set the LD_LIBRARY_PATH just to use some python imports seems like an undesirable burnden for our users.

> (I suspect this used to work because LCM was being built static; likely it would work again if built static.)

Why was this changed?  Should we change it back?

mwoehlke-kitware, 2016-09-13:
> All of my Drake programs apparently have a an rpath set already [...]

Only if you run them from the *build* directory. Try looking at the installed binaries...

That said, `_lcm.so` should be the same, i.e. if you point `PYTHONPATH` to `<build>/externals/lcm/lib/python2.7/site-packages/lcm/_lcm.so` instead of `<install>/lib/python2.7/site-packages/lcm/_lcm.so`, that will probably work also.

> > I suspect this used to work because LCM was being built static
>
>  Why was this changed?

Probably because, by default, LCM doesn't build a PIC static library. Forcing it to build shared was just as easy a fix, and I dislike static libraries anyway for a variety of reasons.

(That said, on further thought, I'd prefer to pass in instead the flag to enable PIC and inherit static-vs-shared from the superbuild, as is the case for most other externals.)

> Should we change it back?

It wouldn't be *my* first preference (see previous comment). There really isn't a "bug" here besides "you're doing it wrong" (although granting it is convenient to not have to set `[DY]LD_LIBRARY_PATH`).

My general philosophy is, if you are going to use installed artifacts, you¹ are responsible for setting `PATH`, `[DY]LD_LIBRARY_PATH`, `PYTHONPATH`, etc.. If you don't want to do that, use the build directory artifacts instead.

(¹ Convenience scripts to set up the environment are also acceptable; "you" here really means "*someone*". It might not be a bad idea to create such a script?)

Ultimately, it's a choice between "convenient" and "correct". *I* always prefer "correct", but YMMV.

jwnimmer-tri, 2016-09-13:
> > All of my Drake programs apparently have a an rpath set already [...]
> Only if you run them from the *build* directory. Try looking at the installed binaries...

Installed binaries still have an rpath:
```
jwnimmer@call-cc:~/jwnimmer-tri/drake-distro/build$
readelf -d ./debug/install/bin/rigidBodyLCMNode
...
Library rpath: [/home/jwnimmer/jwnimmer-tri/drake-distro/build/debug/install/lib]
```

> That said, `_lcm.so` should be the same, i.e. if you point `PYTHONPATH` to `<build>/externals/lcm/lib/python2.7/site-packages/lcm/_lcm.so` instead of `<install>/lib/python2.7/site-packages/lcm/_lcm.so`, that will probably work also.

I don't understand how that is relevant, or helps.  The python script always loads `_lcm.so` successfully; but post-#2582 that library requires `liblcm.so.1`, which is not found by the loader.

Here is what files I have:
```
jwnimmer@call-cc:~/jwnimmer-tri/drake-distro/build/debug$ \
  find . -name liblcm.so.1 -o -name _lcm.so | sort
./externals/lcm/lcm/liblcm.so.1
./externals/lcm/lib/python2.7/site-packages/lcm/_lcm.so
./install/lib/liblcm.so.1
./install/lib/python2.7/site-packages/lcm/_lcm.so
```

I think, though, what you're saying is code that does `import lcm` from within a Drake build has to add an entry to `LD_LIBRARY_PATH` that is congruent with their additions to `PYTHONPATH`.  Is that right?

If so, that means we can never tell users to just run a python program (unless we keep the #3403 hack around).  They either have to run a `set_environment.sh` script first, or munge paths by hand.  That seems unfortunate, given that it used to work just fine a couple months ago.

> My general philosophy is, if you are going to use installed artifacts, you¹ are responsible for setting `PATH`, `[DY]LD_LIBRARY_PATH`, `PYTHONPATH`, etc.. If you don't want to do that, use the build directory artifacts instead.

The build directory artifacts don't work either, do they?  They still would fail to load `liblcm.so.1` unless the `LD_LIBRARY_PATH` was munged?

mwoehlke-kitware, 2016-09-13:
> Installed binaries still have an rpath.

Yike. It looks like drake, in a somewhat nasty way, is [monkeying with RPATH](https://github.com/RobotLocomotion/drake/blob/3b2f5a02fac45302e06d0276c8a0b133d555f795/cmake/pods.cmake#L677).

(There are some legitimate reasons to do this, but the way drake is doing it ­— at least, the *function* that is doing it — is a bit scary. Most projects, however, including LCM, don't touch any of the RPATH stuff and just use the default behavior, which is to strip RPATH from installed artifacts.)

> The build directory artifacts don't work either, do they? They still would fail to load `liblcm.so.1` unless the `LD_LIBRARY_PATH` was munged?

No, the `_lcm.so` from the build directory should have an RPATH. It does on my machine; `ldd` of `_lcm.so` from the install directory can't find `liblcm.so.1`, but the one in the build directory does find it.

> I think, though, what you're saying is code that does `import lcm` from within a Drake build has to add an entry to `LD_LIBRARY_PATH` that is congruent with their additions to `PYTHONPATH`. Is that right?

Yes. This is "normal". *In general* (i.e. for any random package, not necessarily drake), adding a non-standard prefix to `PYTHONPATH` implies needing to add the same prefix to `[DY]LD_LIBRARY_PATH`, if whatever you're trying to use uses non-Python-module shared libraries.

> If so, that means we can never tell users to just run a python program [...]

They already have to set `PYTHONPATH` though, don't they? It seems that having a script would be easier than setting `PYTHONPATH` by hand...

Alternatively, maybe LCM's `__init.py` could be modified with something like #3403. (Or we can force it to be built static, though that has additonal drawbacks.) My concern, however, is that all alternatives to setting `[DY]LD_LIBRARY_PATH` are work-arounds. They may work for specific packages, but there is always the chance of some other package coming along and needing a new work-around. Setting `[DY]LD_LIBRARY_PATH` is TRTTD.

(An alternative I *might* be comfortable with would be to push RPATH stuff into *all* of the externals. I'd want to get opinions from some folks with more experience fiddling with install RPATH's on doing so, however...)

jwnimmer-tri, 2016-09-13:
> No, the _lcm.so from the build directory should have an RPATH.

Oh, indeed it does.  Sorry!
I guess I will explore this option.

> > If so, that means we can never tell users to just run a python program [...]
> They already have to set `PYTHONPATH` though, don't they?

Nope, because we can fix `PYTHONPATH` automatically for them:
https://github.com/RobotLocomotion/drake/blob/4c18e9b926a35fb1540527ad8f4531e094f1ea61/drake/examples/Cars/steering_command_driver.py#L29

Fixing `LD_LIBRARY_PATH` is more difficult:
https://github.com/RobotLocomotion/drake/blob/4c18e9b926a35fb1540527ad8f4531e094f1ea61/drake/examples/Cars/steering_command_driver.py#L37

-- 

In general, I still don't understand why it was desirable for the CMake rewrite of LCM's build system to change how `_lcm.so` came out.  Why it is better to have two shared libraries and associated path questions, instead of just the one (`_lcm.so`)?

mwoehlke-kitware, 2016-09-13:
> Why it is better to have two shared libraries[...]?

Why are shared libraries used, ever? For one, they reduce code duplication (i.e. space used on disk). IIRC, 'the old way' was also building some objects (sources) twice.

jwnimmer-tri, 2016-09-13:
Anyway, it sounds like upstream LCM isn't broken, just different, so I can take it from here in terms of fixing up my demo scripts.

mwoehlke-kitware, 2016-09-14:
TBH, I wouldn't mind @ashuang's opinion on the matter :smile:.

@ashuang, the setuptools build of the LCM Python module recompiles the LCM C sources, which duplicates object code but results in a stand-alone `.so`. The CMake build instead links to `liblcm.so`, potentially requiring that `[DY]LD_LIBRARY_PATH` is set in addition to `PYTHONPATH`. Do you think that's okay, or should we (i.e. LCM's CMake build) either go back to duplicating the object code or provide some other work-around for this?

ashuang, 2016-09-14:
Hi @mwoehlke-kitware . Yes, the setuptools build of LCM Python was intentional in that regard. I thought it was more desirable to have a standalone `_lcm.so` and to not have to worry about library paths and/or environment variables.  My recommendation would be to go back to duplicating the object code so that the python module can be a standalone `.so`. Disk space is cheap these days, and the duplicated object code is just a few hundred kb I think.

jwnimmer-tri, 2016-09-21:
I assume the next step here is an lcm-proj issue to request this feature, or else directly to a pull request that puts back the old behavior.  Does that sound correct @mwoehlke-kitware?

mwoehlke-kitware, 2016-09-22:
For the moment, see https://groups.google.com/forum/#!topic/lcm-dev/XKu7u7jpuck. If there are no objections, I'll open an issue.

mwoehlke-kitware, 2016-09-28:
> I assume the next step here is an lcm-proj issue to request this feature, or else directly to a pull request that puts back the old behavior.

lcm-proj/lcm#110

jwnimmer-tri, 2016-10-05:
It looks like perhaps LCM upstream is repaired now, and we could resolve this issue by bumping our hash, is that correct?

mwoehlke-kitware, 2016-10-17:
yeeeees... and no. I think the upstream Python module is actually broken ATM pending lcm-proj/lcm#128. Since this issue has a work-around presently, I've been procrastinating until LCM and the lcmtypes stuff all settle in hopes of only having to bump everything once. At least, it's probably best to wait for lcm-proj/lcm#128.

wxmerkt, 2016-10-18:
https://github.com/lcm-proj/lcm/pull/128 is in so upstream LCM has Python fixed and this issue should be resolved by bumping LCM to the most recent master. Matlab bindings are still broken modulo https://github.com/lcm-proj/lcm/pull/137 and there isn't a fix for lcm-spy/bot-spy yet.

jwnimmer-tri, 2016-11-08:
I think we've bumped the hash recently.  Can this be fixed now?

jwnimmer-tri, 2016-11-20:
I will look into pushing this to a conclusion.

PR #4271 (2016-11-28): Fix automotive_demo paths to work again
- Fix binary executable paths to work again (regression from a few weeks ago when bindir changed).
- Remove the hack that works around #3397 break (thus; this fixes #3397) now that our LCM sha is new enough.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/4271)
closed 2016-11-28 by jwnimmer-tri