RobotLocomotion/drake #3406: Rename drake-distro/drake/examples/Cars to drake-distro/drake/mysomething

jwnimmer-tri, 2016-09-10:
I think the library of vehicular and/or automotive systems (and related toolkit elements) should be an in-tree tookit as part of Drake.  It will very soon outgrow a simple example, if it hasn't already.  Let's agree to move it into a new home at the `drake-distro/drake/mysomething` top level.  If there are any objections to that, speak up now.

(I will debate the name for `mysomething` in a second message, to better allow for voting and/or discussion.)

jwnimmer-tri, 2016-09-10:
My best suggestion is `drake-distro/drake/automotive`.  @RussTedrake do you have any feelings if that is an appropriate vocabulary to use for your vision?

liangfok, 2016-09-10:
I wonder whether we could create general bins that transcend "automotive". Something like:

* `drake-distro/drake/actuators`
* `drake-distro/drake/common`
* `drake-distro/drake/controllers`
* `drake-distro/drake/examples`
* `drake-distro/drake/force_elements`
* `drake-distro/drake/models` -  This may be a submodule per #3257.
* `drake-distro/drake/models_tri` - This may be an optional submodule per #3257.
* `drake-distro/drake/optimizers`
* `drake-distro/drake/parsers`
* `drake-distro/drake/perception`
* `drake-distro/drake/planners`
* `drake-distro/drake/plants`
* `drake-distro/drake/rigid_body_mechanics`
* `drake-distro/drake/road_networks`
* `drake-distro/drake/sensors`
* `drake-distro/drake/simulation` - maybe "solvers" instead of "simulation"?
* `drake-distro/drake/system_id`
* `drake-distro/drake/systems`

An automotive application could then pick-and-choose various components from each of the above bins.

jwnimmer-tri, 2016-09-10:
@liangfok I think many of those make sense, and should happen (or already exist), but with `mysomething` I'm looking for a home for the actual automotive-specific elements, and applications that incorporate them.  That is, all of the vehicular system blocks we're going to write, agent models, etc.  Generic planners or sensors and such should certainly be available outside of the `mysomething` directory.

liangfok, 2016-09-10:
:+1: for `drake-distro/drake/automotive`

RussTedrake, 2016-09-10:
I like “vehicles” better, but agree that it is not sufficient.  +1 for automotive.

jwnimmer-tri, 2016-09-11:
Yeah.  I tried to find a variant of "vehicles" that worked, but to no avail.  Thanks for the input.

maddog-tri, 2016-09-13:
How about "cars" instead of "automotive"?  awmomotiff is a mouthful and the 2.5x character count does not express any greater or deeper meaning than simply "cars".

liangfok, 2016-09-13:
Well, in my mind, "automotive" encompasses a wider set of vehicles than "cars". For example, I would include a truck in the "automotive" category but not in a "car" category.

PR #3436 (2016-09-13): Rename drake/examples/Cars to drake/automotive
Closes #3406.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/3436)
closed 2016-09-13 by jwnimmer-tri