RobotLocomotion/drake #3407: Reduce unit tests' CMake boilerplate

jwnimmer-tri, 2016-09-10:
Writing this over and over again is getting a bit boring:
```
add_executable(foo_bar_test foo_bar_test.cc)
target_link_libraries(foo_bar_test drakeSomething drakeSomething2)
add_test(NAME foo_bar_test COMMAND foo_bar_test)

add_executable(foo_baz_test foo_baz_test.cc)
target_link_libraries(foo_baz_test drakeSomething drakeSomething2)
add_test(NAME foo_baz_test COMMAND foo_baz_test)

add_executable(foo_quux_test foo_quux_test.cc)
target_link_libraries(foo_quux_test drakeSomething drakeSomething2)
add_test(NAME foo_quux_test COMMAND foo_quux_test)
```

I assume there's a way to wrap a macro around this, so as to write approximately this:
```
drake_libraries_required_by_this_subdirectory="drakeSomething drakeSomething2"
drake_add_cc_unit_test(foo_bar_test)
drake_add_cc_unit_test(foo_baz_test)
drake_add_cc_unit_test(foo_quux_test)
```

jwnimmer-tri, 2016-09-10:
@david-german-tri What do you think?  Could you delegate this?

david-german-tri, 2016-09-12:
I agree something like this would be nice.  I'm not sold on the directory-global deps list - different tests do have different deps - but it could easily be shifted to:

```
drake_add_cc_unit_test(foo_bar_test DEPENDS drakeSomething)
```
@jamiesnape and @billhoffman, does this seem like a reasonable practice to you?

jamiesnape, 2016-09-12:
I would certainly merge these parts:
```
add_executable(foo_baz_test foo_baz_test.cc)
target_link_libraries(foo_baz_test ${GTEST_BOTH_LIBRARIES})
add_test(NAME foo_baz_test COMMAND foo_baz_test)
```
Personally, I would leave the other `target_link_libraries` calls.

jamiesnape, 2016-09-12:
If we decide on a naming convention for test macros now, I can ensure #3306 conforms to avoid a million more merge conflicts as we proceed.

jwnimmer-tri, 2016-09-12:
I think #3306 only touches matlab tests, and this issue only touches C++ tests?

jamiesnape, 2016-09-12:
If you decide on `drake_add_cc_unit_test`, I would like the MATLAB to be `drake_add_matlab_unit_test`, for example.

jwnimmer-tri, 2016-09-12:
> ... different tests do have different deps ...
> ... Personally, I would leave the other target_link_libraries calls ...

If you look at `target_link_libraries` use in recent (systems/framework/test, examples/Cars/test) you will see the lines are identical for all unit tests.  This will be common.

If there are differences, I think they will be additive, not subtractive, in which case a single `target_link_libraries` for the additional item on that one test will be easy.  And in the _rare_ care they are subtractive, we could just skip the macro, or add an escape hatch to it.

david-german-tri, 2016-09-12:
`If you decide on drake_add_cc_unit_test, I would like the MATLAB to be drake_add_matlab_unit_test, for example.`
I prefer `drake_add_matlab_test` as currently written in #3306, so my vote would be to proceed with #3306 as is.

jwnimmer-tri, 2016-09-21:
Is there an existing example of this kind of macro in Drake that I could crib from?  The repetitiveness is making merging branches actively hard, so I guess I will write this myself.

jamiesnape, 2016-09-21:
Probably not. Create a ticket and assign it to me, then I will have a PR up by the end of the day.

jwnimmer-tri, 2016-09-21:
This ticket should suffice ;).  And thanks!

jamiesnape, 2016-09-21:
What do you all think of the signature of the function* in #3547? We could go further in terms of simplification and additional libraries to link in excess of `${GTEST_BOTH_LIBRARIES}`, or less far and allow arbitrary and multiple source file names rather than simply `<name>.cc`. I would suggest that we start with this, though.

(* which may or may not actually work, it is just up for feedback)

jwnimmer-tri, 2016-09-21:
I think this is a good start and would evict most of the boilerplate.

> ... allow arbitrary and multiple source file names rather than simply `<name>.cc`.

This appears to be rare.  Let's finish this pass and then see if we need it.

> ... additional libraries to link in excess of `${GTEST_BOTH_LIBRARIES}` ...

As in upthread, I think I would prefer this, but I'm more than happy to get this version landed and port us to use it, and then reconsider whether and how to compact it further in a second pass.

PR #3547 (2016-09-21): Add drake_add_cc_test function
Closes #3407.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/3547)
closed 2016-09-21 by jwnimmer-tri