RobotLocomotion/drake #3414: Finish carving out geometry and gradient from `drake/util` into new homes

jwnimmer-tri, 2016-09-12:
The goal is for these include statements to go away, replaced by paths into `common` or `util` (or removed) as necessary:
```
./systems/plants/constraint/RigidBodyConstraint.cpp:11:#include "drake/util/drakeGeometryUtil.h"
./systems/plants/joints/FixedAxisOneDoFJoint.h:12:#include "drake/util/drakeGradientUtil.h"
./systems/plants/joints/QuaternionFloatingJoint.h:7:#include "drake/util/drakeGeometryUtil.h"
./systems/plants/joints/RollPitchYawFloatingJoint.h:7:#include "drake/util/drakeGeometryUtil.h"
./systems/plants/RigidBody.cpp:5:#include "drake/util/drakeGeometryUtil.h"
./systems/plants/RigidBodyTree.cpp:13:#include "drake/util/drakeGeometryUtil.h"
./systems/plants/RigidBodyTree.h:26:#include "drake/util/drakeGeometryUtil.h"
./systems/plants/test/testKinematicsCacheChecks.cpp:11:#include "drake/util/drakeGeometryUtil.h"
./systems/robotInterfaces/QPLocomotionPlan.cpp:19:#include "drake/util/drakeGeometryUtil.h"
```

This may involved relocating the helper functions.

jwnimmer-tri, 2016-09-12:
/CC @david-german-tri and @hongkai-dai perhaps one of you could take this one over for us?

david-german-tri, 2016-09-12:
This parallelizes well.  I'm happy to work on chunks of it in the background.  Before starting to work on a piece of it, I'll post here; could others please do the same?

hongkai-dai, 2016-09-12:
I can handle the drakeGeometryUtil include

jwnimmer-tri, 2016-12-02:
Was this close intentional?  There's a bunch of non-forwarding-methods still left in drakeGeometryUtil.  Should we just delete them?

hongkai-dai, 2016-12-02:
@jwnimmer-tri you are right. I thought many of these functions can be replaced by using AutoDiffScalar, but we have not implemented the template version with AutoDiffScalar yet.

david-german-tri, 2017-04-13:
@hongkai-dai, I'm unassigning myself.

jwnimmer-tri, 2018-11-14:
Fixed in #9935 -- util is in the attic now.

closed 2018-11-14 by jwnimmer-tri