RobotLocomotion/drake #3419: Build Drake with  fatal error: rtwtypes.h: No such file or directory

wuyou33, 2016-09-12:
[ 20%] Built target expmap_test
[ 20%] Built target autodiff_test
[ 21%] Built target drakeGeometryUtil
[ 22%] Built target drakeUtil
[ 22%] Built target publishLCMLog
[ 22%] Building CXX object util/CMakeFiles/lcmLogger.dir/lcmLogger.cpp.o
In file included from /home/james/drake-distro/drake/util/lcmLogger.cpp:11:0:
/home/james/MATLAB/simulink/include/simstruc.h:153:22: fatal error: rtwtypes.h: No such file or directory
 #include "rtwtypes.h"
                      ^
compilation terminated.
make[5]: *** [util/CMakeFiles/lcmLogger.dir/lcmLogger.cpp.o] Error 1
make[4]: *** [util/CMakeFiles/lcmLogger.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [drake-prefix/src/drake-stamp/drake-build] Error 2
make[1]: *** [CMakeFiles/drake.dir/all] Error 2
make: *** [all] Error 2

RussTedrake, 2016-09-13:
this looks like an issue with the matlab headers.  not sure we’re the right ones to fix it for you..?

wuyou33, 2016-09-14:
@RussTedrake It has been fixed,Thanks.

closed 2016-09-14 by wuyou33