RobotLocomotion/drake #3421: Improve Robustness of Drake→DrakeVisualizer LoadRobot Message

liangfok, 2016-09-12:
Currently, the `LoadRobot` message is sent once and if it is lost, DrakeVisualizer has no way to recover. Periodically re-sending `LoadRobot` messages is not currently an option since it results in an expensive operation within DrakeVisualizer involving deleting all of DrakeVisualizer's prior state and rebuilding it from scratch.

Longer term, we'll want to generalize the communication protocol to allow objects to be dynamically added and removed from DrakeVisualizer.

rpoyner-tri, 2016-09-12:
More generally, drake-visualizer could likely benefit from an interface with a detail level somewhere between LoadRobot/DrawRobot and LCMGL. I may have a look at all this during the course of work on cars + road networks effort over the next week or two.

rpoyner-tri, 2016-09-12:
There is a work-around that one can do in launcher scripts to make sure that drake-visualizer is ready when the simulation process starts. I'll comb my notes and try post more info about that here.

rpoyner-tri, 2016-09-12:
Ah, yes, the work-around. When drake-visualizer is ready, it publishes a message on LCM channel DRAKE_VIEWER_STATUS, of type drake.lcmt_viewer_command, with command_type STATUS (== 0) and the command_data "loaded". A tolerably correct launcher should wait for this event before launching a program that sends DRAKE_VIEW_LOAD_ROBOT. 

Instead, lazy folk like me have written code like https://github.com/RobotLocomotion/drake/blob/master/drake/examples/Cars/run_demo_multi_car.sh#L30

liangfok, 2016-09-12:
Since LCM is based on UDP, there's no guarantee that the `DRAKE_VIEW_LOAD_ROBOT` message will get through even after waiting for a `DRAKE_VIEWER_STATUS` message. This unreliability suggests that the `DRAKE_VIEW_LOAD_ROBOT` should be re-sent if Drake can determine that the previous one didn't work, perhaps by analyzing the values within the `DRAKE_VIEWER_STATUS` it continues to periodically receive.

jwnimmer-tri, 2016-09-12:
In theory the `VIEWER_STATUS` might be lost.  In practice, UDP on localhost will be perfect (enough) to not worry about packet loss.  The only hazard to worry about is application launch order and process (re)start, which is solved by choosing different message primitives, not adding more ceremony around the current ones.

liangfok, 2016-09-12:
Are we assuming / requiring that Drake and DrakeVisualier always runs on the same host?

rpoyner-tri, 2016-09-12:
Unless you pass custom parameters to LCM, your UDP messages will have ttl=0 and never leave the local host.

liangfok, 2016-09-12:
Obviously, it would be nice to improve the communication protocol to not require that IPC messaging remain local and/or be reliable.

rpoyner-tri, 2016-09-12:
So we have several issues:

1. the current drake visualizer protocol is brittle (which i interpret as the core of #3421)
2. LCM is not a reliable transport (but good-enough on local host)
3. LCM's default configuration does not support remote operation

As for 1, we can probably make progress, but it will involve PR's against `director`.

I do not lose sleep over 2.

If 3 is a problem we need to solve, there are loads of ways to do it, but they require actual thought, because distributed anything is hard. I would argue that it is out of scope for this issue.

liangfok, 2016-09-12:
OK. I've renamed this PR to focus on brittleness with respect to process starting order, and I created #3422 to handle brittleness due to message loss.

rpoyner-tri, 2016-09-21:
I consider any work on this issue blocked, pending resolution of #3344.

david-german-tri, 2017-04-18:
I'm mildly skeptical there's actual work to do here, but I've reassigned it to @SeanCurtis-TRI for assessment.

liangfok, 2017-04-18:
I recommend closing this in favor of #4343.

closed 2017-04-18 by liangfok