RobotLocomotion/drake #3422: Improve Robustness of Drake→DrakeVisualizer LoadRobot Message In Terms of Message Loss

liangfok, 2016-09-12:
Currently, message loss may result in Drake Visualizer not receiving a `LoadRobot` message resulting in an unrecoverable error.

This issue should be addressed after #3421.

rpoyner-tri, 2016-09-12:
I don't understand the scope of what is wanted here? Are we "fixing" the properties of UDP multicast in LCM? Are we choosing another transport layer?

liangfok, 2016-09-12:
Nope. This issue does not replace LCM or its underlying use of UDP. It simply establishes a new Drake-level communication protocol that is not brittle against message loss.

liangfok, 2016-09-12:
For example, one idea is to modify Directory to tell Drake "hey I haven't received a `LoadRobot` message yet" or "hey, I just received a draw message containing links I'm not aware of, please send me another `LoadRobot` message".

jwnimmer-tri, 2016-09-12:
I cannot imagine a solution to #3421 that does not also solve this issue.

liangfok, 2016-09-12:
@rpoyner-tri proposed a launch script solution for #3421 that would not solve this.

jwnimmer-tri, 2016-09-12:
That's a work-around (on a per-demo basis), not a solution.

liangfok, 2016-09-13:
@jwnimmer-tri I clarified the difference by updating the descriptions of both this issue and #3421. PTAL.

jwnimmer-tri, 2016-09-13:
Okay.  I guess if #3421 assumes processes never restart, then some solutions there would not fix this issue.  I still find it likely that any reasonable rework of the message choices would solve both of these issues at once.

liangfok, 2016-09-13:
:+1: for a single PR that can fix both this issue and #3421. The only reason why I created two issues is because I sensed some pessism in https://github.com/RobotLocomotion/drake/issues/3421#issuecomment-246486286 that it could be solved in a single PR.

Here's to incremental PRs!

rpoyner-tri, 2016-09-21:
Solving this PR will require some pretty deep semantic changes to the protocol, since it would no longer be adequate to share various resources by file name. I am unconvinced that anyone actually wants this.

liangfok, 2016-09-21:
Note that solving this PR does **not** require ensuring that Drake and Drake Visualizer can run on different machines. However, this PR can be thought of as a baby step towards that direction.

I think this PR can be solved by a Drake-level protocol involving the retransmission of the `lcmt_viewer_load_robot` message and optimizations to Director that enables it to process such a message without reloading everything.

rpoyner-tri, 2016-09-21:
Well, if we're not considering ttl>0, I don't think message loss is even worth discussing. Anything else this issue could address is already covered by #3421.

closed 2016-09-21 by rpoyner-tri