RobotLocomotion/drake #3426: Feature request: make it easier to make an lcmtypes library

tkoolen, 2016-09-13:
I'd love to have a CMake function that takes a list of .lcm file names and a 'collection name', and then performs the tasks currently done in `drake/lcmtypes/CMakeLists.txt` for `drake_lcmtypes` (where what I'm calling the 'collection name' is 'drake').

My use case: I'd like to put new .lcm files specific to Valkyrie in `drake/examples/Valkyrie/lcmtypes`. I don't think they should be in `drake/lcmtypes`. So I need to do basically the same thing as in `drake/lcmtypes/CMakeLists.txt`, just in a different folder, with different .lcm files, and the generated headers should be installed to a different folder.

Related: https://github.com/RobotLocomotion/drake/issues/2486.

cc: @mwoehlke-kitware, could you let me know if you or someone else from Kitware has time for this within the next week? If not, I'll try to do it myself, or if you prefer, I'll just copy the code from `drake/lcmtypes/CMakeLists.txt` for now. Thanks!

mwoehlke-kitware, 2016-09-13:
I'll try to take a look tomorrow; probably can't today. This might make sense being added to LCM upstream...

mwoehlke-kitware, 2016-09-20:
On further thought, I'm less convinced that we should be hiding details like this. *Maybe* in whatever winds up being "the Drake LCM types repo", but even there, I'm not that big a fan of burying `add_library` and the like in a 'do lots of magic' function.

OTOH, having a well documented sample of how to create LCM binding libraries would certainly be helpful...

tkoolen, 2016-09-23:
I disagree. Having gone through the CMake tutorial, it really looks like creating LCM bindings libraries is a very straightforward process (as it should be), and should be done in one very specific way. If there's no function that does the setup for you, then the same boiler plate code will be copied all over the place, making it harder to maintain. Either that or people will get lazy and put all the LCM types into one big library, as is currently the case.

mwoehlke-kitware, 2016-10-07:
> it really looks like creating LCM bindings libraries [...] should be done in one very specific way

That's just the thing; that's *not* the case.

Take a simple example (sneak peak of some coming changes):

```cmake
lcm_wrap_types(
  C_SOURCES sources
  C_HEADERS c_install_headers
  CPP_HEADERS cpp_install_headers
  ...some .lcm files...
)

lcm_add_library(my-lcmtypes C ${sources} ${c_headers})
target_include_directories(my-lcmtypes INTERFACE
  $<BUILD_INTERFACE:${CMAKE_CURRENT_BINARY_DIR}>)

lcm_add_library(my-lcmtypes-cpp CXX ${cpp_headers})
target_include_directories(my-lcmtypes-cpp INTERFACE
  $<BUILD_INTERFACE:${CMAKE_CURRENT_BINARY_DIR}>)

lcm_install_headers(
  ${c_install_headers}
  ${cpp_install_headers}
  DESTINATION include/lcmtypes
)

install(TARGETS my-lcmtypes my-lcmtypes-cpp
  EXPORT ${PROJECT_NAME}Targets
  RUNTIME DESTINATION bin
  LIBRARY DESTINATION lib${LIB_SUFFIX}
  ARCHIVE DESTINATION lib${LIB_SUFFIX}
  INCLUDES DESTINATION include/lcmtypes
)
```

...but wait! Did you want to force the C library to be `STATIC`? `SHARED`? (`MODULE`?) What if you want to compile additional sources into the library? Do you want Python bindings always, if Python is available, or never? Where did you want to install the headers? Do you believe in `LIB_SUFFIX` or `LIB_INSTALL_DIR`? How do you name your install exports?

*All of these things are variable*, and hence, the user needs to be able to specify them. It's well known programming practice in C++ that it is better to have several small functions that clearly state what they are doing than one large, mysterious function (not to mention that said monolithic function is much more complicated, leading to increased maintenance costs). Consider:

```C++
  // Bad: what does this do??
  auto s = new QSpinBox(layout, "spin", 5, 0, 100, 1, 10);

  // Better
  auto s = new QSpinBox(layout);
  s->setName("spin");
  s->setRange(0, 100);
  s->setValue(5);
  s->setSingleStep(1);
  s->setPageStep(10);
```

Not only is it easier to maintain the implementation in the second case, but it is *much* more clear to someone reading the logic what is going on.

(That said, it occurs to me that a) `C_INCLUDE` support is broken, and b) `lcm_add_library` could take care of the `target_include_directories` also... but I remain opposed to trying to have one function generate the bindings *and* create one or more libraries and/or JAR's *and* install things. Such a function would be a towering unstable monolith of a black box that does no one any favors. There's a reason CMake itself has no built-in functions like that.)

mwoehlke-kitware, 2016-10-07:
> That said, it occurs to me that [...] `lcm_add_library` could take care of the `target_include_directories` also

Oops, no, I take that back... the include directory is also variable. (The directory to which the bindings are generated is variable.) The above is as simplified as is reasonably possible, at least for LCM. (As previously stated, a project like Drake might "enforce" conventions that would allow further abstraction, but LCM itself does not, and *should* not.)

tkoolen, 2016-10-07:
Alright, good arguments. I thought there not that many things would be variable within the context of Drake. I think the example CMake code is actually quite nice like this. And I suppose if all you need is C++ it'll actually be pretty short. (I was wondering where `c_headers` and `cpp_headers` come from though.)

mwoehlke-kitware, 2016-10-07:
> I thought there not that many things would be variable within the context of Drake.

*Within Drake*, there are not. That's why I proposed that possibly *Drake* (not LCM itself) could abstract some (more) of this away. (In particular, I'd like a `drake_install_libraries` that would have all the exports and library destinations. Ideally, all libraries, not just LCM libraries, would use this.)

> I was wondering where `c_headers` and `cpp_headers` come from though.

Those are just the variable names that receive the lists of those files that `lcm_wrap_types` generates (and are later used to build the libraries and install headers). They could as easily be `foo` and `bar`, although that would be... odd :smile:. (For example, I observe now that the above example, adapted from [lcm](https://github.com/lcm-proj/lcm)/[test/types/CMakeLists.txt](https://github.com/lcm-proj/lcm/blob/debbad5392e13ae4bda0475de40dc0efb1743656/test/types/CMakeLists.txt), uses `sources`, where I'm sure I meant to use `c_sources`.) If you've used any of the functions from [FindQt4](https://cmake.org/cmake/help/v3.6/module/FindQt4.html), those tend to follow the same convention (although the output variables there are positional rather than named arguments).

tkoolen, 2016-10-07:
Ah OK, I think you just meant to use `c_install_headers` and `cpp_install_headers` in the example above.

mwoehlke-kitware, 2016-10-07:
Well... sort-of :smile:. In [lcm](https://github.com/lcm-proj/lcm)/[test/types/CMakeLists.txt](https://github.com/lcm-proj/lcm/blob/debbad5392e13ae4bda0475de40dc0efb1743656/test/types/CMakeLists.txt), which I copied for the above example, they actually *aren't* installed, so are "correct" (again, keeping in mind this is all just convention) in that context. But yes, I "should" have added `_install` for the above example to be consistent with convention.

jwnimmer-tri, 2016-11-08:
I just happened upon this ticket, and thought I'd drop a little FYI that our beta Drake-Bazel build system supports this: https://github.com/RobotLocomotion/drake/blob/6c591ea1bbad36518900570a90d0f29f00027a43/drake/lcmtypes/BUILD#L15

Once I've removed / replaced CMake support for those LCM messages, then those `*.lcm` files can move around within the tree (i.e., not all have to be dumped into `drake-distro/drake/lcmtypes`), and they can be rolled up into separately-named libraries, etc.

PR #6738 (2017-09-29): Purge CMake not used by MATLAB
Toward #3129. Closes #3342, closes #3426, closes #4113, closes #4525, closes #5356, and closes #5572.

----
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/6738)
closed 2017-09-29 by jamiesnape