RobotLocomotion/drake #3429: ROS CI Unable to Log Due to Insufficient Permissions

liangfok, 2016-09-13:
# Problem Definition

The error below was taken from the console output: https://drake-jenkins.csail.mit.edu/job/linux-clang-experimental-ros/62/console

```
10:41:51 2: WARNING: cannot create log directory [/home/ubuntu/workspace/linux-clang-experimental-ros/log]. Please set ROS_LOG_DIR to a writable location.
10:41:52 2: Traceback (most recent call last):
10:41:52 2:   File "/opt/ros/indigo/share/rostest/cmake/../../../bin/rostest", line 36, in <module>
10:41:52 2:     rostestmain()
10:41:52 2:   File "/opt/ros/indigo/lib/python2.7/dist-packages/rostest/__init__.py", line 273, in rostestmain
10:41:52 2:     _main()
10:41:52 2:   File "/opt/ros/indigo/lib/python2.7/dist-packages/rostest/rostest_main.py", line 167, in rostestmain
10:41:52 2:     results_file = xmlResultsFile(pkg, outname, is_rostest, env=env)
10:41:52 2:   File "/opt/ros/indigo/lib/python2.7/dist-packages/rosunit/core.py", line 102, in xml_results_file
10:41:52 2:     raise IOError("cannot create test results directory [%s]. Please check permissions."%(test_dir))
10:41:52 2: IOError: cannot create test results directory [/home/ubuntu/workspace/linux-clang-experimental-ros/build/drake_ros_common_test/test_results/drake_ros_common_test]. Please check permissions.
```

# Question

Should environment variable `ROS_LOG_DIR` be set to a different location?

liangfok, 2016-09-13:
I'm going to elevate the priority of this issue to high because I just realized that it is causing #3409 to fail CI.

jhoare, 2016-09-13:
``ROS_HOME`` is set to be ``DASHBOARD_WORKSPACE`` which is where all jenkins build artifacts should be. https://github.com/RobotLocomotion/drake-ci/blob/master/ctest_driver_script_catkin.cmake#L110

jhoare, 2016-09-13:
It looks like re-running this in #3409 worked correctly, so it seems like there was something weird at the OS level?

liangfok, 2016-09-13:
Yes, something strange at the OS level. Since it passed the second time around, I'll downgrade the priority of this PR back to low.

jwnimmer-tri, 2016-10-31:
Maybe close this, as unreproducible?

liangfok, 2016-11-05:
This issue just occurred in #4040. I will leave it closed for now unless it becomes more problematic.

liangfok, 2016-11-05:
Since this issue just occurred in #4012 (https://drake-jenkins.csail.mit.edu/job/linux-gcc-experimental-ros/531/), I shall re-open this.

liangfok, 2016-11-08:
@hongkai-dai just encountered this issue here:

https://drake-jenkins.csail.mit.edu/job/linux-gcc-experimental-ros/653/consoleFull

liangfok, 2016-11-23:
https://drake-jenkins.csail.mit.edu/job/linux-gcc-experimental-ros/1116/consoleFull

jonathan-owens, 2016-11-29:
https://drake-cdash.csail.mit.edu/testDetails.php?test=5050820&build=221987

jonathan-owens, 2016-11-29:
https://drake-cdash.csail.mit.edu/testDetails.php?test=5068729&build=222847

liangfok, 2016-11-29:
@SeanCurtis-TRI experienced this here:
https://drake-jenkins.csail.mit.edu/job/linux-gcc-experimental-ros/1293/console

Lucy-tri, 2016-12-05:
https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-gcc-nightly-matlab-ros/130/console

liangfok, 2016-12-05:
Again: https://drake-jenkins.csail.mit.edu/job/linux-gcc-continuous-ros-debug/722/console

liangfok, 2016-12-06:
Again: https://drake-jenkins.csail.mit.edu/job/linux-gcc-experimental-ros/1522/

liangfok, 2016-12-27:
Again: https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-gcc-continuous-ros-debug/846/consoleFull

mwoehlke-kitware, 2017-01-03:
https://drake-jenkins.csail.mit.edu/job/linux-gcc-continuous-ros-release/881/

liangfok, 2017-01-07:
https://drake-jenkins.csail.mit.edu/job/linux-clang-continuous-ros-debug/954/

liangfok, 2017-01-14:
https://drake-jenkins.csail.mit.edu/job/linux-clang-continuous-ros-release/985/

liangfok, 2017-01-16:
https://drake-jenkins.csail.mit.edu/job/linux-gcc-experimental-ros/2662/

liangfok, 2017-01-16:
https://drake-jenkins.csail.mit.edu/job/linux-gcc-experimental-ros/2666/

jonathan-owens, 2017-01-17:
https://drake-jenkins.csail.mit.edu/job/linux-clang-nightly-ros-release/174/

liangfok, 2017-01-18:
https://drake-jenkins.csail.mit.edu/job/linux-gcc-experimental-ros/2700/

david-german-tri, 2017-01-26:
https://drake-jenkins.csail.mit.edu/job/linux-gcc-experimental-ros/2981/

liangfok, 2017-02-13:
https://drake-jenkins.csail.mit.edu/job/linux-gcc-experimental-ros/3600

liangfok, 2017-02-28:
https://drake-jenkins.csail.mit.edu/job/linux-gcc-experimental-ros/4203/

liangfok, 2017-03-01:
https://drake-jenkins.csail.mit.edu/job/linux-gcc-continuous-ros-debug/1215/

liangfok, 2017-03-01:
https://drake-jenkins.csail.mit.edu/job/linux-gcc-experimental-ros/4296

liangfok, 2017-03-24:
https://drake-jenkins.csail.mit.edu/job/linux-gcc-experimental-ros/5220

liangfok, 2017-03-25:
https://drake-jenkins.csail.mit.edu/job/linux-gcc-experimental-ros/5252

liangfok, 2017-03-26:
https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-clang-continuous-ros-debug/1447/

PR #6783 (2017-08-15): Remove ROS from within Drake
Closes #3418, closes #3429, closes #4711, closes #5579, and closes #6674.

Relates in part to #5012.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/6783)
closed 2017-08-15 by sammy-tri