RobotLocomotion/drake #3433: Report: CMake sometimes finds gflags in a different Drake checkout

david-german-tri, 2016-09-13:
I'm going to try to reproduce and then add detailed steps here.

@ggould-tri @sammy-tri FYI

ggould-tri, 2016-09-13:
Thanks, David.  Recap of my experience below:

- I had a clone of my personal drake fork in `~/git/drake-ggould`, which was up-to-date
- I had a clone of Naveen's personal drake fork in `~/git/drake-naveen`, which was two weeks out of date
- I built in the top level of my clone `~/git/drake-ggould` using https://github.com/ggould-tri/public-sandbox/blob/master/Scripts/build-drake.sh (which pretty much just encapsulates the documented build process)
- The build failed due to a linker error on `gflags` and the output indicated that it was using the `libgflags.a` from `~/git/drake-naveen`

Unfortunately I then concluded the problem was that I had done something wrong and deleted all of the directories involved.  Only when @sammy-tri mentioned the same problem did I realize otherwise.

jwnimmer-tri, 2016-09-13:
My local builds also have buckets-o-fun going on with gflags linking now (linebreaks added for readability):
```
jwnimmer@call-cc:~/jwnimmer-tri/drake-distro/build/debug/drake/bin$
readelf -d rigidBodyLCMNode
...
Library rpath: [
  /home/jwnimmer/jwnimmer-tri/drake-distro/build/debug/drake/lib:
  /home/jwnimmer/jwnimmer-tri/drake-distro/build/clang-debug/externals/gflags/lib:
  /home/jwnimmer/jwnimmer-tri/drake-distro/build/debug/install/lib:]
```

See how a different build variant crept in there?

PR #3440 (2016-09-14): Prevent CMake from consulting the User Package Registry
This helps `find_package(gflags)` find gflags in the current tree, not some other random tree.  Fixes #3433, though builds that have already been bitten will need to scrub `gflags_DIR` from `CMakeCache.txt` manually.

https://gflags.github.io/gflags/#cmake

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/3440)
closed 2016-09-14 by david-german-tri