RobotLocomotion/drake #3435: Drake Proper Should Not Use ROS Environment Variables

liangfok, 2016-09-13:
# Problem Definition

Drake's `drake-distro/drake/systems/plants/xmlUtil.cpp` currently contains a method called [populatePackageMap()](https://github.com/RobotLocomotion/drake/blob/master/drake/systems/plants/xmlUtil.cpp#L307) that accesses the following two environment variables that belong to ROS:

1. ROS_ROOT
2. ROS_PACKAGE_PATH

This causes problem when you have multiple development clones of Drake one of which is _inside_ a ROS Catkin workspace and while the others are _outside_ of a ROS Catkin workspace.

Basically, the problem I ran into is one where the clones of Drake that are outside of the ROS Catkin workspace end up searching for resources belonging to Drake inside a Catkin workspace. This problem could be avoided by modifying Drake proper to not make use of ROS' environment variables.

# Proposed Solution

Modify Drake proper to not access ROS environment variables. Remove the `populatePackageMap()` method.

Instead, the path information contained within the aforementioned ROS environment variables can passed to Drake via command-line arguments. This will allow me to ensure that clones of Drake outside of a ROS Catkin workspace to not attempt to access resources belonging to another clone of Drake in a ROS Catkin workspace.

liangfok, 2016-11-20:
@jwnimmer-tri informs me that he ran into a seemingly infinite loop when executing `ctest -VV -R rigid_body_constraint` after performing a Bazel build. I suspect this may be caused by `populatePackageMap()`, but haven't verified this.

patmarion, 2016-11-27:
Quick proposal: the call to populatePackageMap() should be removed from the urdf parser.  Instead, an application could manually call populatePackageMap("ROS_PACKAGE_PATH") if it needed to populate a package map with that environment variable, then pass the result to the urdf parser.  This is how Director functions, it populates a package map in its own application specific way, then passes the data to drake's URDF parser.  Currently, populatePackageMap() is always called by parser_urdf.cc, even if the caller passed in a fully populated package map.

patmarion, 2016-11-27:
Note, drake's examples will still need to call populatePackageMap() in order to maintain their functionality, but at least they would have the option to restrict the search to a specific location (like a path given on the command line).

PR #4363 (2016-12-12): Removes Downward Package Crawling From Parsers
This PR eliminates the "downward crawling" that was being done by the parsers as they search for ROS packages. Instead of crawling down the directory tree for the packages, the parsers now crawl up provided the original directory is within `drake-distro`. Alternatively, applications can provide a pre-populated package map. A utility method is provided that populates this map with models in `drake/examples/`. Thus this PR resolves #3435.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/4363)
closed 2016-12-12 by jwnimmer-tri