RobotLocomotion/drake #3444: examples/PR2/runPassive are failing on linux-matlab-open-source builds

jhoare, 2016-09-14:
It appears that after https://github.com/RobotLocomotion/drake/pull/3262 was merged, the linux-matlab-open-source continuous CI builds have had a test fail for ``examples/PR2/runPassive``

See:
https://drake-jenkins.csail.mit.edu/view/Continuous/job/linux-gcc-continuous-matlab-open-source/653/
https://drake-jenkins.csail.mit.edu/view/Continuous/job/linux-gcc-ninja-continuous-matlab-open-source/563/

tkoolen, 2016-09-14:
I can't reproduce this locally, with GCC and Matlab 2016a (nor with clang on OSX with 2016a). The CI builds use 2015b, so that could be a difference. This one is a bit frustrating.

tkoolen, 2016-09-14:
@hongkai-dai, do you have any insights? I don't know how to debug this.

tkoolen, 2016-09-14:
Related, I suppose: https://github.com/RobotLocomotion/drake/issues/1702.

jhoare, 2016-09-14:
Are you also building in the "open-source" configuration? The regular matlab builds are passing fine, is it possible the "open-source" configuration is using a different solver or something for this test?

tkoolen, 2016-09-14:
There might be differences, yes. How do I build exactly what's built in CI?

jhoare, 2016-09-14:
Check out the drake-ci repository into a `ci` subfolder within `drake-distro`, edit the `run_local.bash` to have the right paths/environment variables, and run the shell script. 

https://github.com/RobotLocomotion/drake-ci/blob/master/run_local.bash

tkoolen, 2016-09-14:
Thanks, didn't know that, I'll try again.

jwnimmer-tri, 2016-09-15:
If we can't fix-forward this today, I propose that we either disable the test, or revert the PR.  Having a broken build in Continuous is undesirable.

tkoolen, 2016-09-15:
I still can't reproduce this locally. I tried the `run_local.bash` thing with
```
 export compiler=gcc
 export coverage=false
 export debug=false
 export documentation=false
 export matlab=true
 export minimal=false
 export openSource=true
 export ros=false
 export track=experimental
```

and the test passes (if I run it by itself). Remaining differences between CI and my setup include Matlab R2015b vs my Matlab R2016a, and running all tests in succession instead of just `examples/PR2/runPassive.m`.

jwnimmer-tri, 2016-09-15:
Is there any chance a `CMakeCache.txt` would have contained stale values, defeating the `run_local.bash` attempts to, e.g., turn off snopt.

tkoolen, 2016-09-15:
> disable the test

Sounds good to me.

tkoolen, 2016-09-15:
@jwnimmer-tri, nope, clean checkout and build. Also made sure that `checkDependency('gurobi')`, `checkDependency('fastQP')`, and `checkDependency('snopt')` fail.

jwnimmer-tri, 2017-06-19:
Closing as won't-fix, because #6369 will be removing the relevant software.

If I'm confused and this should in fact stay open, please re-open an clarify the remaining unsolved problem(s) in a new comment here.

closed 2017-06-19 by jwnimmer-tri